roleoroleo/yi-hack-MStar #392: no /etc/resolv.conf after flash

cheese1, 2021-10-06:
hi

some days ago my camera (on older version 0.4.0 or 0.4.1) started to boot-loop. 
i upgraded to 0.4.3, which seemed to work, but no some other troubles started: (no more bootloop):
* i cannot connect via official yi home app
i think the reason is that my / partition got full and i have now "no space left on device".
```
ln -s /tmp/resolv.conf /etc/resolv.conf
ln: /etc/resolv.conf: No space left on device
```
i no longer am able to flash to an older version, the cam simply reboots and very fast boots into 0.4.3
i CAN use the web-interface
i can ssh into the cam

etc is:
```
/home/app # ls -l /etc/
total 5
lrwxrwxrwx    1 root     0               20 Aug 30 21:31 TZ -> /home/yi-hack/etc/TZ
drwxrwxr-x    3 root     0                0 Mar  4  2019 Wireless
lrwxrwxrwx    1 root     0               10 Aug 30 21:31 back.bin -> /fake/file
lrwxrwxrwx    1 root     0               26 Aug 30 21:31 hostname -> /home/yi-hack/etc/hostname
lrwxrwxrwx    1 root     0               10 Aug 30 21:31 hosts -> /tmp/hosts
drwxrwxr-x    2 root     0                0 Aug 30 21:31 init.d
-rwxrwxr-x    1 root     0               51 Mar  4  2019 inittab
lrwxrwxrwx    1 root     0               24 Aug 30 21:31 passwd -> /home/yi-hack/etc/passwd
-rw-r--r--    1 root     0             1267 Aug 30 21:31 profile
```

tmp:
```
/home/yi-hack/etc # ls -l /tmp/
total 100
-rw-------    1 root     0                9 Jan  1  1970 MTK
lrwxrwxrwx    1 root     0               23 Jan  1  1970 audio -> /home/app/audio_file/us
prw-------    1 root     0                0 Jan  1  1970 audio_in_fifo
-rw-------    1 root     0                0 Jan  1  1970 audio_in_fifo.requested
-rw-------    1 root     0              222 Sep 16 15:06 debug_alarm.txt
-rw-------    1 root     0                0 Jan  1  1970 debug_oss.txt
-rw-------    1 root     0              137 Sep 16 14:57 debug_p2p.txt
-rw-------    1 root     0               24 Sep 16 15:17 dns1
-rw-------    1 root     0                2 Sep 16 15:17 done
-rw-------    1 root     0               13 Sep 16 15:17 gw1
prwx------    1 root     0                0 Sep 16 14:57 h264_high_fifo
-rw-r--r--    1 root     0               32 Sep 16 15:21 hosts
-rw-------    1 root     0                0 Jan  1  1970 httpd.conf
-rw-------    1 root     0            18112 Sep 16 15:21 log.txt
-rw-------    1 root     0                0 Jan  1  1970 log_oss.txt
srwx------    1 root     0                0 Jan  1  1970 logsock
-rw-------    1 root     0               14 Sep 16 15:17 mask
-rwx------    1 root     0             2592 Jan  1  1970 mmap.info
-rw-------    1 root     0              624 Sep 16 14:57 onvif_srvd.conf
-rw-------    1 root     0               65 Sep 16 15:17 resolv.conf
drwxr-xr-x    6 root     0            32768 Jan  1  1970 sd
drwx------    4 root     0               80 Sep 16 14:57 var
-rwxrwxrwx    1 root     0              190 Jan  1  1970 wpa_supplicant.conf
```

i CAN fake a working dns by creating /tmp/hosts, but i dont know which ips i would need to e.g connect via app (and maybe reflash stock)

i COULD use upgrade_firmware, if i knew what parameters i would need.

any suggestions?

yours
josef

cheese1, 2021-10-06:
deleting iperf did NOT solve the "no space"-problem.

roleoroleo, 2021-10-07:
This is strange.
The upgrade procedure is made by the bootloader.
If the sd card is correctly detected, the upgrade must start.

Try this:
[h201c_0.4.3.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7304986/h201c_0.4.3.tar.gz)

cheese1, 2021-10-08:
the sd-card should have been correctly used as i can record on that and remove/add files.
as i'm currently on 0.4.3 your fresh h201c_0.4.3.tar.gz wont help, as its the same version and therefore cannot get flashed. i would need some other version.

anyway. i will try to format the sd and start fresh.

bepece1, 2021-10-09:
> holy cow! whatever it was: i now have a completly working camea again. steps:
> 
> * repartition (remove partition and recreate) sd-card with a 12G partition
> * format fat32
> * add your new files
> * boot up the camera
> * i think i have recognised two distinct flashes, a shorter first on and a longer second one
> * the camera rebooted and did their left/right look
> * voila, working
>   thanks
>   i'm very grateful
> 
> ```
> /home/yi-hack # df -h
> Filesystem                Size      Used Available Use% Mounted on
> /dev/root                 1.9M      1.8M     48.0K  98% /
> tmpfs                    29.7M      1.8M     27.9M   6% /dev
> /dev/block/mtdblock3     12.2M     10.8M      1.4M  88% /home
> tmpfs                    32.0M    104.0K     31.9M   0% /tmp
> /dev/block/mmcblk0p1     11.7G     66.2M     11.6G   1% /tmp/sd
> /home/yi-hack # ls -l /etc/ /tmp/sd/
> /etc/:
> total 5
> lrwxrwxrwx    1 root     0               20 Oct  7 15:20 TZ -> /home/yi-hack/etc/TZ
> drwxrwxr-x    3 root     0                0 Mar  4  2019 Wireless
> lrwxrwxrwx    1 root     0               10 Oct  7 15:20 back.bin -> /fake/file
> lrwxrwxrwx    1 root     0               26 Oct  7 15:20 hostname -> /home/yi-hack/etc/hostname
> lrwxrwxrwx    1 root     0               10 Oct  7 15:20 hosts -> /tmp/hosts
> drwxrwxr-x    2 root     0                0 Oct  7 15:20 init.d
> -rwxrwxr-x    1 root     0               51 Mar  4  2019 inittab
> lrwxrwxrwx    1 root     0               24 Oct  7 15:20 passwd -> /home/yi-hack/etc/passwd
> -rw-r--r--    1 root     0             1267 Oct  7 15:20 profile
> lrwxrwxrwx    1 root     0               16 Oct  7 15:20 resolv.conf -> /tmp/resolv.conf
> 
> /tmp/sd/:
> total 24272
> -rwx------    1 root     0         12284537 Oct  7 15:28 h201c_0.4.3.tar.gz
> -rwx------    1 root     0         10816240 Oct  7 15:21 home_h201c
> drwx------    2 root     0             8192 Oct  8 15:37 log
> drwx------    3 root     0             8192 Oct  8 15:41 record
> -rwx------    1 root     0          1717492 Oct  7 15:21 sys_h201c
> drwx------    3 root     0             8192 Jan 14  2021 yi-hack
> ```
> 
> thanks

What allocation unit size has you used when formatting sd flash in FAT32? @cheese1 

[image]

closed 2024-05-08 by github-actions[bot]