roleoroleo/yi-hack-MStar #399: Please add proxychains settings in GUI

areyoufserious, 2021-10-19:
Adding and testing proxychains servers via ssh takes a lot of time and makes insane
please add it on UI like in other hacks, just a copy solution

vin-w, 2021-11-04:
i voted for this feature too.
And if possible I hope the backup could be able to restore the proxychains.conf too. This could help if people reset and restore the camera with proxychain on.

areyoufserious, 2021-11-04:
> i voted for this feature too. And if possible I hope the backup could be able to restore the proxychains.conf too. This could help if people reset and restore the camera with proxychain on.
thanks
where I can vote?

roleoroleo, 2021-11-05:
It's not hard but I have to find the time.

roleoroleo, 2021-12-07:
https://github.com/roleoroleo/yi-hack-MStar/commit/92899f999f289043476198d517ab8f0645297abb

closed 2024-05-08 by github-actions[bot]