roleoroleo/yi-hack-MStar #405: Camera isnt showing in the official app but is still online

Cyberflow187, 2021-11-15:
Good evening,

I don't have no connection thru the Yi app on my cellphone or the desktop to see the feed of the camera.

But when i log onto the webserver of the camera (http://192.xxx.xxx.xxx:8080) it gives me access to the camera even though i don't see it in my official apps.

Did anyone report this issue already ?

I did a factory reset on one of the cameras and now i don't have anymore access to it.
Is it possible to fetch the official firmware and reinstal it ?

I have the impression Yi is filtering and blocking cameras that are modified.

roleoroleo, 2021-11-15:
> Is it possible to fetch the official firmware and reinstal it ?

You can restore the original firmware if you have a backup.
Following the same upgrade procedure you used for the installation of the hack.

> I have the impression Yi is filtering and blocking cameras that are modified.

If you switch off the switch "Disable Cloud" the cam should connect to the yi app.
I think it's not possible that Yi blocks hacked cameras.
I tested now my y203c: as soon as I took the switch off and rebooted, the cam appeared in the app.

What's your model?
What's your serial? US or CN?

nasnaynz, 2021-11-20:
i have the same problem as this, i already switch off disable the cloud but still not showing in my yi home app

my model : yi home camera 1080p
my serial : 6fcny2112d

ethempekin, 2021-12-09:
With h201c camera On Yi app I get offline error but homekit homebridge work without problems. Strange that date of the camera is wrong.

roleoroleo, 2021-12-10:
> my model : yi home camera 1080p my serial : 6fcny2112d

Based on your model, your cam is for use in China.
Probably you need to configure proxychain (if the cam is not in China).

> With h201c camera On Yi app I get offline error but homekit homebridge work without problems. Strange that date of the camera is wrong.

Maybe if the date/time is not ok the cam doesn't connect, I don't know if the authentication is based on a certificate.

ethempekin, 2021-12-10:
It is rather 2020-04-17 in my case the date. I should rather reinstall the yi-hack for this camera? All started when I repositioned it.

roleoroleo, 2021-12-11:
> It is rather 2020-04-17 in my case the date. I should rather reinstall the yi-hack for this camera? All started when I repositioned it.

Yes, try to reinstall the hack.

sphi-nx, 2022-01-17:
Hi @roleoroleo I'm having the same issue.

I can connect to the camera using the IP address at port 8080 but the app can't connect to it. It has been working fine for week but just stopped.

I don't have a backup of the original firmware so i can't install it. I have also tried to install the hack again with a formated and partitioned SD Card but it doesn't seem to  run. Instead I get the message "Welcome to YI Home Camera", followed by "Waiting to connect". 

I also noticed that the date on the camera is a day in the future. I'm not sure if that's the cause of the problem because my other cameras don't exactly have the accurate dates. I have tried to modify the date by changing the timezone but its not working.

My model is y203c and my serial number starts with BFUS

@ethempekin did you ever get your camera to work?

roleoroleo, 2022-01-18:
You can't reinstall the same version, the bootloader checks the hash.
Try installing a previous version.
And then update again.

alextircovnicu, 2022-02-17:
Hi,

Same issue here. It worked flowless for a while then suddenly the camera stopped working from the yi app (although it was working fine on rtsp) . I have deleted the camera from the app and since then  I am unable to pair it again. 
It is scanning the QR code, it is connecting to the wifi (rtsp works fine) but I am ending up with "Pairing time out".  WiFi connection is not persisted on power loss - which makes camera useless.   

Firmware Version  0.3.6
Base Version  4.2.0.0H_201909041620
Model Suffix  y25
Serial Number  9FUSY25XJFNTDX191010
Local Time| Tue Jan 19 06:54:20 Universal 2016
Uptime  0h 12m 58s
Load Average  2.74 2.35 1.35
Free/Total Memory  17104/60824 KB
Free SD Space  66%

sphi-nx, 2022-02-17:
@alextircovnicu try reinstalling an older version. This worked for me when I had the same issue with one of my cameras. Thanks, @roleoroleo for the suggestion.
I may have to do the same for two of my other cameras which started acting up this week.
@roleoroleo is there a way to uninstall the hack?

roleoroleo, 2022-02-17:
To uninstall the hack you have to put sys and home partitions to the sd card.
Like a hack installation, but with the original partitions.

sphi-nx, 2022-02-17:
Thanks for the reply @roleoroleo
Do you mean the original home_y203c and sys_y203c files?
Unfortunately, I don't have those. Anyway I can get them?

roleoroleo, 2022-02-17:
Send me your email to my paypal address (see the main page).

Cyberflow187, 2022-02-17:
@roleoroleo : I can't find your paypal email btw

roleoroleo, 2022-02-18:
> 1080p Home Camera YYS.2016
> 1080p Dome Camera - YHS.2016

Suffix?

sphi-nx, 2022-02-18:
@roleoroleo I think I found your email yesterday and sent you an email.

Looking forward to receiving the files.

Thanks a lot

roleoroleo, 2022-02-19:
Ok, forget the paypal address.
For y203c: https://drive.google.com/file/d/1KQHMBuAel5ZZL4qDgo-rDv8CH9biaHNG/view?usp=sharing
Let me know when you download the file, I will remove this link.

@Cyberflow187 
I need the suffix of your cams.

Cyberflow187, 2022-02-19:
Hi @roleoroleo 
Sorry for the delay i had to go uninstal the camera from the ceilling to go fetch the suffix : 
9FUSY
BFUSY

Cyberflow187, 2022-02-19:
Here's the information for the file prefix : 
9FUSY : y25
BFUSY : h201c

Cyberflow187, 2022-02-19:
[image]
Is this the correct way to configure the cloud by the way ?

Cyberflow187, 2022-02-19:
I'm freaking out because i got 6 cameras at home that i can't use with the app now

But i noticed tonight i can connect to them via its ip address

roleoroleo, 2022-02-20:
y25:
https://drive.google.com/file/d/10QehqMhw0J6gKPQs_yfiGBGCacPTh8vT/view?usp=sharing
h201c:
https://drive.google.com/file/d/1DT-MYfy9N4_oSJ7CvIArSyv5e98ABOi3/view?usp=sharing

> Is this the correct way to configure the cloud by the way ?

If you don't want to use the app and you don't want to save videos, is ok.

Cyberflow187, 2022-02-20:
> y25: https://drive.google.com/file/d/10QehqMhw0J6gKPQs_yfiGBGCacPTh8vT/view?usp=sharing h201c: https://drive.google.com/file/d/1DT-MYfy9N4_oSJ7CvIArSyv5e98ABOi3/view?usp=sharing
>
>
> If you don't want [...]

Ok ok 
I thought that this maybe had something to do with the cameras not working at all.

So i'm going to install the original system file and relink everything see if i can get it up and fuctionning.

Then re-install the Yi-Hack 

But if i check that "Disable Cloud" that would be why i dont have the camera on the app right now no ?

Btw, thank you for helping us out trully appreciate it !

Cyberflow187, 2022-02-20:
Btw, i got the 0.3.9 version of Yi-Hack installed

Wouldn't hurt if i upgrade that

roleoroleo, 2022-02-20:
> But if i check that "Disable Cloud" that would be why i dont have the camera on the app right now no ?

Right.

> Wouldn't hurt if i upgrade that

Yes, update it.

Cyberflow187, 2022-02-20:
Will keep you posted @roleoroleo 

Thx again for the help

Have you found why this happened by any chance ?

Cyberflow187, 2022-02-21:
[image]
@roleoroleo  : Ok i was able to make everything work without the Yi-Hack update (Thx for that help !)

Now i'm asking myself should i update the firmware ? The app gives me an error when i try to update.
If not needed il'l just install the Yi-Hack update

roleoroleo, 2022-02-21:
> Have you found why this happened by any chance ?

The home partition is almost full.
Probably a process try to write on it.

sphi-nx, 2022-02-21:
> Ok, forget the paypal address. LINK REMOVED Let me know when you download the file, I will remove this link.

Thank you so much @roleoroleo. This is really appreciated. 
Do you by any chance also have the y211ga?

Cyberflow187, 2022-02-21:
> > Have you found why this happened by any chance ?
> 
> The home partition is almost full. Probably a process try to write on it.
> 
> Something is wrong in your cam. The versions don't match. 4.2.0 and 4.5.0 belong to 2 different models.

The partition beeing full impacts the cameras not working with the app ?

roleoroleo, 2022-02-21:
It impacts the cameras not working at all.

Cyberflow187, 2022-02-22:
Hi @roleoroleo 

I understand what you mean by the version 4.2 VS 4.5

Why does the Yi app proposes me to upgrade to another version then ?

Cyberflow187, 2022-02-22:
And with that said; which version is the correct one ?

That explains why the app doesn't want to upgrade because there's a version conflict on the camera.

roleoroleo, 2022-02-22:
> Why does the Yi app proposes me to upgrade to another version then ?

Maybe you installed a wrong version?
Dump the bootloader with dd and I will check the problem:
`dd if=/dev/mtd/mtd0 of=/tmp/sd/mtd0.bin`

Cyberflow187, 2022-02-23:
@roleoroleo 10-4, for sure il'l do that in the next dayz i'm pretty loaded by work

Cyberflow187, 2022-12-26:
Hi there @roleoroleo  !

I guess i had a crazy year in 2022 !

I've got a few weeks off ahead and wanted to fix up this issue

I wanted to know; how do i perform a dump of the bootloader ?

I connect to the camera via ssh and write that line of command; right ?

`dd if=/dev/mtd/mtd0 of=/tmp/sd/mtd0.bin`

roleoroleo, 2022-12-26:
> I connect to the camera via ssh and write that line of command; right ?
> 
> `dd if=/dev/mtd/mtd0 of=/tmp/sd/mtd0.bin`

Yes.

Cyberflow187, 2023-01-09:
Hey hi @roleoroleo, i'm really rusted here.

I need to reinstall the yi-Hack-MStar in order to have the ssh service right ?

Because right now i've reinstalled the original system on it

roleoroleo, 2023-01-09:
Yes, you need to reinstall the hack for ssh to work.

closed 2024-05-08 by github-actions[bot]

Cyberflow187, 2024-05-26:
Hi @roleoroleo , sorry for the really long delay on this ! I will try to do this in the next weeks !

Cyberflow187, 2024-05-26:
@roleoroleo What would be the ssh password for the connection ? Cant figure out how to configure that

Cyberflow187, 2024-05-26:
@roleoroleo I tried using the windows ssh client and putty cant connect. always connection denied with the password i've configured on the webpage of the webcam

roleoroleo, 2024-05-27:
If you didn't change it, default password is empty.

Cyberflow187, 2024-06-03:
Hi @roleoroleo 
So :

Username : 
Password : empty

Or empty in like nothing or null ?

roleoroleo, 2024-06-07:
root without password

Cyberflow187, 2024-06-07:
Gotcha, i will give it a try

Cyberflow187, 2024-06-11:
That worked !

U : Root
P : 

But ill need to change the chmod on this i think : 
Tried the command and couldnt write the file : 

/home/yi-hack # dd if=/dev/mtd/mtd0 of=/tmp/sd/mtd0.bin
dd: can't open '/tmp/sd/mtd0.bin': Read-only file system

roleoroleo, 2024-06-29:
This is not normal.
When it happens, probably the sd card is corrupted and the file system is mounted read-only.
Remove it and try a file system check. It should work.