roleoroleo/yi-hack-MStar #410: h305r RTSP high res feed only 360p

olicooper, 2021-11-29:
I am not sure if this is a bug or a limitation of the firmware. My camera is the IeGeek IE80 model.

* The downloadable video files are 1080p, but the RTSP feed is only 360p
* The `snapshot.sh?res=high&watermark=yes` returns a blank response, but when `res=low` I get an image which is 360p
* The 'Snapshot' page `index.html?page=snapshot` is blank

roleoroleo, 2021-11-30:
>     * The downloadable video files are 1080p, but the RTSP feed is only 360p
>     * The RTSP low res feed `ch0_1.h264` doesn't seem to load

What do you mean?
is rtsp working?

To check this problem I need a dump of your stream.
Please run these commands:
```
killall wd_rtsp.sh
killall rRTSPServer
killall h264grabber_h
killall h264grabber_l
h264grabber_h -d -r high > /tmp/sd/high_stream.h264
```

And send me both the output log and the file /tmp/sd/high_stream.h264

olicooper, 2021-12-02:
I think I have captured what you need. Included in the zip is 10 seconds of video feed and the ssh output log. The camera is at a remote location so it may take a few days to get anything else you might need. [ie80-stream-20211130.zip](https://github.com/roleoroleo/yi-hack-MStar/files/7641429/ie80-stream-20211130.zip)
Below is a screenshot of the VLC media info for one of the `.mp4` files I downloaded from the SD card. `stream 0` is `1920x1080` but the `.h264` stream in the zip is only `640x360`. It is like `h264grabber` is capturing the wrong stream from the original feed.
[image]

roleoroleo, 2021-12-02:
Please, run this command:
`ls -laR /proc/mstar/OMX/`

olicooper, 2021-12-02:
The output from that command is too long to fit on the ssh history.. are there specific directories you would like me to query?

roleoroleo, 2021-12-03:
The driver is configured in a different manner.
Run these command, please:
```
cat /proc/mstar/OMX/VMFE0/ENCODER_INFO/OBUF_nAllocLen
cat /proc/mstar/OMX/VMFE0/ENCODER_INFO/OBUF_nFilledLen
cat /proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_nAllocLen
cat /proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_nFilledLen
cat /proc/mstar/OMX/VMFE2/ENCODER_INFO/OBUF_nAllocLen
cat /proc/mstar/OMX/VMFE2/ENCODER_INFO/OBUF_nFilledLen
```

olicooper, 2021-12-03:
Here you go:
```log
/home/yi-hack # cat /proc/mstar/OMX/VMFE0/ENCODER_INFO/OBUF_nAllocLen
153600
/home/yi-hack # cat /proc/mstar/OMX/VMFE0/ENCODER_INFO/OBUF_nFilledLen
1998
/home/yi-hack # cat /proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_nAllocLen
153600
/home/yi-hack # cat /proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_nFilledLen
1796
/home/yi-hack # cat /proc/mstar/OMX/VMFE2/ENCODER_INFO/OBUF_nAllocLen
153600
/home/yi-hack # cat /proc/mstar/OMX/VMFE2/ENCODER_INFO/OBUF_nFilledLen
71424
```

roleoroleo, 2021-12-03:
It seems that the high res stream is in the last buffer.
I will send yuo a custom version of h264grabber.

roleoroleo, 2021-12-03:
Here it is:
[h264grabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7648388/h264grabber.gz)

olicooper, 2021-12-03:
I wasn't sure how to update the system to include the file you sent, but I uploaded the unzipped files to a server and then ran these commands:
```log
wget http://mydomain/h264grabber

killall wd_rtsp.sh
killall rRTSPServer
killall h264grabber_h
killall h264grabber_l

cp /home/yi-hack/bin/h264grabber /home/yi-hack/h264grabber.original
cp /home/yi-hack/h264grabber /home/yi-hack/bin/
chmod 755 /home/yi-hack/bin/h264grabber

reboot -f
```

I can get a low resolution image so the camera is operational, but the video feed doesn't seem to be working and I can't get a high resolution image still. Its difficult to properly debug though as I am doing this remotely. Is there anything I amy have done wrong?

roleoroleo, 2021-12-03:
> I can get a low resolution image so the camera is operational, but the video feed doesn't seem to be working and I can't get a high resolution image still. Its difficult to properly debug though as I am doing this remotely. Is there anything I amy have done wrong?

The update procedure is ok.
What do you mean with "I can get a low resolution image"? Do you mean in the app?

If you run
`h264grabber_l -d -r low > /tmp/sd/low_stream_1.h264`
Does it work?

olicooper, 2021-12-03:
I can make a call to `http://CAMERA_IP:8080/cgi-bin/snapshot.sh?res=low&watermark=yes`

> If you run
> `h264grabber_l -d -r low > /tmp/sd/low_stream_1.h264`
> Does it work?

Unfortunately that doesn't work either:
```log
/home/yi-hack # h264grabber_l -d -r low > /tmp/sd/low_stream_1.h264
debug on
Resolution: 360
vaddr: 0xb45f9000 - paddr: 0x236c0000 - size: 153600
^C
```

olicooper, 2021-12-03:
What about `VVHE0` ?
```log
/home/yi-hack # cat /proc/mstar/OMX/VVHE0/ENCODER_INFO/OBUF_nFilledLen
3035
/home/yi-hack # cat /proc/mstar/OMX/VVHE0/ENCODER_INFO/OBUF_nAllocLen
1044480

roleoroleo, 2021-12-03:
> What about `VVHE0` ?
> 
> ```
> /home/yi-hack # cat /proc/mstar/OMX/VVHE0/ENCODER_INFO/OBUF_nFilledLen
> 3035
> /home/yi-hack # cat /proc/mstar/OMX/VVHE0/ENCODER_INFO/OBUF_nAllocLen
> 1044480
> ```

This path is new for me.
Try this:
[h264grabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7650302/h264grabber.gz)

But... strange behavior.
Based on this thread https://github.com/roleoroleo/yi-hack-MStar/issues/223 high res should work.

olicooper, 2021-12-03:
Same result. I called it `h264grabber2` and didn't bother putting it in the bin folder this time. I also tried the original one again and it didn't hang like the new versions do.

```log
/home/yi-hack # /home/yi-hack/h264grabber2 -d -r low > /tmp/sd/low_stream_1.h264
debug on
Resolution: 360
vaddr: 0xb45f9000 - paddr: 0x236c0000 - size: 153600
time: 3899139316 - len: 15989
frame is SPS
sync lost: 3899339267 - 3899139316
time: 3915136406 - len: 16057
frame is SPS
sync lost: 3915336391 - 3915136406
^C
/home/yi-hack # /home/yi-hack/h264grabber2 -d -r high > /tmp/sd/high_stream_1.h264
debug on
Resolution: 1080
vaddr: 0xb4af9000 - paddr: 0x23300000 - size: 1044480
^C
```

roleoroleo, 2021-12-03:
If your cam is h265 encoded, h264grabber doesn't work.
Could you send me a couple of mp4 file?
Both high and low res.

olicooper, 2021-12-06:
There is only one video quality saved to the SD card, so no high and low versions of the feed. It looks like all versions are encoded in to the same video file based on the VLC screenshot I took.
[image]

I have managed to get hold of some footage today, hopefully this is enough [20M00S60_1.zip](https://github.com/roleoroleo/yi-hack-MStar/files/7662663/20M00S60_1.zip). I think that `VVHE0` is the 1920x1088 h265 feed, and `VMFE0` is the 640x360 feed.

olicooper, 2021-12-06:
I don't know how the feeds are read, but I think live555 supports h265 if that helps.

This is probably a separate request but it feels related because of how h264grabber works and the modification required to support h265 feeds - would it be possible to have an option to read the feeds directly when 'cloud' is disabled so there is no delay? It could automatically switch too.

roleoroleo, 2021-12-11:
The mp4 is not clear enough.
Please try this binary, use it to dump the stream to a file as you already done.
```
/home/yi-hack/h264grabber -d -r high > /tmp/sd/high_stream.h265
```
And send me the output file.

roleoroleo, 2021-12-11:
live555 supports h265, I already used it in the Allwinner-v2 platform.

olicooper, 2021-12-14:
Looks like it worked this time and the file is bigger which is hopeful! [ie80-stream-20211214.zip](https://github.com/roleoroleo/yi-hack-MStar/files/7714893/ie80-stream-20211214.zip)

roleoroleo, 2021-12-15:
Ok, this should be the last version:
[h264grabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7717571/h264grabber.gz)

olicooper, 2021-12-15:
That works, great work! I ran it 2 times, the first time it kept loosing sync every few seconds, but the second time it worked without issue. I also tried the low stream which doesn't work (`h264grabber4 -d -r low`). [ie80-stream-20211215.zip](https://github.com/roleoroleo/yi-hack-MStar/files/7718998/ie80-stream-20211215.zip)

### 'low' stream log
```log
/home/yi-hack # /home/yi-hack/h264grabber4 -d -r low > /tmp/sd/low_stream1.h265
debug on
Resolution: 360
vaddr: 0xb451f000 - paddr: 0x236c0000 - size: 153600
time: 1123755024 - len: 30728
frame is SPS
sync lost: 1123954985 - 1123755024
time: 1135752820 - len: 30693
frame is SPS
sync lost: 1135952780 - 1135752820
^C
```

### first 'high' stream log
```log
...
copy buffer: len 801
copy again buffer: len 801
time: 1063516145 - len: 379
milliseconds: 1639567188074
copy buffer: len 379
copy again buffer: len 379
time: 1063566132 - len: 397
milliseconds: 1639567188120
copy buffer: len 397
copy again buffer: len 397
time: 1063616123 - len: 458
milliseconds: 1639567188177
copy buffer: len 458
copy again buffer: len 458
time: 1063666116 - len: 129576
milliseconds: 1639567188214
copy buffer: len 129576
copy again buffer: len 129576
frame is VPS
sync lost: 1063816085 - 1063666116
time: 1065665747 - len: 152689
frame is VPS
frame lost: 100004
time: 1065765751 - len: 8019
milliseconds: 1639567190322
copy buffer: len 8019
copy again buffer: len 8019
time: 1065815721 - len: 1737
milliseconds: 1639567190369
copy buffer: len 1737
copy again buffer: len 1737
time: 1065865719 - len: 804
milliseconds: 1639567190418
...
```

roleoroleo, 2021-12-15:
For low stream, run again this version:
https://github.com/roleoroleo/yi-hack-MStar/issues/410#issuecomment-991726626
with this syntax:
`/home/yi-hack/h264grabber -d -r low > /tmp/sd/low_stream.h264`
and send me the ouput.

olicooper, 2021-12-15:
The output is below. The file `low_stream1.h264` is 0 bytes.

```log
/home/yi-hack # /home/yi-hack/h264grabber3 -d -r low > /tmp/sd/low_stream1.h264
debug on
Resolution: 360
vaddr: 0xb451f000 - paddr: 0x236c0000 - size: 153600
sync lost: 3320345904 - 3320145938
sync lost: 3320545856 - 3320345904
sync lost: 3320745828 - 3320545856
sync lost: 3320945785 - 3320745828
sync lost: 3321145751 - 3320945785
sync lost: 3321345708 - 3321145751
sync lost: 3321545670 - 3321345708
sync lost: 3321745645 - 3321545670
sync lost: 3321945605 - 3321745645
sync lost: 3322145549 - 3321945605
sync lost: 3322345518 - 3322145549
sync lost: 3322545480 - 3322345518
sync lost: 3322745467 - 3322545480
sync lost: 3322945402 - 3322745467
sync lost: 3323145360 - 3322945402
sync lost: 3323345341 - 3323145360
sync lost: 3323545285 - 3323345341
sync lost: 3323745244 - 3323545285
sync lost: 3323945229 - 3323745244
sync lost: 3324145176 - 3323945229
sync lost: 3324345136 - 3324145176
sync lost: 3324545096 - 3324345136
sync lost: 3324745068 - 3324545096
sync lost: 3324945018 - 3324745068
sync lost: 3325144997 - 3324945018
sync lost: 3325344938 - 3325144997
sync lost: 3325544899 - 3325344938
sync lost: 3325744871 - 3325544899
sync lost: 3325944829 - 3325744871
sync lost: 3326144808 - 3325944829
sync lost: 3326344753 - 3326144808
sync lost: 3326544711 - 3326344753
sync lost: 3326744674 - 3326544711
sync lost: 3326944637 - 3326744674
sync lost: 3327144608 - 3326944637
sync lost: 3327344565 - 3327144608
sync lost: 3327544525 - 3327344565
sync lost: 3327744477 - 3327544525
sync lost: 3327944438 - 3327744477
sync lost: 3328144406 - 3327944438
sync lost: 3328344376 - 3328144406
^C
```

roleoroleo, 2021-12-15:
Here 3 different versions of the grabber.
The 1st reads from VMFE0 buffer.
The 2nd reads from VMFE1 buffer.
The 3rd reads from VMFE2 buffer.
Try them and check if there is an output.
[h264grabbers.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7719402/h264grabbers.tar.gz)

olicooper, 2021-12-15:
The 1st one looks like it works. The stream fails to open in VLC though, just like the stream from https://github.com/roleoroleo/yi-hack-MStar/issues/410#issuecomment-994038017 before you gave me this version of h264grabber https://github.com/roleoroleo/yi-hack-MStar/issues/410#issuecomment-994485541.

```log
/home/yi-hack # /home/yi-hack/h264grabber5_0 -d -r low > /tmp/sd/low_stream2.h264
debug on
Resolution: 360
vaddr: 0xb5a80000 - paddr: 0x235d0000 - size: 153600
time: 1472348206 - len: 1692
milliseconds: 1639576186829
copy buffer: len 1692
copy again buffer: len 1692
time: 1472398198 - len: 1794
milliseconds: 1639576186876
copy buffer: len 1794
copy again buffer: len 1794
time: 1472448189 - len: 1768
milliseconds: 1639576186922
copy buffer: len 1768
copy again buffer: len 1768
time: 1472498183 - len: 1683
milliseconds: 1639576186977
copy buffer: len 1683
copy again buffer: len 1683
time: 1472548172 - len: 1752
milliseconds: 1639576187025
copy buffer: len 1752
copy again buffer: len 1752
time: 1472598158 - len: 1769
milliseconds: 1639576187072
copy buffer: len 1769
copy again buffer: len 1769
time: 1472648160 - len: 1666
milliseconds: 1639576187133
copy buffer: len 1666
copy again buffer: len 1666
time: 1472698148 - len: 1696
milliseconds: 1639576187179
copy buffer: len 1696
copy again buffer: len 1696
time: 1472748147 - len: 1644
milliseconds: 1639576187226
copy buffer: len 1644
copy again buffer: len 1644
time: 1472798130 - len: 1654
milliseconds: 1639576187282
copy buffer: len 1654
copy again buffer: len 1654
time: 1472848115 - len: 1506
milliseconds: 1639576187329
copy buffer: len 1506
copy again buffer: len 1506
time: 1472898104 - len: 1500
milliseconds: 1639576187376
copy buffer: len 1500
copy again buffer: len 1500
time: 1472948099 - len: 1464
milliseconds: 1639576187422
copy buffer: len 1464
copy again buffer: len 1464
time: 1472998088 - len: 1636
milliseconds: 1639576187479
^C
/home/yi-hack # /home/yi-hack/h264grabber5_1 -d -r low > /tmp/sd/low_stream2.h264
debug on
Resolution: 360
vaddr: 0xb451f000 - paddr: 0x236c0000 - size: 153600
sync lost: 1481146586 - 1480946639
sync lost: 1481346548 - 1481146586
sync lost: 1481546503 - 1481346548
sync lost: 1481746471 - 1481546503
sync lost: 1481946455 - 1481746471
sync lost: 1482146416 - 1481946455
sync lost: 1482346365 - 1482146416
sync lost: 1482546317 - 1482346365
sync lost: 1482746290 - 1482546317
sync lost: 1482946251 - 1482746290
sync lost: 1483146211 - 1482946251
^C
/home/yi-hack # /home/yi-hack/h264grabber5_2 -d -r low > /tmp/sd/low_stream2.h264
debug on
Resolution: 360
vaddr: 0xb4245000 - paddr: 0x23870000 - size: 153600
^C
```

roleoroleo, 2021-12-15:
Again:
[h264grabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7720389/h264grabber.gz)

olicooper, 2021-12-15:
Low stream can be read and I can play the file in VLC. 
High stream can also be read, but does _not_ play in VLC.

### low stream
```log
/home/yi-hack # /home/yi-hack/h264grabber6 -d -r low > /tmp/sd/low_stream3.h264
debug on
Resolution: 360
vaddr: 0xb5a80000 - paddr: 0x235d0000 - size: 153600
time: 1204650001 - len: 35241
frame is SPS
time: 1204699980 - len: 2080
milliseconds: 1639580214142
copy buffer: len 2080
copy again buffer: len 2080
time: 1204749979 - len: 1976
milliseconds: 1639580214188
copy buffer: len 1976
copy again buffer: len 1976
time: 1204799970 - len: 1959
milliseconds: 1639580214245
copy buffer: len 1959
copy again buffer: len 1959
time: 1204849978 - len: 1995
milliseconds: 1639580214292
copy buffer: len 1995
copy again buffer: len 1995
time: 1204899952 - len: 1988
milliseconds: 1639580214348
copy buffer: len 1988
copy again buffer: len 1988
time: 1204949954 - len: 1985
milliseconds: 1639580214396
copy buffer: len 1985
copy again buffer: len 1985
time: 1204999931 - len: 1942
milliseconds: 1639580214442
...
```

### high stream
```log
/home/yi-hack # /home/yi-hack/h264grabber6 -d -r high > /tmp/sd/high_stream3.h264
debug on
Resolution: 1080
vaddr: 0xb49f9000 - paddr: 0x23300000 - size: 1044480
time: 1296633335 - len: 100854
frame is VPS
time: 1296683327 - len: 38452
milliseconds: 1639580306142
copy buffer: len 38452
copy again buffer: len 38452
time: 1296733320 - len: 1968
milliseconds: 1639580306189
copy buffer: len 1968
copy again buffer: len 1968
time: 1296783311 - len: 2011
milliseconds: 1639580306237
copy buffer: len 2011
copy again buffer: len 2011
time: 1296833295 - len: 2125
milliseconds: 1639580306294
copy buffer: len 2125
copy again buffer: len 2125
time: 1296883286 - len: 2248
milliseconds: 1639580306341
copy buffer: len 2248
copy again buffer: len 2248
time: 1296933280 - len: 2302
milliseconds: 1639580306395
...
```

roleoroleo, 2021-12-15:
It depends on the start of the file.
Try to use the new grabber with the normal flow: open your rtsp client and connect to the cam.

olicooper, 2021-12-15:
I don't have direct access to the rtsp feed unfortunately, only HLS via a remote client. I used the `set_config.sh` API to enable both high and low rtsp streams on the camera. When reading from the high stream the rtsp client gets a tcp `i/o timeout`, but the low stream works fine. The high stream uri is `rtsp://CAMERAIP/ch0_0.h264` - should it be `ch0_0.h265`?

roleoroleo, 2021-12-16:
The high stream can not work because rRTSPServer is not ready for h265. I have to change it.
But I wanted to check it the low stream was running.

olicooper, 2021-12-16:
Right okay, thank you for the update. Sorry for all this extra work! Will that make the high res snapshot endpoint `cgi-bin/snapshot.sh` work too?

roleoroleo, 2021-12-17:
Do you want to try this beta?
[h305r_0.4.4.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7735744/h305r_0.4.4.tar.gz)

olicooper, 2021-12-17:
Yeah I can do, but could it brick the device? I can upload it remotely to the SD card and reboot, but if it bricks the camera I will not be able to do anything about it.

olicooper, 2021-12-17:
I have updated the firmware, but I am getting `i/o timeout` on the low and high streams now. The low and high res `snapshot.sh` endpoint doesn't work either. Is there anything I can do to help test?

roleoroleo, 2021-12-17:
Bad news...
Try to test the whole chain.
1 - Run h264grabber with an output file and check if the file contains a stream.
2 - Check if /tmp/lowstream and /tmp/highstream exist and read the content.
3 - Find the process id of rRTSPServer and check the environment: `cat /proc/<pid>/environ`

olicooper, 2021-12-19:
1 - h254grabber works fine, both the high and low streams are playable.

2 - There is no `lowstream` or `highstream` in /tmp, but `/tmp/highres` content is `h265^@` and `/tmp/lowres` content is `h264^@`

3 - rRTSPServer environment:
```log
RRTSP_AUDIO=aacLD_LIBRARY_PATH=/lib:/home/lib:/home/ms:/home/app/locallib:/home/lib:/home/yi-hack/lib:/tmp/sd/yi-hack/libSHLVL=4OLDPWD=/home/appNR_LEVEL=0RRTSP_CODEC_LOW=h264PS1=[\w]## RRTSP_CODEC_HIGH=h265PATH=/sbin:/bin:/usr/sbin:/usr/bin:/home/base/wifi:/ext/bin:/home/base/tools:/home/app/localbin:/home/base/tools:/home/yi-hack/bin:/home/yi-hack/sbin:/tmp/sd/yi-hack/bin:/tmp/sd/yi-hack/sbinRRTSP_RES=bothRRTSP_PWD=<pass>OMX_LIB_PATH=/home/msRRTSP_USER=<user>ANDROID_PROPERTY_WORKSPACE=8,49152PWD=/homeTZ=GMT0BST,M3.5.0/1,M10.5.0RRTSP_PORT=554
```

roleoroleo, 2021-12-19:
Sorry, the files are /tmp/lowres and /tmp/highres.
All checks are ok. I will check again rRTSPServer

roleoroleo, 2021-12-19:
This should fix snapshots but I didn't find any problem in RTSP.
[imggrabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7741135/imggrabber.gz)

olicooper, 2021-12-21:
I've tested `imggrabber` and it almost works.. The low res image is fine, but the high res image is black and white, stretched vertically and repeats 4 times horizontally. See [this zipped image](https://github.com/roleoroleo/yi-hack-MStar/files/7753805/yi-imggrabber-test-high.zip). I will reboot and retest everything just-in-case.

roleoroleo, 2021-12-24:
Maybe the image is coded in a different manner.
Please make 2 or 3 copies of the buffer and share them with me.
```
testgrabber > /tmp/sd/buffer1.yuv
testgrabber > /tmp/sd/buffer2.yuv
testgrabber > /tmp/sd/buffer3.yuv
```
[testgrabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7774587/testgrabber.gz)

olicooper, 2021-12-25:
Okay, here you go [buffers.zip](https://github.com/roleoroleo/yi-hack-MStar/files/7775769/buffers.zip).

roleoroleo, 2022-01-08:
After a long reverse I probably found the solution.
Please, check if this image is correct:
[buffer1.yuv2.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7833457/buffer1.yuv2.gz)

Unzip it and load to this site:
https://rawpixels.net/

Configure the size as 1920x1088 and the format as YUV420SP (NV12).

Is it ok?
The colors are ok or inverted?
If it's not ok, try NV21.

olicooper, 2022-01-10:
I downloaded buffer1.yuv.gz and uploaded to rawpixels.net - I can confirm that NV12 is the correct format! Really great work! Sorry this is turning out to be more difficult than it initially seemed!

roleoroleo, 2022-01-10:
Here it is:
[imggrabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7840424/imggrabber.gz)
[snapshot.sh.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7840450/snapshot.sh.gz)

Call imggrabber with "-m h305r".

olicooper, 2022-01-10:
Sorry, can I try a version of imggrabber with NV21 instead of NV12 please? The image is okay now but colours seem wrong after testing.

roleoroleo, 2022-01-10:
Here it is:
[imggrabber_NV21.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7841168/imggrabber_NV21.gz)

If the colors are wrong, please send me a buffer of an image with many colors.

olicooper, 2022-01-10:
Perfect, the latest one works fine!
Still trying to work out what is wrong with the rtsp stream though.. I just don't seem to be able to connect. I've tried wifi, lan cable, with/without password for the rtsp stream etc.

roleoroleo, 2022-01-10:
Try this beta release.
[h305r_0.4.4.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7841569/h305r_0.4.4.tar.gz)

olicooper, 2022-01-10:
Snapshot works with the latest beta, but the colours are wrong again somehow? rtsp still not working.

olicooper, 2022-01-10:
restoring the old h264grabber from v4.3 allows the rtsp stream to work again

roleoroleo, 2022-01-11:
Snapshot ok, I know.
But about h264grabber, I don't know why doesn't work.
I need some debug.
Login with ssh and check these commands:
```
/home/yi-hack # ps | grep h264
  767 root      0:00 h264grabber_l -r low -f
  768 root      1h01 h264grabber_h -r high -f
 2429 root      0:00 grep h264
```
```
/home/yi-hack # ps | grep RTSP
  775 root     13:02 rRTSPServer
 1924 root      0:00 grep RTSP
```
```
/home/yi-hack # cat /proc/775/environ
```
Where 775 is the pid of rRTSPServer
```
/home/yi-hack # ls -la /tmp
```

olicooper, 2022-01-11:
```
/home/yi-hack # ps | grep h264
  851 root      0:00 h264grabber_h -m h305r -r high -f
 1268 root      0:00 grep h264grabber_l
 1273 root      0:00 grep h264

/home/yi-hack # ps | grep RTSP
  855 root      0:00 rRTSPServer
 1360 root      0:00 grep RTSP

/home/yi-hack # cat /proc/855/environ
RRTSP_AUDIO=LD_LIBRARY_PATH=/lib:/home/lib:/home/ms:/home/app/locallib:/home/lib:/home/yi-hack/lib:/tmp/sd/yi-hack/libSHLVL=3OLDPWD=/home/appNR_LEVEL=0PS1=[\w]## RRTSP_CODEC_HIGH=h265PATH=/sbin:/bin:/usr/sbin:/usr/bin:/home/base/wifi:/ext/bin:/home/base/tools:/home/app/localbin:/home/base/tools:/home/yi-hack/bin:/home/yi-hack/sbin:/tmp/sd/yi-hack/bin:/tmp/sd/yi-hack/sbinRRTSP_RES=highRRTSP_PWD=OMX_LIB_PATH=/home/msRRTSP_USER=ANDROID_PROPERTY_WORKSPACE=8,49152PWD=/homeTZ=RRTSP_PORT=554/home/yi-hack

# ls -la /tmp
total 56
drwxrwxrwt    4 root     0              520 Apr 17  2020 .
drwxr-xr-x   15 root     0                0 Jan  1  1970 ..
-rw-------    1 root     0                0 Jan 11 11:45 .count
-rw-------    1 root     0                9 Jan 11 11:45 MTK
lrwxrwxrwx    1 root     0               23 Jan 11 11:45 audio -> /home/app/audio_file/us
prw-------    1 root     0                0 Jan 11 11:45 audio_in_fifo
-rw-------    1 root     0                0 Jan 11 11:45 audio_in_fifo.requested
-rw-------    1 root     0                0 Jan 11 11:45 debug_alarm.txt
-rw-------    1 root     0                0 Jan 11 11:45 debug_oss.txt
-rw-------    1 root     0                0 Jan 11 11:45 debug_p2p.txt
-rw-------    1 root     0               28 Jan 11 11:45 dns0
-rw-------    1 root     0                2 Jan 11 11:45 done
-rw-------    1 root     0               12 Jan 11 11:45 gw0
prwx------    1 root     0                0 Apr 17  2020 h264_high_fifo
-rw-------    1 root     0                5 Apr 17  2020 highres
-rw-------    1 root     0               31 Apr 17  2020 hosts
-rw-------    1 root     0                0 Jan 11 11:45 httpd.conf
-rw-------    1 root     0             4177 Jan 11 11:46 log.txt
-rw-------    1 root     0                0 Jan 11 11:45 log_oss.txt
srwx------    1 root     0                0 Jan 11 11:45 logsock
-rw-------    1 root     0               14 Jan 11 11:45 mask
-rwx------    1 root     0             2648 Jan 11 11:45 mmap.info
-rw-------    1 root     0               65 Jan 11 11:45 resolv.conf
drwx------    8 root     0             8192 Jan  1  1970 sd
drwx------    4 root     0               80 Jan 11 11:45 var
-rwxrwxrwx    1 root     0              405 Jan 11 11:45 wpa_supplicant.conf
```

roleoroleo, 2022-01-11:
The output is ok.
So, if you overwrite h264grabber with the previous beta, it works.

What happens if you redirect the output?
Try for 2-3 seconds and post the file.
```
h264grabber_h -r high > /tmp/pippo.h264
```

olicooper, 2022-01-11:
Based on your comment I tested the latest h264grabber (from [this comment](https://github.com/roleoroleo/yi-hack-MStar/issues/410#issuecomment-1009294097)) with different settings for 10 seconds each, then I tried to play the videos in VLC afterwards. Here are all the video files: [streams_20220111_1.zip](https://github.com/roleoroleo/yi-hack-MStar/files/7847696/streams_20220111_1.zip).

## does work
`h264grabber_l -m h305r -r low > /tmp/sd/stream_20220111_L_1b.h264` - h264 640x360
`h264grabber_h -r high > /tmp/sd/stream_20220111_H_1a.h264` - h264 640x360

## does not work
`h264grabber_h -m h305r -r high > /tmp/sd/stream_20220111_H_1b.h264`
`h264grabber_l -r low > /tmp/sd/stream_20220111_L_1a.h264`

This is the non-debug output for `stream_20220111_L_1a.h264` (which is only 14kb in size). There was no output for the other streams.
```log
/home/yi-hack # h264grabber_l -r low > /tmp/sd/stream_20220111_L_1a.h264
sync lost: 3589870891 - 3589670918
^C
```

I also ran it with debug on for the streams that didn't work: 

```log
/home/yi-hack # h264grabber_l -d -r low > /tmp/sd/stream_20220111_L_1.h264
debug on
Resolution: 360
vaddr: 0xb45f9000 - paddr: 0x236c0000 - size: 153600
time: 3337717324 - len: 14316
frame is SPS
sync lost: 3337917285 - 3337717324
time: 3349715112 - len: 14447
frame is SPS
sync lost: 3349915058 - 3349715112
^C
```

```log
/home/yi-hack # h264grabber_h -m h305r -d -r high > /tmp/sd/stream_20220111_H_1.h264
debug on
Resolution: 1080
vaddr: 0xb4af9000 - paddr: 0x23300000 - size: 1044480
time: 3399355961 - len: 50528
frame is VPS
time: 3399405944 - len: 3195
milliseconds: 1641913490088
copy buffer: len 3195
copy again buffer: len 3195
time: 3399455937 - len: 4024
milliseconds: 1641913490143
copy buffer: len 4024
copy again buffer: len 4024
time: 3399505928 - len: 2301
milliseconds: 1641913490190
...

olicooper, 2022-01-11:
I tried a few older versions via **RTSP** on VLC media player to see what worked and I got the following results:

* [h264grabber3](https://github.com/roleoroleo/yi-hack-MStar/issues/410#issuecomment-991726626) (9688 bytes) - both streams do not work
* [h264grabber4](https://github.com/roleoroleo/yi-hack-MStar/issues/410#issuecomment-994485541) (12652 bytes) - both streams do not work
* [h264grabber5_0](https://github.com/roleoroleo/yi-hack-MStar/issues/410#issuecomment-994759485) (9688 bytes) - low stream works, high stream does not work
* [h264grabber6](https://github.com/roleoroleo/yi-hack-MStar/issues/410#issuecomment-994865506) (12788 bytes) - low stream works, high stream does not work

olicooper, 2022-01-13:
Having rtsp configured with just the low stream still doesn't work.
[image]

I can use telnet on Windows to connect to the camera on port 554, so it is working but the camera just won't stream anything. Can you suggest another Windows RTSP client I can try?

roleoroleo, 2022-01-13:
Ok, try this:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7865431/rRTSPServer.gz)

roleoroleo, 2022-01-14:
I checked again sources and it seems ok.
Let's restart from scratch.
We have to check:
1 - The command line of the grabber
2 - The output of the grabber
3 - The command line of the rtsp server
4 - Some file

1 - Run this command:
```
ps | grep h264grabber
```

2 - Run:
```
h264grabber_l -m h305r -r low -d > /tmp/sd/stream_low.h264
h264grabber_h -m h305r -r high -d > /tmp/sd/stream_high.h265
```
try to play the files with vlc and check the size.

3 - Check the environment of the daemon:
```
ps | grep RTSP
```
get the pid and
```
cat /proc/PID/environ
```

4 - Run:
```
cat /tmp/lowres
cat /tmp/highres
```

olicooper, 2022-01-14:
h264grabber is telling me `invalid option -- 'm'` so I want to make sure I am testing against the correct version.  I think I am testing against the original one from the latest beta you gave me but could you send me the version you'd like me to test so we both know what we're testing against?

[33 more replies]

closed 2022-01-27 by olicooper