roleoroleo/yi-hack-MStar #414: Question about RTSP Stream and ONVIF

idanny76, 2022-01-01:
I am using a TruVision NVR that has 8 channels. The performance is not as I would like. Having to run all 8 cameras on the low stream RSTP. If I have the High RSTP some cameras will not work, because of low resources. I am trying to figure this all out. Question: Will it affect the NVR if I am also using the YI Home app at the same time? By having the NVR pulling stream from the cameras at the same time as the app also sending stream to the Yi app make the cameras overload? 

I got the NVR free from my brother and just wanted to have the longer recording storage free. But the Yi app has good functionality that the NVR does not. The PTZ function on the NVR does not work. I would also like to have the NVR record on 1080p, but I guess thats too much to ask for. Any suggestion or help will greatly be appreciated, thank you.

roleoroleo, 2022-01-06:
> I am using a TruVision NVR that has 8 channels. The performance is not as I would like. Having to run all 8 cameras on the low stream RSTP. If I have the High RSTP some cameras will not work, [...]

Yes, but only while you watch the stream on the app.
If the cam is connected to the app but you don't watch the stream it should not be a problem.
AFAIK a single full hd stream from the cam works properly.
Check if the nvr opens more than one stream at the same time.

> I got the NVR free from my brother and just wanted to have the longer recording storage free. But the Yi app has good functionality that the NVR does not. The PTZ function on the NVR does not work. [...]

If you connectec the cam as ONVIF (not only as RTSP) the PTZ "should" work.
Should because ONVIF is not a good standard (IMHO).
I implemented the ptz support in the onvif daemon but other users noticed problems.

idanny76, 2022-01-06:
I really appreciate you taking the time to reply to something not related to the firmware. I have the cameras connected as ONVIF and RSTP and the PTZ is not working. I am not sure if you are familiar with the Truvision NVR. I can upload settings screenshots if you have time or are willing to take a look. If not, thats ok... I can keep playing around with the settings to see why the low resources and bad NVR performance. It might just be a low end crappy NVR that says can handle 8 cameras and not be true. It might be that these cameras do no have the hardware ability even with the modded firmware to perform very well with RTSP and ONVIF.

roleoroleo, 2022-01-06:
When you write ONVIF and RTSP you mean both?
So 2 connections?
You should connect the cam only with ONVIF; ONVIF will use RTSP to get the stream.

Please send me the screenshots of your settings.

I use a chinese NVR with my 4 cams and it works correctly, so the cam should not be a problem.

idanny76, 2022-01-08:
> When you write ONVIF and RTSP you mean both? So 2 connections? You should connect the cam only with ONVIF; ONVIF will use RTSP to get the stream.
> 
> Please send me the screenshots of your settings.
> 
> I use a chinese NVR with my 4 cams and it works correctly, so the cam should not be a problem.

Both ONVIF and RSTP are on, but the NVR is only pulling the ONVIF. If I turn the RSTP off, the ONVIF does not work. I have it set to provide both high and low quality stream, but the ONVIF on the NVR only recognizes the HIgh quality stream. The only way to get low quality is by setting the RSTP to low quality and not both. Here are the screenshots of the Yi-Hack settings and pictures of the NVR settings
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
.

idanny76, 2022-01-08:
PTZ for some reason does not work on the NVR. I have change the settings to low quality 7 of the 8 cameras... TL-Lo, Video Quality Lowest, Camera stream resolution low I think it was 640x360 and I can view more cameras, but not all 8. I was ok with at least 7 cameras of the 8, but them the next day the NVR would be frozen up. The NVR is a TruVision NVR 11 model is TVN1108S with a 2TB hard drive.

roleoroleo, 2022-01-09:
> Both ONVIF and RSTP are on, but the NVR is only pulling the ONVIF. If I turn the RSTP off, the ONVIF does not work.

This is correct.
NVR setup is ok. It should work.

> I have change the settings to low quality on all 8 cameras... TL-Hi, Video Quality Highest, Camera stream resolution low 640x360 and I can view all 8. Then less than an hour later the camera stream feed will be frozen or blacked out. The NVR is a TruVision NVR 11 model is TVN1108S with a 2TB hard drive. If I try 1080P on just one camera I start having issues. I would have loved to record all cameras at 1080p.

Are you able to detect if the stream is frozen cam side or nvr side?
If you reboot the cam doest it restart?

idanny76, 2022-01-09:
> 

The stream is frozen on some and blackout on other on the NVR side. If I go on the Yi Home app or Kami app..... all the feeds work fine. If I reboot the NVR it starts working good..... but it doesnt take long for the feed to start freezing on some and going black on others. According to the NVR specs.... it can handle up to 4/8/16 Full HD cameras in real-time (30fps). I am unsure if the NVR is defective or a poor quality NVR OR if the yi-hack firmware mod is at fault. Maybe I need to disable the recording to the SD Card and disable the feed to the YI Home servers?

Model : TVN-1108S-2T
Recording Capabilities

    Up to 4/8/16 built-in PoE ports, for 4/8/16 Ch NVR'S
    Plug-and-play camera installation
    Continuous, motion/alarm triggered or scheduled recording modes
    Event recording
    Fully compatible with TruVision Navigator software
    Audio recording with capable IP cameras
    Reverse playback
    Digital zoom on live and recorded video
    Watermarking for legal video authentication
    Free built-in DDNS service
    Mobile apps allow you to connect remotely via smartphone or tablet to keep an eye on facilities, people or valuables
    Start-up Wizards for quick and easy installation
    40/80/160 Mbps for incoming streams respectively for 4/8/16 Ch NVR'S
    Up to 6TB of storage  Familiar, user-friendly interface
    Sub-stream for low bandwidth live viewing
    2TB HDD included

Camera Support

    Up to 4/8/16 Full HD cameras in real-time (30fps), for 4/8/16 Ch NVR'S
    Supports 3rd-party cameras via ONVIF Profile S
    TruVision VGA through 4K (8MPx) (indoor and outdoor)
    TruVision 4/8/16 channel encoders
    PTZ and dome camera control via mouse, front panel and IR remote control

roleoroleo, 2022-01-09:
> 
> The stream is frozen on some and blackout on other on the NVR side. If I go on the Yi Home app or Kami app..... all the feeds work fine. If I reboot the NVR, it starts working good..... but it doesnt take long for the feed to start freezing on some and going black on others. According to the NVR specs.... it can handle up to 4/8/16 Full HD cameras in real-time (30fps). I am unsure if the NVR is defective, at fault or a poor quality NVR OR if the yi-hack firmware mod is at fault. Maybe I need to disable the recording to the SD Card and disable the feed to the YI Home servers?

The stream to the app is different from the rtsp stream. If the stream works on the app this doesn't mean that it also works on the rtsp client.
Try to disable recording to the sd and try to disable cloud services (just a try).
Swap space is enable or disable?
If you use vlc as a rtsp client, does the stream freeze?

roleoroleo, 2022-01-10:
Are you able to set tcp as a transport protocol?

roleoroleo, 2022-01-10:
Another try you can do.
Set the buffers to a higher value:
```
root@yi-hack-allwinner-v2:~# sysctl -w net.core.rmem_max=524288
net.core.rmem_max = 524288
root@yi-hack-allwinner-v2:~# sysctl -w net.core.rmem_default=524288
net.core.rmem_default = 524288
root@yi-hack-allwinner-v2:~# sysctl -w net.core.wmem_max=524288
net.core.wmem_max = 524288
root@yi-hack-allwinner-v2:~# sysctl -w net.core.wmem_default=524288
net.core.wmem_default = 524288
```

idanny76, 2022-01-10:
> Are you able to set tcp as a transport protocol?

Yes, I have it set to Auto. But I opened the RSTP stream that was blacked out on the NVR in the VLC and the stream was good. So its nothing to do with the RSTP stream feed coming from the camera. I notice that if I go to the edit portion of the camera in the NVR and type the password again and apply..... the feed comes back up.

idanny76, 2022-01-10:
> 

How do I do that? Thank you for taking the time to help me out. I haven't tried the disable the cloud and recording to the SD card, since the VLC stream from the camera was good.... basically letting me know it has to do with the NVR. If I dont have a authentication for the ONVIF..... anyone can access the feed if they have the IP and port number, correct?

idanny76, 2022-01-10:
> Another try you can do. Set the buffers to a higher value:
>
> ```
> root@yi-hack-allwinner-v2:~# sysctl -w net.core.rmem_max=524288
> net.core.rmem_max = 524288
> root@yi-hack-allwinner-v2:~# sysctl -w [...]

How or where do I set those buffer values? Thank you for taking the time to help me out. I haven't tried the disable the cloud and recording to the SD card, since the VLC stream from the camera was good.... basically letting me know it has to do with the NVR. If I dont have a authentication for the ONVIF..... anyone can access the feed if they have the IP and port number, correct?

idanny76, 2022-01-10:
So far its working..... Now the PTZ is not working. Do I need to do something in the Yi-Hack to allow the NVR to use PTZ?

roleoroleo, 2022-01-10:
> So far its working.....

What conf did you change?

> Now the PTZ is not working. Do I need to do something in the Yi-Hack to allow the NVR to use PTZ?

PTZ should work automagically when the cam is configured as ONVIF.
But other users noticed problems with PTZ.
To test if it works, try Onvif Device Manager on Windows: the "official" testing software.

About buffer values, if you still need it, you can set them entering in the cam with a ssh connection (like putty), logging in with root user, and writing the commands above.

idanny76, 2022-01-11:
PTZ is working now after reading and following the update you made on this link: https://github.com/roleoroleo/yi-hack-MStar/issues/415

You might want to look into the update process through the web interface on the yi-hack. I did the update from the 0.4.1 to the 0.4.3 through the web interface and the PTZ lines you edited on the link above for the for the ipc_cmd.sh path were still on /tmp and not /home. 

Do you think I should do a fresh install of the 0.4.3 in case other things might not have updated?

roleoroleo, 2022-01-11:
For this model, the update process overwrite the whole partition.
I don't know why this happens but I think that a fresh install doesn't fix the problem.
I will release a new version asap.

idanny76, 2022-01-11:
> For this model, the update process overwrite the whole partition. I don't know why this happens but I think that a fresh install doesn't fix the problem. I will release a new version asap.

Just fixed the outdoor camera by formatting the SD card, installing 0.4.2 first and then doing the web interface upgrade to the 0.4.3. Now the camera is working, but the PTZ file path did not update. Edited both files this time with the correct file path and PTZ is working. Online update is not updating the PTZ file path.

idanny76, 2022-01-11:
Got PTZ fixed, still need to test to see if NVR streams are freezing and blacking out. Before I get into the NVR testing.... I am trying to set the MQTT, everything seems to be pre-filled except the IP and Authentication. Do I use the same IP address for the Yi Camera? Client ID can be the camera name ex. backyard? For authentication use the same as the camera login?

I tried googling a tutorial online, but did not find one. Will keep looking. Thank you for all your help.... really appreciated.

idanny76, 2022-01-12:
Just this morning all the live feed on the NVR was frozen except on two cameras. One of the cameras is a 720p Yi Home cam 27US 1.8.7.0F_201809191400 with alienatedsec/yi-hack-v5 firmware and the other still providing live feed was on of the 1080p on your firmware. I notice comparing those two feeds that the 720p cam live feed had almost no delays, but the 1080p was lagging behind. 

I did read on your firmware description in regards to performance and you said "The performance of the cam is not so good (CPU, RAM, etc...). If you enable all the services you may have some problems. For example, enabling both rtsp streams is not recommended. Disable cloud is recommended to save resources." I have both feeds on, but the NVR is only using the low quality feed. Will having both on still consume camera resources even when the NVR is only accessing one?

The other thing I did notice this morning..... when the feeds are frozen or black, I went to the NVR and re-entered the camera password and the feed came back. I don't seem to be having that issue with the 720p Yi Home camera. This is the first time I notice that and I guessing it has to do with better camera hardware? Or Firmware?

roleoroleo, 2022-01-12:
> I did read on your firmware description in regards to performance and you said "The performance of the cam is not so good (CPU, RAM, etc...). If you enable all the services you may have some [...]

No, only if you consume them. And if you use only low res stream there must be no problem.

> The other thing I did notice this morning..... when the feeds are frozen or black, I went to the NVR and re-entered the camera password and the feed came back. I don't seem to be having that issue with the 720p Yi Home camera. This is the first time I notice that and I am guessing it has to do with better camera hardware? Or Firmware?

I think the problem is that the NVR doesn't restart the stream when there is a corruption.
For example when the network goes down or when a frame is corrupted.

> Not sure if by re-entering the password it reconnects to the camera by requesting the feed? I can also reboot the camera and feed comes back. If I reboot the NVR it also fixes the issue. Without restarting the NVR.... Rebooting the camera or re-entering the password restores feed.

What about the process when the screen is black?
Check if rRTSPServer is running when the cam is offline and if the connection is up:
```
ps | grep RTSP
netstat -anp | grep RTSP
```
Wait 10 seconds and try again the commands above to check if the process change the pid (dies and restarts).

Finally, kill rRTSPServer process, wait for the restart and check if it works.

idanny76, 2022-01-12:
> What about the process when the screen is black? Check if rRTSPServer is running when the cam is offline and if the connection is up:
>
> ```
> ps | grep RTSP
> netstat -anp | grep RTSP
> ```
>
> Wait 10 [...]

in Putty I ran the command then waited 10 seconds, ran the command again, but nothing happened. After I killed the rRTSPServer process and camera came back from blacked out to live feed. Since the cameras are on WiFi there may be like you say a corruption or the feed went out or stopped for a second and the NVR did not restart the connection. How can I fix that? and what about the delay in video feed? Would the delay be solely on the WiFi connection?

login as: root
root@192.168.0.3's password:

                   _     _           _
               _ _|_|___| |_ ___ ___| |_
              | | | |___|   | .'|  _| '_|
              |_  |_|   |_|_|__,|___|_,_|
              |___|
 -----------------------------------------------------
  yi-hack for MStar platform - v. 0.4.3
 -----------------------------------------------------

/home/yi-hack # ps | grep RTSP
 1082 root      7:02 rRTSPServer
 1148 root      0:00 grep RTSP
/home/yi-hack # netstat -anp | grep RTSP
netstat: showing only processes with your user ID
tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN      1082/rRTSPServer
tcp        0      0 192.168.0.3:554         192.168.0.45:43414      ESTABLISHED 1082/rRTSPServer
netstat: /proc/net/tcp6: No such file or directory
udp     1408      0 0.0.0.0:6970            0.0.0.0:*                           1082/rRTSPServer
udp        0      0 0.0.0.0:6971            0.0.0.0:*                           1082/rRTSPServer
udp     1408      0 0.0.0.0:6972            0.0.0.0:*                           1082/rRTSPServer
udp        0      0 0.0.0.0:6973            0.0.0.0:*                           1082/rRTSPServer
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
/home/yi-hack # ps | grep RTSP
 1082 root      7:04 rRTSPServer
 2130 root      0:00 grep RTSP
/home/yi-hack # netstat -anp | grep RTSP
tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN      1082/rRTSPServer
tcp        0   1040 192.168.0.3:554         192.168.0.45:43414      ESTABLISHED 1082/rRTSPServer
netstat: /proc/net/tcp6: No such file or directory
udp     1408      0 0.0.0.0:6970            0.0.0.0:*                           1082/rRTSPServer
udp        0      0 0.0.0.0:6971            0.0.0.0:*                           1082/rRTSPServer
udp     1408      0 0.0.0.0:6972            0.0.0.0:*                           1082/rRTSPServer
udp        0      0 0.0.0.0:6973            0.0.0.0:*                           1082/rRTSPServer
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
/home/yi-hack # login as: root
/home/yi-hack # ps | grep RTSP
 1082 root      7:02 rRTSPServer
 1148 root      0:00 grep RTSP
/home/yi-hack # netstat -anp | grep RTSP
netstat: showing only processes with your user ID
tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN      1082/rRTSPServer
tcp        0      0 192.168.0.3:554         192.168.0.45:43414      ESTABLISHED 1082/rRTSPServer
netstat: /proc/net/tcp6: No such file or directory
udp     1408      0 0.0.0.0:6970            0.0.0.0:*                           1082/rRTSPServer
udp        0      0 0.0.0.0:6971            0.0.0.0:*                           1082/rRTSPServer
udp     1408      0 0.0.0.0:6972            0.0.0.0:*                           1082/rRTSPServer
udp        0      0 0.0.0.0:6973            0.0.0.0:*                           1082/rRTSPServer
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
/home/yi-hack # ps | grep RTSP
 1082 root      7:04 rRTSPServer
 2130 root      0:00 grep RTSP
/home/yi-hack # netstat -anp | grep RTSP
tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN      1082/rRTSPServer
tcp        0   1040 192.168.0.3:554         192.168.0.45:43414      ESTABLISHED 1082/rRTSPServer
Password: netstat: /proc/net/tcp6: No such file or directory
udp     1408      0 0.0.0.0:6970            0.0.0.0:*                           1082/rRTSPServer
udp        0      0 0.0.0.0:6971            0.0.0.0:*                           1082/rRTSPServer
udp     1408      0 0.0.0.0:6972            0.0.0.0:*                           1082/rRTSPServer

udp        0      0 0.0.0.0:6973            0.0.0.0:*                           1082/rRTSPServer
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
/home/yi-hack # kill 1082
/home/yi-hack #

idanny76, 2022-01-12:
I have it on low quality. Is the video feed delay due to WiFi? The 720p yi home has no delays, but all the 1080p Dome do.

idanny76, 2022-01-12:
I tried to set the higher buffer values, but it did not work. I notice it said yi-hack-allwinner-v2 and not MStar. I dont know if that has anything to do with it.

Set the buffers to a higher value:

root@yi-hack-allwinner-v2:~# sysctl -w net.core.rmem_max=524288
net.core.rmem_max = 524288
root@yi-hack-allwinner-v2:~# sysctl -w net.core.rmem_default=524288
net.core.rmem_default = 524288
root@yi-hack-allwinner-v2:~# sysctl -w net.core.wmem_max=524288
net.core.wmem_max = 524288
root@yi-hack-allwinner-v2:~# sysctl -w net.core.wmem_default=524288
net.core.wmem_default = 524288

idanny76, 2022-01-12:
I was trying to install alienatedsec / yi-hack-v5 on one of the 1080p Dome for testing. The reason was, because the only camera that has no delay and doesn't loose connection from the NVR is the 720p Yi Home 27US  1.8.7.0F_201809191400 and its on the Yi-Hack-v5. I copied the files to the SD card, but after a reboot.... the firmware was the same. Do you know if its possible?

roleoroleo, 2022-01-13:
> in Putty I ran the command then waited 10 seconds, ran the command again, but nothing happened. After I killed the rRTSPServer process and camera came back from blacked out to live feed.
Since the cameras are on WiFi there may be like you say a corruption or the feed went out or stopped for a second and the NVR did not restart the connection. How can I fix that?

You are using udp transport.
You should try to use tcp transport instead of udp. It should work better.

> and what about the delay in video feed? Would the delay be solely on the WiFi connection?

Delay depends on many factors.
- server buffering (live555).
- client (for example using webrtc delay is very low).
- network.

> I have it on low quality. Is the video feed delay due to WiFi? The 720p yi home has no delays, but all the 1080p Dome do.

No, wifi is not a problem.

> I tried to set the higher buffer values, but it did not work. I notice it said yi-hack-allwinner-v2 and not MStar. I dont know if that has anything to do with it.

MStar, Allwinner and Allwinner-v2 have the same software.

> I was trying to install alienatedsec / yi-hack-v5 on one of the 1080p Dome for testing. The reason was, because the only camera that has no delay and doesn't loose connection from the NVR is the [...]

I don't know.

idanny76, 2022-01-14:
I have tried everything except disable the recording to SD card and the cloud. I need the cloud, because the app is useful compare to other apps for android. I wanted to use the NVR to get more recording time and have a monitor setup with the cameras in constant view. 

I did manage to get the RSTP link from two other cameras I have...... Hiseeu 5x Zoom that I bought from Amazon (https://www.amazon.com/gp/product/B08RHM6VF2). Those two cameras are also WiFi, but can be connected by Ethernet. I have those two cameras on WiFi and they work perfectly with the NVR. I get sound, PTZ, Zoom, Motion Detection without needing to do anything. I wish the Yi Dome camera with this firmware would work flawlessly. 

Right now the Yi Dome cameras are loosing NVR connection, frozen feed... which turns into a black feed when refreshed. Does this have to do with the live555?

roleoroleo, 2022-01-14:
> Right now the Yi Dome cameras are loosing NVR connection / frozen feed... which turns into a black feed when refreshed. Does this have to do with the live555?

I don't know.

Did you set tcp transport?

Meanwhile I pulled out of the closet my old MISECU NVR.
Installed it yesterday.
At the moment it's working, I will check tomorrow.

idanny76, 2022-01-14:
> > Right now the Yi Dome cameras are loosing NVR connection / frozen feed... which turns into a black feed when refreshed. Does this have to do with the live555?
> 
> I don't know.
> 
> Did you set tcp transport?
> 
> Meanwhile I pulled out of the closet my old MISECU DVR. Installed it yesterday. At the moment it's working, I will check tomorrow.

Yes, all the connections are set to TCP. Mine too works fine for a certain amount of time. After I come back hours later.... the live feed is frozen and when I refresh I get a black screen. Even when I restart the NVR, there is a slight delay on the Yi Dome cameras and it seems to get worst as time passes. The same thing happened to the 720p Yi Home 27US version.... So I don't think its a firmware issue anymore. I have 1GB Fiber Internet connection with WiFi 6 router and can eliminate the WiFi or internet as a problem here. 

The other two Hiseeu 5x Zoom cameras don't have a lag, freeze or any issues.... But that might be because of better hardware. I guess I will see if you experience the same issue or not. Will try to disable the cloud and sd card recording and see what happens.

roleoroleo, 2022-01-14:
I will send you a feedback in a few days.

idanny76, 2022-01-14:
> I will send you a feedback in a few days.

I think you might want to take a look at this post on another TheCrypt0 Yi-Hack. Just found it and will start looking at it now. Thank you for taking your valuable time to check this issue out. I know you do this on your free time and I greatly appreciated. 

[/TheCrypt0](https://github.com/TheCrypt0/yi-hack-v4/issues/89)

roleoroleo, 2022-01-14:
The approach used by TheCrypto is very different.
Because the cams are a previous model.
The rtsp server is also different.
It's the same problem but probably not related.

roleoroleo, 2022-01-16:
I found another rtsp library in C
https://github.com/PHZ76/RtspServer
and I compiled a test server.
Do you want to try it?
[rtsp_server_yi.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7877616/rtsp_server_yi.gz)

idanny76, 2022-01-16:
> I found another rtsp library in C https://github.com/PHZ76/RtspServer and I compiled a test server. Do you want to try it? [rtsp_server_yi.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7877616/rtsp_server_yi.gz) At the moment only 1 video stream, low or high.

Sure I will try it. What do I do to run the new RTSP Server?

roleoroleo, 2022-01-16:
Upload it via ftp to the sd card.
Login with ssh and, kill wd_rtsp.sh and rRTSPServer.
Run with your preferred options.

idanny76, 2022-01-16:
> Upload it via ftp to the sd card. Login with ssh and, kill wd_rtsp.sh and rRTSPServer. Run with your preferred options.

I am not proficient in SSH commands. When you say kill wd_rtsp.sh and rRTSPServer.... I have to first do ps | grep RTSP and ps | grep wd_rtsp to get the id or port number and then I can do the kill command ex. Kill 1010 and kill 1023? Can you explain the how to run to select the options?

roleoroleo, 2022-01-16:
Yes.
Or you can use killall with the name of the process.

idanny76, 2022-01-16:
> Yes. Or you can use killall with the name of the process.

how do I run the new rtsp server after placing it on the sd card?

roleoroleo, 2022-01-16:
If you copied it on /tmp/sd run
```
killall wd_rtsp.sh
killall rRTSPServer
cd /tmp/sd
./rtsp_server_yi -a alaw
```

Which model do you have?

idanny76, 2022-01-16:
Yi dome 1080p

roleoroleo, 2022-01-17:
When you kill wd_rtsp.sh and rRTSPServer the stream stops.
If you run rtsp_server_yi the stream should restart.
The cam comes back to the original situation (rRTSPServer) when you reboot.

But the error `Open /tmp/h264_high_fifo failed.` is not ok.
Try again.

Please note: aac at the moment is not supported.

roleoroleo, 2022-01-17:
A question: with rRTSPServer do you use AAC?

idanny76, 2022-01-17:
> A question: with rRTSPServer do you use AAC?

For RTSP audio I had PCM. I just disabled it.

roleoroleo, 2022-01-17:
Because high stream is disabled.
Try `./rtsp_server_yi -r low -a alaw`

idanny76, 2022-01-17:
> Because high stream is disabled. Try `./rtsp_server_yi -r low -a alaw`

Got this now and the camera did not reload live feed.

 -----------------------------------------------------
  yi-hack for MStar platform - v. 0.4.3
 -----------------------------------------------------

/home/yi-hack # killall wd_rtsp.sh
/home/yi-hack # killall rRTSPServer
/home/yi-hack # cd /tmp/sd
/tmp/sd # ./rtsp_server_yi -r low -a alaw
Resolution: 360p
Codec low: h264
Codec high: h264
Audio: 2
Port: 554
User:
Password: ********
Open /tmp/audio_fifo failed.

roleoroleo, 2022-01-17:
Sorry, the sound is also disabled.
`./rtsp_server_yi -r low -a no`

idanny76, 2022-01-17:
> Sorry, the sound is also disabled. `./rtsp_server_yi -r low -a no`

Finally is running now. Will let you know if the video stream work good or if it freezes/disconnect and goes black.

 -----------------------------------------------------
  yi-hack for MStar platform - v. 0.4.3
 -----------------------------------------------------

/home/yi-hack # killall wd_rtsp.sh
/home/yi-hack # killall rRTSPServer
/home/yi-hack # cd /tmp/sd
/tmp/sd # ./rtsp_server_yi -r low -a no
Resolution: 360p
Codec low: h264
Codec high: h264
Audio: 0
Port: 554
User:
Password: ********
Play URL: rtsp://127.0.0.1:554/ch0_1.h264
RTSP client connect, ip=192.168.0.45, port=49731

idanny76, 2022-01-17:
For some reason it loaded the feed again after the commands and then I got "No Video Link". Will try it again.

idanny76, 2022-01-17:
Lost video connection after maybe 2 minutes. Went into No video link screen. Do I need to keep Putty open on the SSH commands?

roleoroleo, 2022-01-18:
> Do I need to keep Putty open on the SSH commands?

We need a more stable setup.

- Create the folder yi-hack in the sd card.
- Create the file startup.sh in the yi-hack folder.
- Add this content to the file:

```
killall wd_rtsp.sh
killall rRTSPServer
/tmp/sd/rtsp_server_yi -r low -a no &
```

The cam will restart the new rtsp server at boot.
To restore the original rtsp server, remove or rename startup.sh

idanny76, 2022-01-18:
> We need a more stable setup.
>
>     * Create the folder yi-hack in the sd card.
>
>     * Create the file startup.sh in the yi-hack folder.
>
>     * Add this content to the file:
>
>
> ```
> killall wd_rtsp.sh [...]

I did what you said and it started like it normally starts ans then after 2 hours or so I went back and the feed was frozen. I can't say if after following your steps it did connect to the new RTSP server or the old one. All I can say is I created a yi-hack folder on the SD card root, then created a txt file on my PC copied the code to the txt file and renamed the txt to startup.sh. I then uploaded the file inside the yi-hack folder via FTP. I went to the Camera web interface and rebooted the cam. But there is no way for me to verify if it connected to the new RTSP server. If it did connect to the new RTSP server..... It is still freezing.

roleoroleo, 2022-01-19:
> I can't say if after following your steps it did connect to the new RTSP server or the old one.

You can check it entering with ssh and running `ps` command.
Be sure that the new created file has linux new line (LF), not windows CR-LF.
You can set the format with the editor (Notepad for example).

idanny76, 2022-01-19:
> > I can't say if after following your steps it did connect to the new RTSP server or the old one.
> 
> You can check it entering with ssh and running `ps` command. Be sure that the new created file has linux new line (LF), not windows CR-LF. You can set the format with the editor (Notepad for example).

Yes, its running on the new RTSP server. But the live feed still freezing/stops working in the NVR after sometime. As I was writting this the video feed came back on by itself after being blacked out. Maybe because I was sending the SSH command PS it woke the camera feed back up. 

It seems like the camera might be timing out the feed after sometime. I remember now that it happens on the YI Windows app. When I leave a camera feed live for a while, I come back and see that it disconnected. I have also seen a message on the Yi Windows app where it has asked me something in the line of..... If I still want to stay connected to the live feed. This could be basically that..... Yi firmware telling the camera to keep recording, but stop live feed.

 -----------------------------------------------------
  yi-hack for MStar platform - v. 0.4.3
 -----------------------------------------------------

/home/yi-hack # ps
PID   USER     TIME  COMMAND
    1 root      0:02 /init
    2 root      0:00 [kthreadd]
    3 root      2:13 [ksoftirqd/0]
    5 root      0:00 [kworker/0:0H]
    7 root      4:55 [rcu_preempt]
    8 root      0:00 [rcu_sched]
    9 root      0:00 [rcu_bh]
   10 root      0:01 [watchdog/0]
   11 root      0:00 [khelper]
   12 root      0:00 [writeback]
   13 root      0:00 [crypto]
   14 root      0:00 [bioset]
   15 root      0:00 [kblockd]
   16 root      0:01 [kworker/0:1]
   17 root      0:32 [kswapd0]
   18 root      0:00 [fsnotify_mark]
   31 root      0:00 [SCLDAZA_THREAD]
   32 root      0:02 [VIPDazaTask]
   36 root      0:00 [deferwq]
   39 root      0:00 /bin/ueventd
   40 root      0:01 [jffs2_gcd_mtd3]
   41 root      0:00 [jffs2_gcd_mtd2]
   79 root      1:36 [mmcqd/0]
   90 root      0:00 [cryptodev_queue]
  101 root      0:00 [spi0]
  133 root      0:00 [cfg80211]
  138 root      0:10 ./log_server
  139 root      3:10 ./dispatch
  623 root      0:01 [kworker/u2:2]
  640 root      8h00 ./rmm
  693 root     23:27 ./mp4record
  696 root      0:08 ./oss
  697 root      0:00 ./watch_process
  706 root      0:00 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
  732 root      0:00 pure-ftpd (SERVER)
  753 root      0:00 dropbear -R
  758 root      0:02 ipc_multiplexer
  857 root      0:00 ntpd -p pool.ntp.org
  868 root      0:03 [kworker/u2:1]
  915 root      1h03 h264grabber_l -r low -f
  943 root     20:35 rRTSPServer
  954 root      0:31 {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
 1045 root      0:04 onvif_srvd --conf_file /tmp/onvif_srvd.conf
 1052 root      0:05 ./arp_test
 1068 root      0:18 wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --type t
 1089 root      0:00 /usr/sbin/crond -c /var/spool/cron/crontabs/
 1447 root      0:00 /tmp/sd/rtsp_server_yi -r low -a no
 2921 root      0:00 dropbear -R
 3120 root      0:00 sleep 10
 3145 root      0:00 -sh
 3166 root      0:00 ps
 3174 root      0:00 [kworker/0:2]
 3649 root      0:01 [RTW_CMD_THREAD]
 3667 root      0:00 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf
 3692 root      0:00 udhcpc -i wlan0 -b -s /home/app/script/default.script -x h
 3830 root      0:00 ./cloud
 3836 root      0:03 ./p2p_tnp

idanny76, 2022-01-19:
>

roleoroleo, 2022-01-19:
> Yes, its running on the new RTSP server. But the live feed still freezing/stops working in the NVR after sometime. As I was writting this the video feed came back on by itself after being blacked out. Maybe because I was sending the SSH command PS it woke the camera feed back up.

I think it's not related to your command.

> It seems like the camera might be timing out the feed after sometime. I remember now that it happens on the YI Windows app. When I leave a camera feed live for a while, I come back and see that it [...]

The stream though rtsp server is different than yi app and It shouldn't time out.
The frames are grabbed directly from kernel memory and rtsp daemon never stops.

> On the SSH after it shows the connection to the RSTP server one line down... it says "sleep 10".

Yes. The watchdog checks the daemon every 10 seconds.

idanny76, 2022-01-19:
> If this rtsp server is also not working I think there are two possibilities:
> 
>     * the problem is some bad frame
> 
>     * the problem is outside the camera
> 
> 
> To check the 1st step you could try to capture a stream on the sd card, extract it from the cam and try to play with a client software (like vlc).
> 
>     * kill wd_rtsp.sh and rRTSPServer
> 
>     * capture the stream `dd if=/tmp/h264_low_fifo of=/tmp/sd/test.h264`
> 
>     * stop capturing when you want
> 
>     * play the file test.h264 with a PC
> 
> 
> You should see some artifact for a couple of seconds and then it should work properly.

Its been working fine so far now since I changed the file from CR-LF to LF. Will check back in a couple of hours to see if the feed is still live. If not, then I will try the steps to capture a stream and play on VLC.

idanny76, 2022-01-19:
After editing the startup.sh to LF and running the camera.... Two hours now still working good. Will check again tonight. Hopefully it was just startup.sh being on CR-LF that was causing the problem.

idanny76, 2022-01-19:
So far so good. Seems to be working. I am 99.9% that its working, because the connection would not have lasted 6 hours without freezing/loosing connection. Can I try this setup on my other Yi Dome cameras?

roleoroleo, 2022-01-20:
> So far so good. Seems to be working. I am 99.9% that its working, because the connection would have not lasted 6 hours without freezing/loosing connection. Can I try this setup on my other Yi Dome cameras?

Which model?

idanny76, 2022-01-20:
> > So far so good. Seems to be working. I am 99.9% that its working, because the connection would have not lasted 6 hours without freezing/loosing connection. Can I try this setup on my other Yi Dome cameras?
> 
> Which model/suffix?

The camera is still working after 17 hours. Came and checked this morning and the live feed was still working.

Yi Dome 1080p
Base Version | 4.6.0.0A_201908271549
Model Suffix | h201c
Serial Number | BFUS

[92 more replies]

closed 2024-05-07 by github-actions[bot]