roleoroleo/yi-hack-MStar #415: PTZ don't work -- yi 1080p dome

Hataru80, 2022-01-06:
Hello, 

i have:
 
Firmware Version	0.4.3
Base Version	4.6.0.0A_201908271549
Model Suffix	h201c
Serial Number	BFUSY1E7S3D3IX190902

PTZ work on YI-HACK web page but don't work using "ONVIF Device Manager" or "Home Assistant" whit WebRTC Camera, ONVIF_cam and yi_hack_m_c00620.

I don't see error and don't understand why.

Can you help me please?

Thanks you very much,

roleoroleo, 2022-01-06:
Login with ssh and send me the configuration file:
```
cat /tmp/onvif_srvd.conf
```

Hataru80, 2022-01-06:
```
pid_file=/var/run/onvif_srvd.pid
model=Yi Hack
manufacturer=Yi
firmware_ver=0.4.3
hardware_id=BFUS
serial_num=BFUSY1E7S3D3IX190902
ifs=wlan0
port=80
scope=onvif://www.onvif.org/Profile/S

#Profile 0
name=Profile_0
width=1920
height=1080
url=rtsp://%s/ch0_0.h264
snapurl=http://%s:8080/cgi-bin/snapshot.sh?res=high&watermark=yes
type=H264

user=xxx
password=xxx

#PTZ
ptz=1
move_left=/tmp/sd/yi-hack/bin/ipc_cmd -m left
move_right=/tmp/sd/yi-hack/bin/ipc_cmd -m right
move_up=/tmp/sd/yi-hack/bin/ipc_cmd -m up
move_down=/tmp/sd/yi-hack/bin/ipc_cmd -m down
move_stop=/tmp/sd/yi-hack/bin/ipc_cmd -m stop
move_preset=/tmp/sd/yi-hack/bin/ipc_cmd -p %t

```

Thanks you!

roleoroleo, 2022-01-06:
I fixed a bug here:
https://github.com/roleoroleo/yi-hack-MStar/commit/b0e69281889746ba3b01607daf37e6e24481455f#diff-6840841c37f9cb08680edb6c2bc19354d5e2ae9651217ee1033a868232b30b79

Hataru80, 2022-01-06:
TOP!
I have modify the file and now the moviment work. 
I don't remember if zoom must work because don't work

roleoroleo, 2022-01-06:
Optical zoom is not present and digital zoom is not supported.

idanny76, 2022-01-11:
I also have the issue with the PTZ. I did have the 0.4.3 firmware, but the location for was still in /tmp/sd/yi-hack/bin/ipc_cmd. I did not install 0.4.3 originally.... I updated from previous version either 0.4.0 or 0.3.9 within the yi-hack web interface. I guess the update did not modified the location address. I tried to change it manually through FTP to /home/yi-hack/bin/ipc_cmd, but as soon as I reboot the camera.... the file goes back to the old location.

idanny76, 2022-01-11:
> I fixed a bug here: [b0e6928](https://github.com/roleoroleo/yi-hack-MStar/commit/b0e69281889746ba3b01607daf37e6e24481455f)
> 
> But in 0.4.3 it should be already ok.

I also have the issue with the PTZ. I did have the 0.4.3 firmware, but the location for was still in /tmp/sd/yi-hack/bin/ipc_cmd. I did not install 0.4.3 originally.... I updated from previous version either 0.4.0 or 0.3.9 within the yi-hack web interface. I guess the update did not modified the location address. I tried to change it manually through FTP to /home/yi-hack/bin/ipc_cmd, but as soon as I reboot the camera.... the file goes back to the old location.

Edit:
I messed up. Tried editing the src/static/static/home/yi-hack/script/system.sh and then rebooted. Now the camera has no connection. Its outside now and late. I will remove memory card and reinstall the 0.4.3 firmware again. Is there a way to just update those two files for the rest of the cameras by FTP? All the other cameras are on the 0.4.3 but with the wrong file location.

roleoroleo, 2022-01-11:
The camera must work even if you remove the sd card.
You can't access it with ssh or http but must work.

When the cam is bricked you can restore it reinstalling the hack but you can't install the same version because the bootloader checks the hash. So you have to install a previous version and then update to the last one.

idanny76, 2022-01-11:
> The camera must work even if you remove the sd card. You can't access it with ssh or http but must work.
>
> When the cam is bricked you can restore it reinstalling the hack but you can't install the [...]

I think then it must have bricked, because it's not online on the yi home app. I'll remove the memory card and do what you mentioned. Install previous version and then the latest. I think if I can edit that second file it might work. But if you release a new version I can just update on all cameras.

closed 2024-05-07 by github-actions[bot]