roleoroleo/yi-hack-MStar #419: Revert back to original firmware due to blurry image

aqwserf, 2022-01-28:
Hi @roleoroleo 

First of all, thanks for your awesome firmware !

I bought two Yi 1080p Home Camera (both begin with 6FUS). With the first one, I was able to pair it with the Yi App on Android then hack it with your firmware (y203c) and it works great. With the other one, even after a reset, I never heard the "Waiting connection", so I gave a try with your hack + wifi setup and it worked. 

The issue is that the image is blurry (but not from near, only from far). So according to [this](https://help.yitechnology.com/hc/en-us/articles/231352688-Why-are-the-videos-blurry-or-not-clear-) link, I should be able to fix the issue with app. 
I was able to download the "original" firmware from [there](https://yihome-firmware-us.oss-us-west-1.aliyuncs.com/yifirmware/smarthomecam/familymonitor-y20/home_y20m) but the camera flashes orange continuously, so I guess it didn't work :)
I was however able to find the real original firmware [here](https://github.com/shadow-1/yi-hack-v3#going-back-to-stock-firmware) but I guess this is the wrong one (Hi3518e chipset instead of MStar). Am I correct? 
According to this [post](https://github.com/roleoroleo/yi-hack-MStar/issues/15#issuecomment-547849512), feiticeir0 provided original firmwares but there's no `rootfs_y20`. Am I misunderstanding?

If necessary, could you provide me the original firmware (`home_y20` and `rootfs_y20` version 4.5.0?) so I can pair it with the application? Or maybe there's another procedure to pair the camera with the application AND the hack? Or maybe another advice to change the focus manually via ssh/webui?

Thanks!

roleoroleo, 2022-01-28:
> I was able to download the "original" firmware from [there](https://yihome-firmware-us.oss-us-west-1.aliyuncs.com/yifirmware/smarthomecam/familymonitor-y20/home_y20m) but the camera flashes orange continuously, so I guess it didn't work :)

This firmware is only for update, it's not a complete firmware.

> I was however able to find the real original firmware [here](https://github.com/shadow-1/yi-hack-v3#going-back-to-stock-firmware) but I guess this is the wrong one (Hi3518e chipset instead of MStar). Am I correct?

Yes, it's not for this platform.

> According to this [post](https://github.com/roleoroleo/yi-hack-MStar/issues/15#issuecomment-547849512), feiticeir0 provided original firmwares but there's no `rootfs_y20`. Am I misunderstanding?

Again, only partial firmware for update.

> If necessary, could you provide me the original firmware (`home_y20` and `rootfs_y20` version 4.5.0?) so I can pair it with the application? Or maybe there's another procedure to pair the camera with the application AND the hack? Or maybe another advice to change the focus manually via ssh/webui?

If you have a MStar cam with 4.5.0, you need home_y203c and sys_y203c
Not *y20

Send me a pm to the paypal email address.

aqwserf, 2022-01-28:
Thanks for your prompt response @roleoroleo. Email sent to your paypal email account. 
And thanks for the explanation!

aqwserf, 2022-01-31:
Hey @roleoroleo. Finally got it working by opening the camera and turned the lens. Now the image is clear (but purpled). Integration with HA works great with both cameras (detection, stream...) :champagne: 

Thanks again for your support and your great firmware!

closed 2022-01-31 by aqwserf

bepece1, 2022-03-18:
> Hey @roleoroleo. Finally got it working by opening the camera and turned the lens. Now the image is clear (but purpled). Integration with HA works great with both cameras (detection, stream...) 🍾
> 
> Thanks again for your support and your great firmware!

Could you explain how turned the lens???
I have same problem

aqwserf, 2022-03-23:
Hi @bepece1 !
Sorry for the delay. I basically just opened the camera and turned the optic (the round thing), like in this picture (EU - Camera Top View): https://fccid.io/2AFIB-YHS2116/Internal-Photos/Int-Photos-2894309
I was using the live RTSP stream to adjust accordingly