roleoroleo/yi-hack-MStar #422: Request Extend support ieGeek IE82

fabiex75, 2022-02-04:
Hi, can you add support for IeGeek IE82, please ? I can't find any information about chipset or other.
Thanks in advance

fabiex75, 2022-02-04:
..if I may be useful for additional information, let me know.

roleoroleo, 2022-02-05:
It depends on the platform.
Please, start opening the cam and taking some snapshot of the board.

fabiex75, 2022-02-05:
Thanks.I post some screenshots. Let me know if i can help in any way.[image]
[image]

roleoroleo, 2022-02-06:
PleasePlease also the other side.

fabiex75, 2022-02-07:
[image]

roleoroleo, 2022-02-10:
I can't see the cpu..
Is there another board?

idanny76, 2022-02-14:
> No...this is the only One.Maybe Photo is not clear or under wifi module ?

Could be inside the upper part where the camera is encased in.

fabiex75, 2022-02-17:
I don't know how open it more
How can I open it without broken ?

[image](https://user-images.githubusercontent.com/44639348/154422530-df1a77f6-8f0e-47f2-947c-f63a6556ac40.jpeg)
[image]

idanny76, 2022-02-17:
> I don't know how open it more How can I open it without broken ?
> 
> [image]
> 
> [image]

I think it can be separate in half. If there are no visible screws holding those two half together, then they should snap off. You may need to use some force to snap it. There is always a chance it might break, but I think it should come apart..... there is definitely a line showing a separation.  
[image]

fabiex75, 2022-03-29:
Sorry guys but at first attemp I ear little crack and I'm stopped. So, I try to flash with ie80 experimental and works. Now I can connect to my ie82 in http on port 8080. Can I have to do some tests for ensure that all is ok ? Thanx as always

roleoroleo, 2023-04-30:
Do you know the firmware version?
I mean the original one.

closed 2024-05-07 by github-actions[bot]