roleoroleo/yi-hack-MStar #424: Online FW upgrade - not work - y203c

rysuq, 2022-02-13:
Hello,
first of all thanks for your project.
I have a problem to download lastest version of firmware.
After try to download firmware I got message
**Unable to download firmware file.**
My camera information:
Hostname | yi-home-1080p
Firmware Version | 0.4.3
Base Version | 4.5.0.0C_201910080934
Model Suffix | y203c

roleoroleo, 2022-02-13:
The message refers to wget command.
But the link is ok. Check your connection or your firewall.

rysuq, 2022-02-13:
[image]
Hmm, I use Adguard to monitoring DNS.
Also I try to disable my Adblocking dns.

robho, 2022-02-15:
I had this problem also and for me the problem was that the SD card was mounted read-only. wget then failed to download and write the update to the SD card -> update failed.

I'm not sure why yi-hack failed to repair the file system (it looks like there's some code to do it before it's mounted), but to manually repair the file system I did:
* umount /tmp/sd
* fsck.fat -a /dev/block/mmcblk0p1

rysuq, 2022-02-19:
This solution not working for me

```
/home/yi-hack # umount /tmp/sd
/home/yi-hack # fsck.fat -a /dev/block/mmcblk0p1
CP437: Invalid argument
fsck.fat 3.0.26 (2014-03-07)
step 0 1645282219
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
step 1 1645282219
step 2 1645282219
Killed

```

roleoroleo, 2022-02-19:
Try to remove the card and fix it with your PC.

rysuq, 2022-02-26:
After issue in sd card now everything working.

closed 2022-02-26 by rysuq