roleoroleo/yi-hack-MStar #429: Yi 1080p home not storing WiFi configuration

edofullin, 2022-02-26:
The title says it all, every time the cam reboots it asks for the WiFi credentials and does not store them.

Was using 0.3.7, updated to 0.4.4 but the issue persists. Is it a known issue? 
Can I store the credentials on the SD card by chance?

Let me know if you need more info, thanks in advance!

edofullin, 2022-02-26:
Opened the issue too early.

Downloading the configure_wifi.sh script, manually removing the check for the current SSID and password and manually running it from ssh _seems_ to have solved the issue. 

I don't know if this is just a workarounf though

roleoroleo, 2022-02-26:
Sometimes it happens, but the solution you used is correct.

andrzejSlomiany, 2022-03-04:
Hello. I have the same problem every time the camera turns off, I need to configure it. Please what should be deleted in the file configure_wifi.sh ?

roleoroleo, 2022-03-05:
I think I means:
```
if [ "$SSID" == "$CURRENT_SSID" ] && [ "$KEY" == "$CURRENT_KEY" ]; then
    echo "ssid and key already configured"
    exit
fi
```

edofullin, 2022-03-05:
> I think he means:
> 
> ```
> if [ "$SSID" == "$CURRENT_SSID" ] && [ "$KEY" == "$CURRENT_KEY" ]; then
>     echo "ssid and key already configured"
>     exit
> fi
> ```

Correct, I followed [this](https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials-(deprecated-from-0.4.0)) guide first and connected the cam manually with the qr code, sshed into it, navigated to the sdcard path, removed that check and ran the script manually.

It has been working for a few days now and it rebooted several times so I think this is solved.

closed 2022-03-13 by edofullin