roleoroleo/yi-hack-MStar #431: Can't save configuration: "Error while saving" message after upgrading to v0.4.5

Hall9k, 2022-03-17:
I've just upgraded to v0.4.5 via "Online FW upgrade"

roleoroleo, 2022-03-18:
You are right.
I don't know why, but jq is broken.
I will release a new version asap.
In the meantime, use this one:
[jq.gz](https://github.com/roleoroleo/yi-hack-MStar/files/8302878/jq.gz)

Hall9k, 2022-03-18:
Thank you very much roleoroleo. 
Sorry but I don't understand what I have to do with that file,  jq.gz
Should I copy it in a sd card?
Thanks for all your work

Madaraxa, 2022-03-18:
> Thank you very much roleoroleo. Sorry but I don't understand what I have to do with that file, jq.gz Should I copy it in a sd card? Thanks for all your work

Now a New Version ist out (0.4.6) which should have fixed the issue

closed 2024-05-07 by github-actions[bot]