roleoroleo/yi-hack-MStar #433: /bin/busybox is missing poweroff and halt support

meskilla, 2022-04-01:
which leads to a poweroff: applet not found message.

Did you strip it for filesize reasons? Reboot works and is listed in busybox 4.6.0 functions.

What is the safest way to shutdown the cam with a mmc vfat partition still mounted and possibly recording? I can't just cut the power cord, as this will lead to vfat corruption. Is there a stop script instead of shutdown jobs that could kill jobs and safely unmount /tmp/sd before I cut the power ?

roleoroleo, 2022-04-01:
It's stripped by yi.

At the moment there is no way.
You should send reboot command, wait 5 seconds and remove power cord.

meskilla, 2022-04-01:
is there a shell script to safely unmount /tmp/sd first?

roleoroleo, 2022-04-03:
No, but I can try to add it.

Lin-Admin, 2022-04-06:
> 
Thanks. This would be very helpful!

roleoroleo, 2022-04-07:
For MStar I added it. Next release...

closed 2024-05-07 by github-actions[bot]