roleoroleo/yi-hack-MStar #435: FTP to HASSIO

dherkes, 2022-04-08:
I had been using the yi-hack-MStar successfully to upload clips taken by my camera to my HAS installation.   All was well until I upgraded to 4.6, then the FTP just stopped working.  Nothing changed on the HAS end.  The user and password are the same.

roleoroleo, 2022-04-13:
I changed the way it starts.
From daemon mode to crontab.
It should start every 10 minutes with 40 seconds offset.

Check the following steps:
1. cat /var/spool/cron/crontabs/root
2. ps | grep crond

dherkes, 2022-04-15:
I also found I cannot ssh into the camera any longer.  Since I turned telnet off, all I have is the web UI.  Shall I just reinstall 4.5 and go from there?  What settings does the ftp need on the web UI in 4.6
Thank you.

roleoroleo, 2022-04-17:
Apart from the wait it starts, the script is not changed from 4.5 to 4.6.
You need to set the following fields:
[image]

dherkes, 2022-04-17:
Thank you very much.  I'm going to do a reinstallation to get ssh back.  Works great.

dherkes, 2022-04-18:
I updated to 4.7.  Everything works like a charm.  Thank you.

closed 2022-04-24 by dherkes

olivluca, 2022-08-09:
> I changed the way it starts. From daemon mode to crontab. It should start every 10 minutes with 40 seconds offset.
> 
> Check the following steps:
> 
>     1. cat /var/spool/cron/crontabs/root
> 
>     2. ps | grep crond

May I ask why did you change it to cron instead of starting it?

roleoroleo, 2022-08-17:
Because crond is already running.
I think it is more efficient.

olivluca, 2022-08-17:
Yes, but it will delay the sending of the files to the server up to a minute (or up to an hour in some old versions, like the one I have before fixing the crontab).
The less time it takes to upload the file, the less time the eventual thieve has to see the camera and yank it.
(in fact I think that uploading after the fact is already too late, maybe there is a better way).

roleoroleo, 2022-08-18:
If you have this kind of need, probably, it would probably be better to save the stream to a dvr.

olivluca, 2022-08-18:
Currently the camera is connected via cellular data so I cannot stream continuosly (a local dvr is not an option since it could be stolen at the same time).
I already setup mqtt to send a picture on motion start, maybe I could implement an option to send several pictures every n seconds during a motion event.

roleoroleo, 2022-08-18:
Use startup.sh to run your custom progrmas.
You could use ipc_notify command to do "something" when a motion occurs.
Keep in mind that high res snapshot is cpu expensive, so slow...