roleoroleo/yi-hack-MStar #436: Incorrect video filename (doesn't match time)

phoenix1887, 2022-04-12:
Hi, I've been using this software (and other versions allwinner etc) for over a year across many cams and have continually been updating every once in awhile when new versions come out. However I just updated one camera that has the mStar software on it to the latest version of the software and discovered a bug where it's saving the video file with a filename that doesn't match the time. This is causing problems in software I've written that syncs video files from various times across cameras as it's not grabbing the right files that coordinate with the times gathered from other cameras as it once did.

For example I'm using all my cameras with Los Angeles time zone string specified PST8PDT,M3.2.0,M11.1.0

The video file which corresponds to 4-12 at 11:09am is: http://rear:8080/record/2022Y04M12D18H/09M00S60.mp4
(Note the hour is 18 not 11, 7 hours off from what it should be)

The correct filename for this time should be:
http://rear:8080/record/2022Y04M12D11H/09M00S60.mp4

Please note that in the UI it correctly shows that this incorrect filename is associated with 11:00 a.m. it's just that the filename itself is incorrect and doesn't match the time. Therefore it doesn't match the filename for this time gathered on other cameras that i have. Therefore it has been causing problems with the wrong files being synced for this camera in scripts that I have to back up files for the cameras. 

I haven't noticed any of these incorrect filename problems with any of the other cameras using the various kinds of software and various versions of this mstar software. Therefore I believe this bug is coming from either the most recent version or within the last 2-4 updates.

Hope this issue can be fixed as right now the only solution I have is to downgrade to a previous version of the software. I'll have to figure out which is the last version that works. Let me know if you know what version I should downgrade to.

Best,

roleoroleo, 2022-04-16:
I played a lot with timezone in the last versions and I'm working on it in these days.
But I don't understand the current state.
Please explain better how it was before and how it is now.

rathaROG, 2023-01-16:
I think mine has the same problem. It's 1 hour different, and I guess it's related to the folder name? I have to seek the timeline on app 1 hour later; for example, if I want to see the recorded footage at 10:00 AM, I must seek to 11:00 AM.

roleoroleo, 2023-01-21:
Did you set time zone both in the app and in the web page?

rathaROG, 2023-01-21:
> Did you set time zone both in the app and in the web page?

Yes both app and webpage have the exact same time zone, and there's nothing wrong on the event webpage `/?page=eventsdir`. The names of the dir or folder are correct. I have no idea why the app can't seek the time correctly. 

Mine is "9FUS", came with firmware 4.2.0. I used the latest MStar 0.4.8, tried both (4.2.0* y25) and (4.5.0* y203c), but no luck. 

I also has a newer 9FUS with firmware 8.3.0, and there is no such problem with your Allwinner 0.3.2. Everything just work perfectly.

roleoroleo, 2023-01-21:
Do you remember if the problem already existed before installing the hack?

rathaROG, 2023-01-22:
> Do you remember if the problem already existed before installing the hack?

No there was no such problem before. I used the stock firmware for years, and this issue only presents after I installed the hack last week.

roleoroleo, 2023-01-22:
Are you using continuous recording or just events?

roleoroleo, 2023-01-22:
I checked the problem and it has always existed.
I will fix it in the next release.
But this fix will introduce a "problem": the cam will save date/time in UTC (as original fw does).

rathaROG, 2023-01-23:
> I checked the problem and it has always existed. I will fix it in the next release. But this fix will introduce a "problem": the cam will save date/time in UTC (as original fw does).

Yes, continuous recording. I'm glad you identified bug. I'm looking forwards to the next release. Thanks!

smithumble, 2024-02-08:
> I'm looking forwards to the next release.

Wasn't this behavior changed?
I have the same problem in 0.5.4

roleoroleo, 2024-02-08:
The last release changes completely this behavior.
So, please, describe the problem again.
What's the local time shown in Status page?
What's the "Date & time" column in Events page? Local time? GMT? Other?
What's the name of the file/folder structure in/tmp/sd/record? Local time? GMT? Other?

smithumble, 2024-02-08:
I set timezone in settings like this
[image]

In status page there is local time
[image]

Also local time in in Events page
[image]

[image]

Here is file/folder structure. It's also local time.
[image]
The red one is with configured timezone
The greens is when timezone is not set (blank on settings page)

When timezone is set I can't seek in app timeline properly

smithumble, 2024-02-08:
So as I understand the app requires GMT time in file structure

roleoroleo, 2024-02-09:
Please, overwrite syste.sh with this one:
[system.sh.gz](https://github.com/roleoroleo/yi-hack-MStar/files/14224093/system.sh.gz)

smithumble, 2024-02-09:
Looks like it did nothing

Check that I did it right
[image]

md5 of original and replaced
[image]

folders still with local time
[image]

smithumble, 2024-02-09:
oh, sorry, I was wrong

it did something

[image]

looks like strait after reboot it saves first file to folder with GMT time
and next files to folder with local time

smithumble, 2024-02-09:
BTW maybe it makes sense to make a switch for the time zone in the saved files/folders?

Something like "Force GMT for saved video directory tree"

So that those who need the app can enable it and those who don't can have the correct local date/time

roleoroleo, 2024-02-10:
Mmmmmmmhhh
The problem is not yet fixed.
I need to check again.

roleoroleo, 2024-02-12:
After many days of analysis, I understand that this problem is very complex.

If you use the cam with the factory firmware, it works in GMT: the file/folder structure is saved in GMT and the software translates the time for you (with the timezone that you configure in the app).
The cam knows the timezone because cloudAPI gets this value from the cloud and sets it in memory, but the cam continues to work in GMT.

When you switch off the cloud, cloudAPI_fake replies to the cam with the same configuration sent by the cloud, so there shouldn't be any differences.

Other elements impact this behavior: ntpd, timezone, set_tz_offset_osd.
ntpd: if you use ntpd, the time comes from an internet server, but should be the same time sent from the cloud (not tested).
timezone: if you set timezone in the webgui, /etc/TZ is filled and all the processes can read the file (to check again).
set_tz_offset_osd: if you enable time osd in the webgui, the cam changes the time used to save the file. If you don't use the cloud it's not a problem, if you use the cloud you will experiment the problem reported in the video above.

I didn't finish the test because it's very difficult and time expensive. But I will do...

roleoroleo, 2024-03-01:
I did all the tests I could and there is no full solution.
I managed to fix the problem about folder structure but I didn't find a solution for the timeline in the app.
When I switch on the the "Time OSD" flag the timeline is wrong.
Please, try this beta:
[h201c_0.5.4.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/14461323/h201c_0.5.4.tar.gz)

closed 2024-06-30 by github-actions[bot]