roleoroleo/yi-hack-MStar #437: Home Assistant ONVIF reboot command doesn't work

apmillen, 2022-04-13:
I have integrated the my Yi Dome 1080p camera into Home Assistant using ONVIF and the stream and PTZ all work as expected. However the reboot command does not seem to do anything to the camera.

Am I missing something?
If this isn't a feature the firmware offers, is there another way of rebooting the camera remotely?

roleoroleo, 2022-04-16:
Reboot via onvif is not currently supported.
I could try to add it but I need time...
At the moment you can use this service:
http://ip:8080/cgi-bin/reboot.sh

apmillen, 2022-04-19:
Works like a charm. Thank you!

roleoroleo, 2022-04-19:
Check the last release of the hack.
I added the feature:
https://github.com/roleoroleo/yi-hack-MStar/commit/bcb052bbebbebc701634bf4cb8a3febe45d2e1ec

apmillen, 2022-04-19:
Ahhh brilliant! I'll give it a go when i get a chance!

closed 2024-05-06 by github-actions[bot]