roleoroleo/yi-hack-MStar #441: YI 1080p Home Camera 6FUS doesn't connect to WI-FI

Creo2005, 2022-05-20:
I bought a cheap camera that did not read the code from the phone and did not connect to WI-FI.
MStar msc313e
I flashed it with firmware [y203c_0.4.7.tgz](https://github.com/roleoroleo/yi-hack-MStar/releases/tag/0.4.7#:~:text=18%20Apr%202022-,y203c_0.4.7.tgz,-11.6%20MB)
After that I flashed my WI-FI data with this [yi-hack folder](https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials-(deprecated-from-0.4.0))
Then the camera start and yellow flash then blue LED started flashing.
But the camera still won't connect, endless reload.
Attaching the log file after flashing the settings and log file from Putty, 
and can't stop downloading firmware in Putty (Ctrl-C, Enter etc. not help)
Please help me start the camera.
Maybe desoldering the memory chip and flashing it on the programmer will help?
[log.txt](https://github.com/roleoroleo/yi-hack-MStar/files/8742373/log.txt)
[putty.log](https://github.com/roleoroleo/yi-hack-MStar/files/8742377/putty.log)

roleoroleo, 2022-05-21:
I can confirm that your cam is a y203c.
Try again with y203c_0.4.7

Creo2005, 2022-05-21:
> y203c_0.4.7
What means try again?
Make a SDcard again with [y203c_0.4.7.tgz](https://github.com/roleoroleo/yi-hack-MStar/releases/tag/0.4.7#:~:text=18%20Apr%202022-,y203c_0.4.7.tgz,-11.6%20MB) ?
and after with [yi-hack folder](https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials-(deprecated-from-0.4.0))?
And another question - in  file configure_wifi.cfg:
wifi_ssid=My AP 2.4GHz
wifi_psk=password$ --------- "$" it must be after the password?

roleoroleo, 2022-05-21:
Yes, both.
password$ is the whole password.
You don't need to add a $ sign at the end.

Creo2005, 2022-05-24:
> Yes, both. password$ is the whole password. You don't need to add a $ sign at the end.

I tried again with no result.
Yellow flash after bluу flash then the camera clicks and so on in a circle.
Attach log on sd card after flash (add name to zip file

[flash_fw_log_no_wifi_dev.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/8763242/flash_fw_log_no_wifi_dev.tar.gz)
[flash_pswd_log_no_wifi_dev.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/8763243/flash_pswd_log_no_wifi_dev.tar.gz)
)

What else can be done?

Creo2005, 2022-05-24:
On the router in the log file I do not see attempts to connect :(

Creo2005, 2022-05-24:
Where to find this file - wpa_supplicant.conf ? He was absent in https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials-(deprecated-from-0.4.0)

Creo2005, 2022-05-24:
MAC-filter OFF

roleoroleo, 2022-05-24:
Data are ok and the cam is configured properly. I don't know why it doesn't work.
You should try again to sync the cam with the app.

Creo2005, 2022-05-24:
> 

> Data are ok and the cam is configured properly. I don't know why it doesn't work. You should try again to sync the cam with the app.

The fact of the matter is that I bought a camera used and the seller said that the camera does not connect to the app. This is true - the camera does not accept the QR code from the phone - nothing just happens.

roleoroleo, 2022-05-24:
It depends on the cam, but there are other methods to sync the app without using the qr code.
Try the audio method if available.

Creo2005, 2022-07-13:
Bought on Aliexpress W-fi module - "RTL8188FTV USB WIFI Wireless Network Card Adaptor"  and replace :) All is OK!!
[image]
[image]
[image]

andriej, 2022-09-12:
I have same issues with 9FUS except that it doensn't connect on place where it's supposed to (two APs with same SSID and roaming enabled) but it does on different location (3rd AP with same SSID/pass) - all 3 of them on different channels...

closed 2024-05-06 by github-actions[bot]