roleoroleo/yi-hack-MStar #444: Snapshot taken at the beginning of each video recording.

twosheds, 2022-06-02:
I have a 1080p Dome camera (BFUS) which I use to record a video when it detects motion. It has now started to take a snapshot at the beginingof each video that it records. I've checked all my settings and can't see any setting that might cause this. Is this a bug or is there a way of stopping this behaviour ?

Thanks for any advice that anyone might give.

twosheds, 2022-06-04:
Can this feature  be switched off ?

roleoroleo, 2022-06-04:
Officially it is not possible.
But you could use startup.sh to change crond configuration:
```
killall crond
cat /var/spool/cron/crontabs/root | grep -v thumb.sh > /tmp/cron_root
mv /tmp/cron_root /var/spool/cron/crontabs/root
/usr/sbin/crond -c /var/spool/cron/crontabs/
```

closed 2024-05-06 by github-actions[bot]