roleoroleo/yi-hack-MStar #447: "Get recorded videos" web request return image instead of a video

marcor2196, 2022-06-18:
Hi, when i perform this web request on my camera:
http://192.168.1.47:8080/cgi-bin/getlastrecordedvideo.sh?type=2

I receive this response:
http://192.168.1.47:8080/record/2020Y05M06D10H/59M00S60.jpg

An image instead of a video, indeed if I change the "type" parameter to 3 in order to get the video player in the browser, the player try to start and then stops.

According to the Wiki this request should return always a video, right?
I receive always an image, also if I set the oldness parameter.

Thank you!

roleoroleo, 2022-06-25:
You are right.
The service was not updated after I added the snapshot service.
Try these commits:
https://github.com/roleoroleo/yi-hack-MStar/commit/778119d41b4e814025ba0f43b99760d405b26ffe
https://github.com/roleoroleo/yi-hack-MStar/commit/4d244cbfa58a0f5baf554504e2c4363f7a253d3d

marcor2196, 2022-06-27:
Hi, I can confirm you that seems fixed.
Now I received correctly videos.

However seems there is another problem with the "record" web request.
When I perform this request: http://192.168.1.47:8080/cgi-bin/record.sh?time=15
I received the response {"error":"false"}, but the recorded video it's not 15 seconds long.
Sometimes it's 17 seconds long or 38 seconds long or 55...

My final goal would be trigger manually a recording request of 15 seconds with "record" web request, get this recorded video with the "get recorded video" web request and finally send to my smartphone (all with Home Assistant).

It would be great if with a dedicated parameter, the "record" web request will respond directly with the recorded video (in the same way of request "get recorded video" with "type=3" response) instead of "error" json response.

If it's useful my camera is "Yi 1080p Home 9FUS".

Thank you so much for the help and for your time!

roleoroleo, 2022-06-29:
> However seems there is another problem with the "record" web request. When I perform this request: http://192.168.1.47:8080/cgi-bin/record.sh?time=15 I received the response {"error":"false"}, but [...]

This web service starts a records with a fake motion start message followed by a stop message after TIME seconds.
Unfortunately I can't control the exact duration of the video because it's managed by original yi processes.

closed 2024-05-06 by github-actions[bot]