roleoroleo/yi-hack-MStar #448: ssh login failed

bonuzzz, 2022-07-01:
I can't login to camera through ssh and telnet after reset to defaults. I received incorrect password (root login) even with set password in web. I have this situation with all 5 mstar cameras, but in 2 allwinner cameras ssh works without problem
0.4.7 fw
[config.7z.zip](https://github.com/roleoroleo/yi-hack-MStar/files/9030833/config.7z.zip)

roleoroleo, 2022-07-05:
In the config the password is "admin".
Did you try it?

I will check the reset procedure.

roleoroleo, 2022-07-05:
Now it should work:
https://github.com/roleoroleo/yi-hack-MStar/commit/2eb42be9d6ef52e75932295e2ab8ac3d6425f24c

Overwrite option was missing.

bonuzzz, 2022-07-05:
@roleoroleo could you upload test firmware, because now I don't have any methods to edit script in cameras

roleoroleo, 2022-07-05:
I don't remember your model.

roleoroleo, 2022-07-06:
Here it is:
[y203c_0.4.7.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/9052433/y203c_0.4.7.tar.gz)

bonuzzz, 2022-07-07:
Unfortunately I can't test it because I can't downgrade firmware. Folder .fw_upgrade just created and that's all

roleoroleo, 2022-07-07:
It's not necessary to downgrade the firmware.
Simply place home_y203c in the sd card. The checksum is different from your actual version and the bootloader will process it.

bonuzzz, 2022-07-07:
it works! thanks!

closed 2022-07-07 by bonuzzz