roleoroleo/yi-hack-MStar #456: Commands in the console to create a login and password RTSP stream

NeroWard, 2022-08-24:
Prompt commands in the terminal to manually set the RTSP stream login and password and start a secure RTSP stream

roleoroleo, 2022-08-24:
Use -u option for the user and -w option for the password.
Please note that the stream is not secure, there is no encryption.

roleoroleo, 2022-08-24:
Use -u option for the user and -w option for the password.
Please note that the stream is not secure, there is no encryption.
Just authentication.

NeroWard, 2022-08-24:
An example of a command?

roleoroleo, 2022-08-24:
What's your model suffix?

NeroWard, 2022-08-24:
https://github.com/niceboygithub/AqaraCameraHubfw/tree/main/binutils
I have an Aqara G2, but this project uses your work to make RSTP work. Perhaps I need to use the updated rRTSP from your project

NeroWard, 2022-08-26:
Could you build rRTSP into an executable for me? My Linux knowledge is not enough to do it myself

roleoroleo, 2022-08-26:
It's not a simple task.
I need the sdk tool to build a source for your platform.
Cross compiler, libc libraries, etc...

NeroWard, 2022-08-31:
> `rRTSPServer -r both -c h264 -C h264 -a alaw -p 554 -n 0 -u user -w password`

The console says "invalid option".Even if you enter each argument separately

roleoroleo, 2022-09-02:
Probably he changed the source code or it's a very old version.

closed 2024-05-06 by github-actions[bot]