roleoroleo/yi-hack-MStar #458: Missing "motion_start" event/callback

juacas, 2022-08-25:
My camera Yi 1080p Home 4FUS | 4.2.0 seems to work as intended except for the following.

When the camera detects any activity it launches an event to the official App but does not publish motion_start in the MQTT topic CAM_ID/motion_detection
When the motion ends, the camera publish "motion_stop" in the MQTT topic CAM_ID/motion_detection
Other MQTT topics are also never published: motion_detection_image, motion_files.

Another symptom (may be related) is that the birth topic CAM_ID/status changes from online to offline every second.

Regards.

"fw_version": "0.4.7", "home_version": "4.2.0.0H_201909041620", "model_suffix": "y25",

@sistemascotesa

  
Add heading textAdd bold text, <Ctrl+b>Add italic text, <Ctrl+i>
Add a quote, <Ctrl+Shift+.>Add code, <Ctrl+e>Add a link, <Ctrl+k>

roleoroleo, 2022-08-25:
Please check if you are using a "unique" client id.
Don't share it with other services or devices because the broker will close the connection.

juacas, 2022-08-25:
Hello. It is the only Yi Cam in my network.

The mosquitto's logs shows the very same you are suggesting: somehow there are continuous reconnections (several per second) to MQTT broker from the same clientID.

I hope this report will help to identify the issue.

My  observations are:

1. The Camera begins to reboot after a minute (maybe it gets overheated). Starts up un-paired. It was needed to pair ir with the WIFI  with the QR code of Yi's app.
2. After a pair of  reboots, the topic that now is published every second is `CamID/camera_setting` (Previously it was `CamID/status`)
3. I disable "MQTT Camera Settings Advertise", "MQTT Telemetry Advertise" and it gained some stablility. However, the camera reboots randomly after some minutes ( rmm process aparently crashes ).
4. 7. I disable all "Start on boot" and "Periodic start" on HomeAssistant integration page and the MQTT connections goes normal.
5. There are only two connections for two different clientID :

- Yi1080pcalleCliendID
- Yi1080pcalleCliendID_c

6. I fully disable  MQTT. As expected, the offending connections stopped.
8. I enabled "Home Assistant MQTT discovery advertise service to start on boot". 
9. The problems with the connections reappears. But the MQTT clientID is not the expected "Yi1080pcalleCliendID" but "Yi1080p-calle-host" which is the "hostname".
10. Enable only "MQTT Link Advertise" and no bad connections are seen in logs.
11. Enable "Home assistant MQTT discovery advertise" and the problem returns: multiple reconnections using "hostname" as MQTT clientID.
12. Enable "Home assistant MQTT discovery advertise" with "periodic start" with cron "0 * * * *" and there is no extra connections... let's see in the hour o'clock. 

It seems that there is something broken in "Home assistant MQTT discovery advertise" with "Start on boot".

```
2022-08-25T19:15:54: New client connected from 192.168.160.1:41358 as Yi1080pcalleCliendID_c (p2, c1, k15, u'xxx').
2022-08-25T19:15:54: New connection from 192.168.160.1:41359 on port 1883.
2022-08-25T19:15:54: New client connected from 192.168.160.1:41359 as Yi1080pcalleCliendID (p2, c1, k120, u'xxx').
2022-08-25T19:16:07: New connection from 192.168.160.1:41364 on port 1883.
2022-08-25T19:16:07: New client connected from 192.168.160.1:41364 as Yi1080p-calle-host (p2, c1, k60, u'xxx').
2022-08-25T19:16:07: Client Yi1080p-calle-host disconnected.
2022-08-25T19:16:07: New connection from 192.168.160.1:41365 on port 1883.
2022-08-25T19:16:07: New client connected from 192.168.160.1:41365 as Yi1080p-calle-host (p2, c1, k60, u'xxx').
2022-08-25T19:16:07: Client Yi1080p-calle-host disconnected.
```

I hope this can help.

juacas, 2022-08-25:
By the way, returning to the issue's title,  "motion_start" events are missing. MQTT only receives "motion_stop" events.

Is there any way I can help to diagnose it?

Thank you for your superb work and help.

roleoroleo, 2022-08-26:
Normally, when rmm crashes and the cam reboots is a problem memory: memory is too low.
Remove unnecessary services.
Disable all "MQTT Advertise & Homeassistant MQTT Discovery" at the moment. If you use HA please try this integration https://github.com/roleoroleo/yi-hack_ha_integration

How did you configure the motion section in the app? Baby crying? Human detection?

juacas, 2022-08-26:
Human detection: ON
Baby crying: OFF

but has the same behaviour with:

Human detection: OFF
Baby crying: OFF

roleoroleo, 2022-08-26:
With human detection off it should work.Try again.
-------- Messaggio originale --------Da: Juan Pablo de Castro ***@***.***> Data: 26/08/22  14:56  (GMT+01:00) A: roleoroleo/yi-hack-MStar ***@***.***> Cc: roleo ***@***.***>, Comment ***@***.***> Oggetto: Re: [roleoroleo/yi-hack-MStar] Missing "motion_start" event/callback (Issue #458) 
Human detection: ON
Baby crying: OFF
but has the same behaviour with:
Human detection: OFF
Baby crying: OFF

roleoroleo, 2022-08-26:
With human detection off it should work.
Try again.

juacas, 2022-08-27:
You're right. With Human detection off it works.
Thanks. 

Is this a known issue or a bug?

Regards.

roleoroleo, 2022-08-27:
It's a know issue but I completely changed the way I manage these messages.
So, probably I fixed the problem..
Next release.

closed 2024-05-06 by github-actions[bot]