roleoroleo/yi-hack-MStar #461: HighRes Snapshot functionality not working (Dome X y30)

thoemmes86, 2022-09-16:
Hi,

I'm currently running 3 Yi Dome X cameras (y30) with this firmware (version 0.4.7).
For whatever reason, for 1 of those cameras the Snapshot functionality in HighResolution is not working.
Reboot, Firmware reinstall, Reset... are not solving this issue. Configuration is identical with the other 2 cameras.

WORKING:
http://IP-ADDRESS:8080/cgi-bin/snapshot.sh?res=low&watermark=yes

NOT WORKING:
http://IP-ADDRESS:8080/cgi-bin/snapshot.sh?res=high&watermark=yes
No file available.

Has anybody an idea how to solve this?

Thanks.

roleoroleo, 2022-09-19:
Try to run the command manually.
`imggrabber -m y30 -r high -d`

thoemmes86, 2022-09-19:
Following I get as result:
/home/yi-hack # imggrabber -m y30 -r high -d
debug on
Resolution: 1080
VMFE0
vaddr: 0xb47eb000 - paddr: 0x23510000 - size: 368640
copy buffer: len 368640
copy again buffer: len 368640
convert YUV image
Segmentation fault

roleoroleo, 2022-09-21:
Segfault is not normal...
Does high res stream work?
Please check the yi firmware revision (with the app or with `# cat /home/homever`).
Both for working cam and not working cam.

roleoroleo, 2022-09-22:
Ok, I don't know why but...
Run the grabber with stdout redirection and send me the resulting file (both cams):
```
h264grabber -m y30 -r high > /tmp/sd/cam1.h264
h264grabber -m y30 -r high > /tmp/sd/cam2.h264
```
Leave the process running for 10 seconds and break with CTRL-C.

thoemmes86, 2022-09-23:
Attached the links to the files:
https://1drv.ms/u/s!AqxFu6uAVIMUgaJeZQ0gyYAQG-2_Vw?e=1VWpZv
https://1drv.ms/u/s!AqxFu6uAVIMUgaJfmpEa0x-wbyrqEA?e=hrtNqM

roleoroleo, 2022-09-30:
I don't know why but the stream is different.
Different SPS, PPS, etc...

Please send me 3-4 copies of the file /dev/fshare_frame_buf
Copy it, wait 5 secs, copy it again, wait...
I need to check better the stream.

wlcims, 2023-08-11:
Use the yi app to modify the H265 stream to H264 stream, and the high-resolution stream will be ok.

closed 2024-05-06 by github-actions[bot]