roleoroleo/yi-hack-MStar #464: MTU is 1200 bytes -> packet fragmentation

robho, 2022-11-15:
In /home/app/script/default.script the MTU for wlan0 interface is set to 1200 bytes:

> echo "1200" > /sys/class/net/wlan0/mtu

This means that many of the RTP packets sent by the RTSP server need to be fragmented into two packets.

I've removed the MTU change and am now using the default of 1500 bytes. It looks to work well for me and now all RTP packets are sent without being fragmented. But I've only tested local playback, I don't use the cloud services.

Do you know if there's a good reason for lowering the MTU? It may be a good idea to use the default MTU size and let other network devices deal with fragmentation if these other devices need to lower the MTU to reach the destination.

roleoroleo, 2022-11-16:
If I remember correctly, video packets are about 1000 bytes, so probably it does not depend on the MTU.
But I will check it.

robho, 2022-11-16:
Here's a screenshot from wireshark with MTU = 1200:
[image]
.. and here's a screenshot from wireshark with default MTU (1500):
[image]

roleoroleo, 2022-11-17:
I tested your changes with the app and I didn't notice any problem.
And the performance is better.
So I will add it to the startup script.
I think it's not enough because the wifi script probably updates mtu frequently.
I will also add it to the watchdog script.

Thank you for your help.

robho, 2022-11-17:
Great!

Is the reason for not modifying default.script that the user should be able to uninstall yi-hack?

roleoroleo, 2022-11-17:
Yes.
I prefer to work only on the hack files.
And I don't know if default.script is the only file involved in this procedure.

robho, 2023-04-12:
Fixed by 48fd5b35f9e27b1969faeea8c19de56ed74755bb and included in release 0.4.8 -> closing this one

closed 2023-04-12 by robho