roleoroleo/yi-hack-MStar #472: Yi Home Dome 1080P on v.0.4.8 Reboot Loop

idanny76, 2022-12-09:
I have had two problems and I am not sure if its just happening with two cameras. Where I have done a reboot from the web interface and then it gets stuck on a reboot loop. The only way I have been able to fix it is to delete the recording and leave the flash files only.

roleoroleo, 2022-12-09:
Disable snapshots for records.

idanny76, 2022-12-10:
> Disable snapshots for records.

I think it worked. Will let you know if I have any issues with reboot loop. Just have one left that I have to remove the SD card to be able to stop the reboot loop.

idanny76, 2022-12-19:
> Disable snapshots for records.

I had another issue with the v0.4.8.... It was causing the NVR to lag on live view and just the whole NVR was very slow. I thought it was the hard drive or that the NVR needed to reboot. I rebooted and also shut down and restarted the NVR and it was still lagging. I went ahead and ran a test on the hard drive and also formatted it and it did not fix the issue. I decided to revert the firmware to the 0.4.7 and the problem went completely away. Not sure how the 0.4.8 firmware caused the issue, but it made the NVR useless. Right now its all working smooth with the 0.4.7. For me the 0.4.8 firmware was causing too many problems.

roleoroleo, 2022-12-19:
How is connected the nvr to the cam?
Through onvif driver or directly in rtsp?

idanny76, 2022-12-19:
> How is connected the nvr to the cam? Through onvif driver or directly in rtsp?
> 
> There are 3 changes related to onvif/rtsp:
> 
>     * Update onvif_srvd to the last commit
> 
>     * Update live555 to the last version
> 
>     * Avoid packets fragmentation
> 
> 
> But they should be improvements... I could try to send you old versions of onvif_srvd and rRTSPServer to check if the problem is there.

I think it's Onvif
[image]

roleoroleo, 2022-12-20:
If you want to spend some time to help me solve the problem...
These are the applications of the new version 0.4.8:

[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-MStar/files/10266342/rRTSPServer.gz)
[onvif_srvd.gz](https://github.com/roleoroleo/yi-hack-MStar/files/10266341/onvif_srvd.gz)
[h264grabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/10266340/h264grabber.gz)

Please try to overwrite you existing binary (0.4.7) starting from rRTSPServer, then onvif_srvd and finally h264grabber.
Move the original binary to the sd card, copy the new one and check if the problem with your NVR appears.
```
- decompress the archive with your PC
- # killall rRTSPServer
- # mv /home/yi-hack/bin/rRTSPServer /tmp/sd
- copy rRTSPServer with a ftp client to /home/yi-hack/bin
- chmod 0755 /home/yi-hack/bin/rRTSPServer
- # reboot
- test the stream with your NVR
- # del /home/yi-hack/bin/rRTSPServer
- # mv /tmp/sd/rRTSPServer /home/yi-hack/bin
- chmod 0755 /home/yi-hack/bin/rRTSPServer
```

The same for onvif_srvd and h264grabber.

idanny76, 2022-12-20:
> If you want to spend some time to help me solve the problem...
> These are the applications of the new version 0.4.8: [...]

Sure.... I'll see if I can test it out today.

idanny76, 2022-12-23:
I did one file at a time. Basically I 
-Renamed rRTSPServer to rRTSPServer.old. 
-Copied the new rRTSPServer file and chmod to 755. 
-rebooted and camera worked fine. 

Then
-Removed the new rRTSPServer and renamed the old rRTSPServer
-Renamed onvif_srvd to onvif_srvd.old and added the new file and chmod to 755.
-Rebooted camera and did not get video on NVR.

Same steps were followed with the h264grabber and after reboot no video on the NVR. So I decided to renamed all the three 0.4.7 files to ,old and copy the new ones, chmod all 3 to 755 and rebooted. After reboot, I did not get any video.

Not sure why the NVR is not getting video with the new files from 0.4.8 in the 0.4.7 firmware. I do get video if I upgrade to the 0.4.8 firmware.

idanny76, 2022-12-23:
Going to play around by renaming the new files .new and renaming the .old to file name randomly and rebooting to see when I get video.

roleoroleo, 2022-12-23:
If you are using rtsp_server_yi I need to check the difference between 0.4.7 and 0.4.8.

idanny76, 2022-12-23:
> If you are using rtsp_server_yi I need to check the difference between 0.4.7 and 0.4.8.

There is only one scenarios where I get video with the new files... I tried the different combinations and only the new rRTSPServer with the old h264grabber and old onvif_srvd. But I am using the rtsp_server_yi RTSP alternative server.

-Video
Old h264grabber
Old onvif_srvd
New rRTSPServer

-No Video
Old h264grabber
Old rRTSPServer
New onvif_srvd

-No Video
Old h264grabber
New onvif_srvd
New rRTSPServer

-No Video
New h264grabber
New rRTSPServer
Old onvif_srvd

-No Video
New h264grabber
New rRTSPServer
New onvif_srvd

idanny76, 2023-01-17:
Upgraded the cameras from the 0.4.7 to the 0.4.8 again to test RTSP server without the alternate. There is something not right with the 0.4.8 version. Rebooting the camera takes a very long time and other times it just keeps rebooting until it finally connects online. Two of the cameras micro SD cards went bad during the upgrade, because the cameras stayed looping until they finally came back online and the SD cards were damaged. Not sure why rebooting the cameras on the 0.4.8 takes longer than the 0.4.7.

idanny76, 2023-01-17:
Also, I used to be able to do Focus with the NVR with the Onvif connection..... but after some updates I am just able to zoom in and out only. I am using Onvif with all the cameras so I can use the PTZ function on the NVR. With RTSP connection I loose all the PTZ functions.

roleoroleo, 2023-01-21:
If you are using alternate rtsp server rRTSPServer is not the problem.
Please, try the old grabber with this new onvif_srvd:
[onvif_srvd.gz](https://github.com/roleoroleo/yi-hack-MStar/files/10472618/onvif_srvd.gz)

GuybrushX, 2023-02-08:
I can confirm that issue: reboot loop ☹️

I didn't try any fixes yet -> what is the current state? Can I help somehow providing logs or is a newer unreleased version available?

idanny76, 2023-02-08:
> I can confirm that issue: reboot loop ☹️
> 
> I didn't try any fixes yet -> what is the current state? Can I help somehow providing logs or is a newer unreleased version available?
> 
> The camera was used with the stock firmware for a couple of months until I found this repo and gave it a try. So there have been recordings and maybe other data from the stock firmware on the SD card already (32GB, Sandisk).
> 
> After updating to this version it was working for some time, than I configured MQTT and I think this somehow caused the reboot loop but not sure.

I have been busy and haven't tried to replaced the old grabber with this new onvif_srvd. I am just switched to the 0.4.7 version, because I lost two sd cards to the reboot loop issue. If you want to try the file fix and let us know if it did fix the reboot issue. The weird thing is that it only happens on come cameras and not all.

idanny76, 2023-02-23:
I tried the new 0.4.9 update and it works perfect in some cameras. In other cameras, but the same type.... they keep rebooting until it finally goes online and start working good. The issue is that in some cameras after it does update after about 8 reboots more or less....... If I ever need to reboot those specific cameras they always go into a reboot loop that will eventually connect. Not sure why some cameras work fine and others don't in the sense of reboots. Its just doesn't make sense when all the cameras are the same and have the same firmware. It weird that some just have to reboot many times before they go online and other reboot and go online immediately.

roleoroleo, 2023-02-25:
I try to restart from scratch.
Pleae, share your system.conf (clear personal data).

It could be a memory issue or a race condition during system start.

idanny76, 2023-02-25:
> I try to restart from scratch. Pleae, share your system.conf (clear personal data).
> 
> It could be a memory issue or a race condition during system start.

Here is the system.config
[system.zip](https://github.com/roleoroleo/yi-hack-MStar/files/10831026/system.zip)

roleoroleo, 2023-02-26:
Please, try to add a sleep between these lines in system.sh:
```
        LD_PRELOAD=/home/yi-hack/lib/ipc_multiplex.so ./dispatch &
        LD_LIBRARY_PATH="/home/yi-hack/lib:/lib:/home/lib:/home/ms:/home/app/locallib" ./rmm &
```
```
        LD_PRELOAD=/home/yi-hack/lib/ipc_multiplex.so ./dispatch &
        sleep 2
        LD_LIBRARY_PATH="/home/yi-hack/lib:/lib:/home/lib:/home/ms:/home/app/locallib" ./rmm &
```
Lines 123 and lines 161.

If this change doesn't work, we will try to start the system step by step.

idanny76, 2023-02-26:
I will try in 30 minutes and reply back.

idanny76, 2023-02-26:
> Please, try to add a sleep between these lines in system.sh:
>
> ```
>         LD_PRELOAD=/home/yi-hack/lib/ipc_multiplex.so ./dispatch & [...]

I did it and it is still on a reboot loop. I did notice that the camera after it reboots it never goes all the way back to the half way position. Not sure if when it reboots, it is supposed to go back to the last position or is it supposed to go back to the half way position and then to the last position.

idanny76, 2023-02-26:
I think I fixed... Let me do another cameras and I'll let you know.

idanny76, 2023-02-26:
The fix for me was in FTP connection deleting all the content of the SD Card and rebooting the camera. That's all it took to fix the reboot loop. Something after an update causes something on the SD card that makes the reboot issue. I did try just deleting the firmware files in the SD Card only, but that did not fix it. What fixed it was deleting the Firmware files, the log folder and the record folder. Not sure where the issue is.... but it could be in either the log or record folder.

idanny76, 2023-02-26:
It's crazy that just by deleting everything on the SD Card fixed the reboot loop on those specific cameras. Not sure what on the record or log folder was causing the camera to go into a reboot loop. Now all the cameras boot up very quickly.

roleoroleo, 2023-02-27:
Mmmmmmh...
It could be the thumbnails feature.
crond runs a process that creates snapshots for all mp4 files in the folder /tmp/sd/record
The process uses a lot of memory when it converts the frame to jpg image.
This process can be switched off with the option SNAPSHOT_VIDEO (system.conf).

idanny76, 2023-02-27:
> Mmmmmmh... It could be the thumbnails feature. crond runs a process that creates snapshots for all mp4 files in the folder /tmp/sd/record The process uses a lot of memory when it converts the frame to jpg image. This process can be switched off with the option SNAPSHOT_VIDEO (system.conf).

I went ahead and disabled it anyways. I disabled all the snapshot features. At least its nothing to do with the firmware and it was a simple fix by just deleting everything on the SD Card. Hope it helps other people with reboot loop issues.

GuybrushX, 2023-02-28:
Thanks for the update and the comments.

I upgraded to 0.4.9 as well and formatted the SD card to be safe -> the upgrade had to be done from within the web interface and needed a few tries, somehow it didn't work using the SD card but that could be me.

Since the camera was running for some time with the original firmware the SD card was very full and rotating recordings so far with the original firmware.

So far the camera seems to work well without any reboots. I'll report back after some time and will also try it on a 2nd identical device for comparison.

Summary:
- The camera is running with stock 0.4.9 now
- SD card was formatted completely to avoid issues
- I've only changed the hostname and the timezone -> btw. the timezone is ignored in the snapshot so it's showing the time in UTC.

roleoroleo, 2023-03-01:
> the timezone is ignored in the snapshot so it's showing the time in UTC.

What's the suffix of your cam?
Do you mean the time in the bottom right corner?

GuybrushX, 2023-03-01:
> > the timezone is ignored in the snapshot so it's showing the time in UTC.
> 
> What's the suffix of your cam? Do you mean the time in the bottom right corner?

I assume you mean that suffix:
Model Suffix	h201c

And yes, exactly that time in the bottom right corner on the "Snapshot" page.

roleoroleo, 2023-03-03:
Where did you configure timezone?
Webgui? Yi app? Both?

GuybrushX, 2023-03-03:
Both.

roleoroleo, 2023-03-04:
What's your timezone?

GuybrushX, 2023-03-06:
I've tried a few - for example UTC+2.

And the camera is stuck in a reboot loop again... I have recording siwtched on so maybe the card is (almost?) full again causing that loop somehow?

roleoroleo, 2023-03-06:
Try to switch off all options related to snapshot:
[image]

GuybrushX, 2023-03-06:
Same issue. What I did:
- Disconnected the USB cable to power off the cam
- Removed the SD card
- Connected to USB/power again
- Switched off both settings
- Disconnected USB cable
- Inserted SD card again with the same data like before
- Connected the cam to USB again -> still reboots

The SD card has 3494 files and 75 folders in the "record" folder. ~500MB from 32GB are left.

The camera was working fine for like 2 or 3 days until I've noticed the reboot loop again -> the camera started to turn and doing some self-test/calibration I guess, turns completely to 1 side, turns to the centre and after a few seconds it reboots.

Without the SD card everything worked well so it is indeed somehow related to the SD card (content) I guess.

GuybrushX, 2023-03-06:
I've renamed the folder to "recorddd" and it still reboots.

The SD card is unchanged -> no new "record" folder was created. Also nothing new in the "log" folder on the SD card.

idanny76, 2023-03-06:
Empty the SD card and see if the reboot stops. Delete the log and record and see if the reboot stops. But it would be nice what in the Sd card is causing this reboot issue. My conflict with it being the memory card is that this does not happen on the 0.4.7 firmware and before. I notice the reboot loop issue only happen on the 0.4.8 and 0.4.9 version. Something in the new firmware versions is causing issues with the SD card.

roleoroleo, 2023-03-06:
I compared 0.4.7 with 0.4.9 and I don't understand where the problem could be.
If you want to help me to find it, let me know.
I could prepare a special startup script with a lot of logs and something else.

roleoroleo, 2023-03-06:
I will prepare the script.

idanny76, 2023-03-07:
> I will prepare the script.

I am not sure how it would help if I try the script, because I no longer have the reboot issue after deleting the SD Cards. They all reboot fine on the 0.4.9 firmware. GuyBrushX should be able to delete the SD Card through FTP connection. I am not sure how the Yi or Kami app formats the SD Card. I did not format the SD through the app and just deleted everything from within the FTP.

roleoroleo, 2023-03-07:
> I did not format the SD through the app and just deleted everything from within the FTP.

Do you mean the ftp folder in the ftp server or the folder "record" inside the sd card?

idanny76, 2023-03-07:
> > I did not format the SD through the app and just deleted everything from within the FTP.
> 
> Do you mean the ftp folder in the ftp server or the folder "record" inside the sd card?
> 

The folder record

roleoroleo, 2023-03-22:
I don't know if this problem is related to the reboot loop, but you should try to apply this little patch.
https://github.com/roleoroleo/yi-hack-MStar/commit/222c90aef6bf9e51b9e07d3d6a8b1c15b8df863d

idanny76, 2023-03-24:
> I don't know if this problem is related to the reboot loop, but you should try to apply this little patch. [222c90a](https://github.com/roleoroleo/yi-hack-MStar/commit/222c90aef6bf9e51b9e07d3d6a8b1c15b8df863d)

Tried the patch but it doesn't fix the problem. Removing the SD card would fix it or formatting it. The problem is that after a while something gets written to the SD card and then a reboot is done at a later point the reboot loop happens and it does eventually connect, but after a long period of time stuck in the loop.

idanny76, 2023-03-24:
The whole thing is crazy. I renamed the record folder to record.old and created a new folder named record and the bootloop happens. I tried your script change and it still happens.

idanny76, 2023-03-24:
> I'm starting to think there's something wrong with the home partition. What about the free space?
> 
> ```
> /home/yi-hack # df
> Filesystem           1K-blocks      Used Available Use% Mounted on
> /dev/root                 1920      1876        44  98% /
> tmpfs                    30412      1872     28540   6% /dev
> /dev/block/mtdblock3     12480     11312      1168  91% /home
> /dev/block/mtdblock3     12480     11312      1168  91% /home/app/cloudAPI
> tmpfs                    32768       268     32500   1% /tmp
> ```

The Sd card is a 32gb and I its right now 29gb filled.

idanny76, 2023-03-24:
> I'm starting to think there's something wrong with the home partition. What about the free space?
>
> ```
> /home/yi-hack # df
> Filesystem           1K-blocks      Used Available Use% Mounted on [...]

/home/yi-hack # df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 1920      1876        44  98% /
tmpfs                    30428      1872     28556   6% /dev
/dev/block/mtdblock3     12480     11324      1156  91% /home
/dev/block/mtdblock3     12480     11324      1156  91% /home/app/cloudAPI
tmpfs                    32768        96     32672   0% /tmp

roleoroleo, 2023-03-24:
I don't know if the problem is related but the available space in /dev/root is too small.
It's not ok.
Give it a temporary try: remove some file that you don't use.
For example:
/home/yi-hack/usr/libexec/sftp-server
or
/home/yi-hack/bin/onvif_srvd

idanny76, 2023-03-24:
> I don't know if the problem is related but the available space in /dev/root is too small. It's not ok. Give it a temporary try: remove some file that you don't use. For example: /home/yi-hack/usr/libexec/sftp-server or /home/yi-hack/bin/onvif_srvd

This might help. I have two identical cameras and both are on the same firmware. One does not have the bootloop issue and the other does. I have them both here.

idanny76, 2023-03-24:
> /home/yi-hack/usr/libexec/sftp-server

deleted /home/yi-hack/usr/libexec/sftp-server, put in the sd card and is in bootloop. Without the SD card it works perfect. If I format the SD card it works also, but if I reboot at a later time... maybe after sd card is almost full then bootloop happens. I just deleted the sd card on another camera with the bootloop and then rebooted after recording for some time and it boots fine.

idanny76, 2023-03-24:
Update..... Good lead here to where the issue is. instead of deleting the record folder to fix the problem.... I made some space on the SD card about 7GB free and it bootloop again. Then I deleted more files to at least 50% free and then camera stopped bootloop. It does have to do with the amount of space available that it is causing the camera to go into bootloop.

idanny76, 2023-03-24:
What I will try now is adding a large file to the SD card and reboot to see if it goes into a bootloop.

roleoroleo, 2023-03-24:
Did you set this option?
[image]

idanny76, 2023-03-24:
> Did you set this option? [image]

I have the same camera with same firmware working fine with 99% full sd card and it reboots fine. I just filled up the memory card on the faulty one with 10gb free space and it did reboot. Filling it more now to see if I can cause it to bootloop, because right now is booting fine. I want to be doing the bootloop again to try those changes and then insert sd card.

idanny76, 2023-03-24:
Yep. filled up the camera with 3gb free space and it started doing the bootloop.

idanny76, 2023-03-24:
> Another try please: try to kill the watchdog wd_rtsp.sh (or disable it from system.sh) and check if the cam remains on.

Whats the putty command to kill the watchdog wd_rtsp.sh or how to disable it on the system.sh?

idanny76, 2023-03-24:
> 

Getting this error:
/home/yi-hack # cat /var/spool/cron/crontabs/root
cat: can't open '/var/spool/cron/crontabs/root': No such file or directory

idanny76, 2023-03-24:
> Did you set this option? [image]

I am sure this will solve the bootloop, but it doesn't seems to be a problem with the other cameras on the same firmware with 99% storage filled.

idanny76, 2023-03-24:
> Please run this command:
> 
> ```
> /home/yi-hack # cat /var/spool/cron/crontabs/root
> * * * * * /home/yi-hack/script/thumb.sh cron
> 0 * * * * sleep 20; /home/yi-hack/script/clean_records.sh 30
> ```

Ran the command in the configuration web interface where it says Expert Crontab and the camera booted with only 12% space left in the memory card. What I did notice is that the root file that was missing on /var/spool/cron/crontabs/root now appears. Why would the root file be missing and now appears? Here is a screenshot of the Crontab. I did paste this command on the box provided but only the bottom two lines remained after I rebooted.

Code:
cat /var/spool/cron/crontabs/root
* * * * * /home/yi-hack/script/thumb.sh cron
0 * * * * sleep 20; /home/yi-hack/script/clean_records.sh 30

What remained:
cat /var/spool/cron/crontabs/root
* * * * * /home/yi-hack/script/thumb.sh cron
[image]

[21 more replies]

closed 2024-05-05 by github-actions[bot]