roleoroleo/yi-hack-MStar #480: Yi-hack OneDrive file upload feature

denven, 2023-01-03:
@roleoroleo Hi, 

I've just written some scripts to support camera uploading files to OneDrive storage, I did some tests on my own camera, it works.
However, more tests and feedback is welcomed to improve the quality. Would you help add a link in the Readme file of this repository and [yi-hack-MStar.gdrive](https://github.com/roleoroleo/yi-hack-MStar.gdrive)  for some advertisement, 

[yihack-onedrive-uploader](https://github.com/denven/yihack-onedrive-uploader)

roleoroleo, 2023-01-03:
Do you want to create a new pr?

denven, 2023-01-04:
> Do you want to create a new PR?

Do you mean the doc update PR? I am fine to add a bit of words and create a new PR if you feel okay to put it inside.

For merging the feature code, maybe currently is not the time, since I only tested on my own camera. It is better to check in after more tests and user feedback.

roleoroleo, 2023-01-04:
Yes, I mean a PR for the readme.

denven, 2023-01-05:
> Yes, I mean a PR for the readme.

Thanks, please add me as a committer.

roleoroleo, 2023-01-06:
No problem to add your user as a committer, but if you want to add a PR you don't need it.
Simply fork the repo in your account, edit the file and create a pr.

denven, 2023-01-06:
Oh, I though I need 

> No problem to add your user as a committer, but if you want to add a PR you don't need it. Simply fork the repo in your account, edit the file and create a pr.
got you, I will do it this way, thanks.

closed 2024-05-05 by github-actions[bot]