roleoroleo/yi-hack-MStar #482: Telecamera non più accessibile in alcun modo

Klay4, 2023-01-24:
Ciao,
(scrivo in italiano per semplicità essendo che da quel che ho visto pure te sei italiano ahhaah)
dopo aver resettato la camera, ho per sbaglio ho disabulitato tutto... ssh, ftp e sito web.
Ho provato più volte a cambiarci firmware per cercare di recuperare la telecamera in qualche modo ma nulla... Si connette al wi-fi ma non posso accedere a nessuna gui e non posso neanche usarla dall'app essendo che il "disable cloud" era abilitato.
Cosa posso fare per recuperarla?
Grazie

roleoroleo, 2023-01-25:
Pui caricare una versione precedente dell'hack a mano (non quella che hai attualmente).
Copi i 2 file nella SD e fai il reboot.
La configurazione viene sbiancata.

Klay4, 2023-01-27:
Provo con [y203c_0.4.7.tgz](https://github.com/roleoroleo/yi-hack-MStar/releases/download/0.4.7/y203c_0.4.7.tgz) ? Ho una yi home 1080p. Domani provo e ti faccio sapere, grazie mille!

Klay4, 2023-01-27:
Grazie mille, funziona!

closed 2023-01-27 by Klay4