roleoroleo/yi-hack-MStar #491: Date and time overlap

maldini89, 2023-03-02:
If possible, please add a date and time overlay to the video stream. It would be very convenient to see this when viewing the archive.

roleoroleo, 2023-03-04:
Sorry but it's not possible.

denven, 2023-04-24:
I guess it is likely to achieve this by rewriting video files from the camera RTSP stream, rather than having it from recorded files on the camera.

roleoroleo, 2023-04-26:
The cam should decode the h264 video stream, add the tag and then encode again the stream.
There are no resources to do this job.

denven, 2023-04-27:
> The cam should decode the h264 video stream, add the tag and then encode again the stream.
> There are no resources to do this job.
> 

Yes, that's what I am saying. Not on the camera, instead, do this job at a client end when reading the stream and encode the stream into video files with the time.

roleoroleo, 2023-04-30:
Yes, you are right.
Maybe it's better with a hardware accelerated device.

maldini89, 2023-04-30:
If so, is it possible to expect this dream to come true? :)

closed 2024-05-04 by github-actions[bot]