roleoroleo/yi-hack-MStar #492: 9FUS - firmware 0.4.9 : RTSP don't work

LeidenSpain, 2023-03-09:
Hi,

I have a camera that had firmware 0.4.7 and RSTP worked perfectly.
I have updated to the latest version because the integration with Home Assistant has indicated it to me, and now I find that the RSTP does not work

I have tried to connect it with VLC and I don't really see anything, which used to work before.

Let me know if I can provide any information that will help locate the problem.

Regards

roleoroleo, 2023-03-12:
There are no changed in the last update about rtsp.
Please, try to reconfigure the cam.

LeidenSpain, 2023-03-13:
Hi,

I've downgraded to 0.4.7 and it's working again.
I update again and tell how it went.

Updated to 0.4.9 and it has stopped working

LeidenSpain, 2023-03-13:
and without touching anything, just went back to go.
I'll keep an eye on it as it seems somewhat unstable

roleoroleo, 2023-03-13:
What about 0.4.8, just to understand where look for the problem.

LeidenSpain, 2023-03-13:
I haven't even tried 0.4.8
Before I had 0.4.7, it worked, so I went back to that to do the tests.

By the way, I have both cameras with 0.4.9, I just connected through VLC, and it has worked without problems on both.

I don't know what to tell you, what used to give me a problem in both cameras now seems to work fine

closed 2024-05-04 by github-actions[bot]