roleoroleo/yi-hack-MStar #496: [h201c] Yi 1080p Dome BFUS FTP password reset each camera restart

majkers, 2023-03-29:
Hi,
each time I restart camera my FTP password resets. I have connect via SSH and use passwd command to make it work again.

roleoroleo, 2023-03-30:
Is it properly saved in the sytem.conf config file?

majkers, 2023-03-31:
No it is not:

`FTP_USERNAME=
FTP_PASSWORD=
SSH_PASSWORD=THIS_IS_MY_PASSWORD`

of cource I changed my password there

majkers, 2023-03-31:
You can check https://github.com/roleoroleo/yi-hack-MStar/issues/495 Also snapshot option in not properly saved

roleoroleo, 2023-03-31:
Are you using weird chars?
Please, try with a stupid password. Maybe a problem in the string validation algorithm.

And as I said in the other issue, clean the cache of the browser; I changed the js code in the last update.

majkers, 2023-03-31:
the only weird char might be @

roleoroleo, 2023-03-31:
@ is supported.
I tested with my cam and it works.
Are you using version 0.4.9?

majkers, 2023-03-31:
OK. I opened new incognito window

[image]

and set new root password to 123Test

and in system.conf only this:

[image]

majkers, 2023-03-31:
Yes I am using 0.4.9 version

roleoroleo, 2023-03-31:
There is a misunderstanding.
FTP_PASSWORD field is the password used to put files to a ftp server (see Events page).
If you want to login from your PC to the cam through FTP, the right field is SSH_PASSWORD
Are you able to login with a ftp client to the cam with "123Test" as a password?

majkers, 2023-04-06:
I don't know what is wrong with 0.4.9 firmware @roleoroleo After a few days of working average load grows a lot and I see my RTSP stream from go2rtc project do not run fluently. So I restart my camera and it won't come back online. Just like it is bricked. I had to downgrade and upgrade to make it work again. I think I will go back to 0.4.8 version and close this issue for now. Unfortunately IMHO 0.4.9 is not stable

closed 2023-04-06 by majkers

roleoroleo, 2023-04-06:
Sorry but I wasn't able to find the stability problem.

majkers, 2023-04-06:
One time it even didn't start record process... Maybe next version will fix those issues :) I keep my fingers crossed