roleoroleo/yi-hack-MStar #497: w: dd recovery instructions for stock firmware

meskilla, 2023-03-30:
hello roleo,

Prior to flasing yi-hack Mstar I successfully soldiered uart to my white dome 1080p cam, then dumped the stock firmware 4.6.0 using a dd snippet I found in your bugtracker:

```
dd if=/dev/mtdblock0 of=/tmp/sd/backup/mtdblock0.bin
dd if=/dev/mtdblock1 of=/tmp/sd/backup/mtdblock1.bin
dd if=/dev/mtdblock2 of=/tmp/sd/backup/mtdblock2.bin
dd if=/dev/mtdblock3 of=/tmp/sd/backup/mtdblock3.bin
dd if=/dev/mtdblock4 of=/tmp/sd/backup/mtdblock4.bin
dd if=/dev/mtdblock5 of=/tmp/sd/backup/mtdblock5.bin
```
If I where to revert this now, can I do it safely like this?
```
dd of=/dev/mtdblock0 if=/tmp/sd/backup/mtdblock0.bin
dd of=/dev/mtdblock1 if=/tmp/sd/backup/mtdblock1.bin
dd of=/dev/mtdblock2 if=/tmp/sd/backup/mtdblock2.bin
dd of=/dev/mtdblock3 if=/tmp/sd/backup/mtdblock3.bin
dd of=/dev/mtdblock4 if=/tmp/sd/backup/mtdblock4.bin
dd of=/dev/mtdblock5 if=/tmp/sd/backup/mtdblock5.bin

sync
reboot
```

Back before I flashed yi-hack Mstar I didn't know about the proper method of using the python scripts [[1]](https://github.com/roleoroleo/yi-hack-MStar/wiki/Dump-your-backup-firmware)

roleoroleo, 2023-03-30:
Yes, it should work.
Or you could simply save the image file into the sd card with the right name.
The bootloader will load them for you.

- sys_h201c for rootfs partition (mtd2)
- home_h201c for home partition (mtd3)

Pay attention to the endianness of the files, it depends on dd.
So, before to load the files, try to mount them to a linux machine to check if the files are ok.
This is a simple script to mount a jffs2 image:

```
#!/bin/bash

## Script to mount jffs2 filesystem using mtd kernel modules.
## EMAC, Inc. 2009

if [[ $# -lt 2 ]]
then
    echo "Usage: $0 FSNAME.JFFS2 MOUNTPOINT [ERASEBLOCK_SIZE]"
    exit 1
fi

if [ "$(whoami)" != "root" ]
then
    echo "$0 must be run as root!"
    exit 1
fi

if [[ ! -e $1 ]]
then
    echo "$1 does not exist"
    exit 1
fi

if [[ ! -d $2 ]]
then
    echo "$2 is not a valid mount point"
    exit 1
fi

if [[ "$3" == "" ]]
then
<------>esize="128"
else
<------>esize="$3"
fi

# cleanup if necessary
umount /dev/mtdblock0 &>/dev/null
modprobe -r mtdram &>/dev/null
modprobe -r mtdblock &>/dev/null

modprobe mtdram total_size=32768 erase_size=$esize || exit 1
modprobe mtdblock || exit 1
dd if="$1" of=/dev/mtdblock0 || exit 1
mount -t jffs2 /dev/mtdblock0 $2 || exit 1

echo "Successfully mounted $1 on $2"
exit 0

```

meskilla, 2023-03-30:
Thank you for the script. The jffs2 both look fine so I removed /etc/back.bin and rebuilt both images following your backup instruction [notes](https://github.com/roleoroleo/yi-hack-MStar/wiki/Dump-your-backup-firmware#notes)

Flashing worked fine and I got the yi software running again. Funny enough my wifi password was still saved somewhere, I found out it is part of mtd5 labeled "conf" by using a hex editor.

Can you help me erase these credentials as well so I can start over with the pairing process? mtd5 is not a normal jffs2 it seems. file magic says my dd image is a

`dBase III DBT, version number 0, next free block index 1, 1st item "<long hex code>"`

meskilla, 2023-03-30:
ah look at that ... I noticed your configure_wifi.sh does raw write credentials  into mtd5 with 2 seek offsets:
```
CURRENT_SSID=$(dd bs=1 skip=28 count=64 if=/dev/mtd/mtd5 2>/dev/null)
CURRENT_KEY=$(dd bs=1 skip=92 count=64 if=/dev/mtd/mtd5 2>/dev/null)
```
is it save to clear both values like this ?
```
dd if=/dev/zero bs=1 skip=28 count=64 of=/dev/mtd/mtd5 2>/dev/null)
dd if=/dev/zero bs=1 skip=92 count=64 of=/dev/mtd/mtd5 2>/dev/null)
```
or should I just perform

`flash_eraseall /dev/mtd/mtd5`

and the rootfs will recreate the whole database?

roleoroleo, 2023-03-31:
> is it save to clear both values like this ?

Yes, it's the same command I use to set credentials.
No problem if you have a backup copy.

> or should I just perform
> 
> `flash_eraseall /dev/mtd/mtd5`
> 
> and the rootfs will recreate the whole database?

Probably yes, but I'm not sure.

meskilla, 2023-03-31:
Thanks for confirming, you do write a "\0\0\0\0" bit to flag the connected state, what should this look like if the cam is disconnected?

roleoroleo, 2023-03-31:
AFAIK this bit means that the pairing process is successfully completed.
So you have to set to 0 to restart the pairing process.

meskilla, 2023-03-31:
It worked! now the welcome message is playing again. mtd enumeration must have changed from firmware 4.6 to 4.9 - with fw 4.6 the last partition is mtd5 not mtd6.

roleoroleo, 2023-03-31:
> mtd enumeration must have changed from firmware 4.6 to 4.9 - with fw 4.6 the last partition is mtd5 not mtd6.

This is strange.
The hack shouldn't change this kind of settings.

closed 2024-05-04 by github-actions[bot]