roleoroleo/yi-hack-MStar #500: Can someone help me? My camera is yi 1080p home camera 4FCN.

krishuang0610, 2023-04-12:
Can someone help me?
My camera is yi 1080p home camera 4FCN
I've been unsuccessful using y23 from the repository. I've downloaded versions from multiple repositories with no success.
[image]
[image]

roleoroleo, 2023-04-14:
What's the firmware version?

krishuang0610, 2023-04-14:
> What's the firmware version?
Thank you.
This problem has been solved. I put all the versions in the TF card, and the update has been successful. After the update is successful, it shows Yi 1080p Home 4FCN ===> h201c, not Yi 1080p Home 4FCN ==>y23 in the list.
[image]

closed 2024-05-04 by github-actions[bot]