roleoroleo/yi-hack-MStar #503: Which configuration item will disable the thumbnail feature of video files?

denven, 2023-04-30:
Hi, @roleoroleo 

The feature is enabled by default when I hacked with the completely new firmware, but it was disabled after some time of use or configuration changes. I've no idea which item will switch the video thumbnail feature.  

Can you tell me about this? Thanks

[image]

roleoroleo, 2023-05-01:
Both these settings:
[image]

denven, 2023-05-02:
> Both these settings: [image]

However, it may not take effect since I am enabling both, I just checked.

roleoroleo, 2023-05-04:
"Snapshot" option enable/disable all snapshot features.
"Snapshot for recorded video" enable/disable the creation of the thumbnails for the recorded video.
I know, it's not a very proper name, but the thumbnail for the cam is like a snapshot.

So, if you want to enable the thumbnails you need to enable both options.

denven, 2023-05-04:
No jpg files when both items are enabled.

[image]

[image]

roleoroleo, 2023-05-04:
Strange...
Try to run the process manually:
`/tmp/sd/yi-hack/script/thumb.sh cron`
Probably there is an error.

denven, 2023-05-07:
@roleoroleo I did some checks based on your advice.

- Step 1: check if custom settings are saved

```bash
/home/yi-hack # cat etc/system.conf | grep -e ^SNAPSHOT.*=yes
SNAPSHOT=yes
SNAPSHOT_VIDEO=yes
```

- Step 2: check if the already enabled service is added to crontab

```bash
/home/yi-hack # cat /var/spool/cron/crontabs/root
* * * * * /home/yi-hack/script/thumb.sh cron
0 * * * * sleep 20; /home/yi-hack/script/clean_records.sh 90
```

- Step 3: check if the scheduled service script in crontab runs into failures
```bash
/home/yi-hack # tail -f -n5 /tmp/thumb.log
2023-05-07 [01-10-00] [INFO] === SERVICE START ===
2023-05-07 [01-10-00] [WARN] Adjusting folder permissions to 0755 ...
2023-05-07 [01-10-01] [INFO] checkFiles
2023-05-07 [01-10-01] [ERROR] checkFiles: create jpg FAILED - [/tmp/sd/record/2023Y05M02D10H/50M59S01.mp4].
2023-05-07 [01-10-01] [INFO] === SERVICE STOPPED ===

/home/yi-hack # cat /tmp/thumb.log | grep "jpg FAILED"
2023-05-07 [00-40-01] [ERROR] checkFiles: create jpg FAILED - [/tmp/sd/record/2023Y05M02D10H/50M59S01.mp4].
2023-05-07 [00-41-01] [ERROR] checkFiles: create jpg FAILED - [/tmp/sd/record/2023Y05M02D10H/50M59S01.mp4].
2023-05-07 [00-42-03] [ERROR] checkFiles: create jpg FAILED - [/tmp/sd/record/2023Y05M02D10H/50M59S01.mp4].
2023-05-07 [00-43-01] [ERROR] checkFiles: create jpg FAILED - [/tmp/sd/record/2023Y05M02D10H/50M59S01.mp4].
2023-05-07 [00-44-02] [ERROR] checkFiles: create jpg FAILED - [/tmp/sd/record/2023Y05M02D10H/50M59S01.mp4]
```

So far, I know the `thumb.sh` which is used to snapshot image files doesn't work properly. Although it runs every minute, it fails each time as well, no image files will be generated.

More investigations on the failures:

- check the script `thumb.sh` source code, and find these code lines 
```bash
minimp4_yi -t 1 $file $BASE_NAME.h26x
imggrabber -f $BASE_NAME.h26x -r low -w > $BASE_NAME.jpg
```
```bash
/home/yi-hack # find / -type f -name minimp4_yi -o -name imggrabber
/home/yi-hack/bin/imggrabber
/home/yi-hack/bin/minimp4_yi
```

- Check the video file existence, since it continues to work on the same video files once failure happens and doesn't move on to new files.
```bash
/home/yi-hack # ls /tmp/sd/record/2023Y05M02D10H/50M59S01.mp4
/tmp/sd/record/2023Y05M02D10H/50M59S01.mp4
/home/yi-hack #cd /tmp/sd/record/2023Y05M02D10H/

/tmp/sd/record/2023Y05M02D10H # minimp4_yi -t 1 50M59S01.mp4 50M59S01.h26x
/tmp/sd/record/2023Y05M02D10H # imggrabber -f 50M59S01.h26x -r low -w > 50M59S01.jpg
watermark on
/tmp/sd/record/2023Y05M02D10H # echo $?
250
/tmp/sd/record/2023Y05M02D10H # ls 50M59S01.*
50M59S01.h26x  50M59S01.jpg   50M59S01.mp4
```
From the above, the image file is generated, I downloaded it and found it is not viewable. As we can see the snapshot has been generated **with an error no 250**, and the script exits after logging the error. 

```bash
	echo "${L_FILE_LIST}" | while read file; do
			if [ $? -ne 0 ]; then
				logAdd "[ERROR] checkFiles: create jpg FAILED - [${file}]."
				rm -f $BASE_NAME.h26x
				rm -f $BASE_NAME.jpg
				return 0  # line 102
			fi
      ....
      done
```
**Each time when it fails, it will continue to fail at the same place, and will not move on to the newer or other files.**

I think the video file is likely incomplete or corrupted at the time when the camera reboots or shutdown, the `imggrabber` will always fail to take a snapshot from a damaged file, which results in script `thumb.sh` interrupting at the same file.

- Change the code to ignore the error and continue even when `imggrabber` fails to grab a frame image.
```bash
			minimp4_yi -t 1 $file $BASE_NAME.h26x
			if [ $? -ne 0 ]; then
				logAdd "[ERROR] checkFiles: demux mp4 FAILED - [${file}]."
				rm -f $BASE_NAME.h26x
				# return 0
                                continue
			fi
			imggrabber -f $BASE_NAME.h26x -r low -w > $BASE_NAME.jpg
			if [ $? -ne 0 ]; then
				logAdd "[ERROR] checkFiles: create jpg FAILED - [${file}]."
				rm -f $BASE_NAME.h26x
				rm -f $BASE_NAME.jpg
				# return 0
                                continue
			fi
```
- Check the result after the above change, and I find the snapshot image files are all generated.
```bash
/tmp/sd/record # cd 2023Y05M07D09H/
/tmp/sd/record/2023Y05M07D09H # ls
06M02S58.jpg  07M00S60.jpg  26M56S03.jpg  27M00S60.jpg  28M00S60.jpg  39M18S42.jpg  40M00S60.jpg  43M20S40.jpg  44M00S60.jpg  51M24S36.jpg  52M00S60.jpg
06M02S58.mp4  07M00S60.mp4  26M56S03.mp4  27M00S60.mp4  28M00S60.mp4  39M18S42.mp4  40M00S60.mp4  43M20S40.mp4  44M00S60.mp4  51M24S36.mp4  52M00S60.mp4
```

denven, 2023-05-07:
And there is a performance issue in the `thumb.sh` script since it runs every minute with repetitive checks of all existing files.
This is not favorable, we can change it to check the single file with a timestamp close (earlier) to its run time, 1-2 minutes is ok to save resources. Most likely, all the new files generated after the time feature is enabled will be captured with the snapshots.

roleoroleo, 2023-06-04:
Thank you for your analysis.
I will change the script asap.

closed 2024-05-03 by github-actions[bot]