roleoroleo/yi-hack-MStar #505: 0.4.9 snapshot issue

marcor2196, 2023-05-26:
Hi, after the 0.4.9 update, the HTTP call to the snapshot endpoint (/cgi-bin/snapshot.sh) respond with code 200 OK but the response it's empty, don't contain the image.

Also the snapshot page in the web control page of camera don't show the image but I imagine that the service it's the same.

denven, 2023-05-31:
@marcor2196 For this issue, you can check issue #503 to see if it is the issue you have, I commented how to fix or bypass the failure in the thread.
> Also the snapshot page in the web control page of camera don't show the image but I imagine that the service it's the same.

marcor2196, 2023-06-16:
Updated to 0.5.0, I believe that the problem is not solved.

When I call the snapshot endpoint, the request sometimes remain pending and other times respond with empty result.

Other strange thing it's that the webserver needs the port 8080 to work, contrary to the changelog of the last update that say to use the 80.

[image]

roleoroleo, 2023-06-16:
About the port, new 0.5.0 version doesn't change your configuration.
It changes only the default setting (for new installations).
If the problem is not solved, probably it's not the same problem described in https://github.com/roleoroleo/yi-hack-MStar/issues/503

We can try to debug the problem.
Open a ssh connection with the cam and run the snapshot manually:
`imggrabber -m $SUFFIX -r high -w -d > /tmp/test.jpg`

marcor2196, 2023-06-16:
Here the output:

**I've removed the duplicated "copy again buffer" output, actually is repeated other several times, more than 100 times roughly.**

```
watermark on
debug on
Starting program
Resolution: 1080
VMFE0
vaddr: 0xb4b44000 - paddr: 0x235a0000 - size: 3502080
copy buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
convert YUV image
Adding watermark
Encoding jpeg image
Free memory
```

With all the "copy again buffer" it took several minutes to end the command.

roleoroleo, 2023-06-16:
> With all the "copy again buffer" it took several minutes to end the command.

This is the problem.
I don't know why, probably imggrabeer doesn't have enough cpu to read the buffer fast enough.
I will check the code.

Meanwhile, could you try to disable all unnecessary processes and check again the command above?

marcor2196, 2023-06-16:
I have only few thing activated: RTSP, NTPD, MQTT and local recording with cloud disabled.
Other intensive processes are all deactivated.

Before upgrade to 0.5.0 I have downgraded to 0.4.8 from the 0.4.9 and work correctly.
So the problem was introduced in 0.4.9.

roleoroleo, 2023-06-16:
Ok, try this binary and add another parameter to set process priority:
[imggrabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11770040/imggrabber.gz)
`imggrabber -m $SUFFIX -r high -w -p 0 -d > /tmp/test.jpg`

marcor2196, 2023-06-16:
I have tried your binary and with the new command the process is really fast, in one or two seconds it ended.

But if I tried without the parameter of process priority, the time needed it's the same as before.

roleoroleo, 2023-06-16:
This means that the process doesn't have enough cpu.
I restored the default priority.
https://github.com/roleoroleo/yi-hack-MStar/commit/106f54b52066e3d8c3253128b2a6e85d0b43f33b

Overwrite your /home/yi-hack/bin/imggrabber with this one:
[Uploading imggrabber.gz…]()

marcor2196, 2023-06-16:
Overwrited the binary, now works correctly, also the snapshot endpoint.

Thank you so much for your time and the work :)

closed 2023-06-16 by marcor2196