roleoroleo/yi-hack-MStar #506: y203c, 6FUS, 0.4.9: eventsdir is empty

jiumon13, 2023-06-02:
Hostname | yi-hack
Firmware Version | 0.4.9
Base Version | 4.5.0.0C_201910080934
Model Suffix | y203c
Serial Number | 6FUS*****
Local Time | Fri Jun 2 10:58:10 EEST 2023
Uptime | 0h 7m 44s
Load Average | 1.96 1.39 0.68
Free/Total Memory | 18740/60824 KB
Free SD Space | 100%

I tried my best to solve this problem. But success has not been achieved:
1) /cgi-bin/eventsdir.sh - answering that there is no records.
2) ssh -> ls -la /tmp/sd/record/ - also empty
3) /index.html?page=eventsdir - No events

Only two things doubt me:
1) Inside YI Home iOS apk in storage settings i've error: "error detected, formatting required", but formatting from apk doesn't help to desaper error "error detected, formatting required". Is it means my SD card is broken or something?
2) SD card capacity is small? 256MB. I suppose no, but..

Could you please give some advice? Thx.

closed 2023-06-02 by jiumon13

roleoroleo, 2023-06-04:
Yes.
If the card is not ok, the folder /tmp/record exists but it's not mounted on the card.
So records are not saved.

jiumon13, 2023-06-08:
@roleoroleo excuse me, probably you know, is it possible somehow mount network disk to camera? like smb