roleoroleo/yi-hack-MStar #508: Camera stopped working

personshelldon, 2023-06-17:
Hello! I have a Europe version of the camera (4FUS). Today YI home App stopped working (the server not responding) and I forgot my password to the camera... I updated the firmware to 0.5.0 and reset the camera. The only thing that I can do within the app now is to connect to WiFi, but the full setup can not be finished. After WiFi is connected I can access the camera through the HTTP and I fully disabled Cloud features. RTSP is enough for me. But there is a problem - after rebooting the camera I always hear "Waiting for connect..." and I need to set up WiFi again and again - the process can not be fully finished. After that, everything is working until the next reboot. All my settings are preserved.

Is there any way to set up WiFi without the Yi Home app and avoid resetting the WiFi every reboot? Thank You.

roleoroleo, 2023-06-18:
Try to setup wifi credentials from the webgui.
It should enable the flag "Ok, I'm connected".

personshelldon, 2023-06-18:
@roleoroleo I can access the configuration page, but it is infinite loading on this page (in the right top corner), and no WiFi networks are available

roleoroleo, 2023-06-19:
Ok, try the method explained in the wiki.
https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials-(deprecated-from-0.4.0)

personshelldon, 2023-06-19:
@roleoroleo thank You! This helped. But to run the scripts, I need to place them on an SD card and reset the camera again. Without the "reset", the scripts do not work.

closed 2023-06-19 by personshelldon