roleoroleo/yi-hack-MStar #509: Onvif stopped working after update to 0.5.0 - Yi 1080 Home 6FUS (4.5.0)

cklimowicz, 2023-06-19:
Hi, I have several Yi 1080 Home 6FUS (4.5.0) cameras connected to HA via Onvif.
Everything was fine until update 0.5.0.
Updated 2 cameras 0.4.9 to 0.5.0 and onvif stopped working, stream via Rstp works ok.
Cameras with version 0.4.9 work properly via onvif.

roleoroleo, 2023-06-19:
I noticed that HA is not so standard with onvif.
It doesn't understand spaces inside the xml.
I need to change some code.

Does Onvif Device Manager works?

cklimowicz, 2023-06-20:
Onvif Manager works ok.
I have connected 2 more cameras with 0.4.9 version

roleoroleo, 2023-06-20:
There was several problems, I think HA is not completely compliant with the standard specification.
But no problem, I fixed it.
I will send a new beta hack to test.

roleoroleo, 2023-06-20:
Here it is:
[y203c_0.5.0.tgz](https://github.com/roleoroleo/yi-hack-MStar/files/11800391/y203c_0.5.0.tgz)

Please try it and let me know if it's ok.

cklimowicz, 2023-06-20:
Thank you very much.
I don't know if I can test it now because I only have remote access to the cameras.
Unless I can upload via ssh.
I'll let you know

roleoroleo, 2023-06-20:
I can send you just the differences...
[update_1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11801088/update_1.tar.gz)

- Remove the folder /home/yi-hack/www/onvif
- Copy in the folder /home/yi-hack/www the content of the archive update_1

[wsd_simple_server.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11801091/wsd_simple_server.gz)
- Kill the application wsd_simple_server
- Overwrite the binary /home/yi-hack/bin/onvif_simple_server with the content of the archive wsd_simple_server

And finally reboot the cam.

cklimowicz, 2023-06-20:
Ok, I checked both variants.
First, I uploaded the files (home../sys..) to the sd card via ssh and did a reboot. 
  In HA through the onvif manager, the camera was not visible.
Second, I replaced the Onvif directory and wsd_simple_server and rebooted. 
Unfortunately the same, the camera cannot be seen through Onvif.

Finally, I uploaded the files with version 0.4.9 and restarted. 
Onvif started working again.

cklimowicz, 2023-06-20:
I noticed that version 0.5.0 lacks Onvif port configuration.
I don't know if that has anything to do with it.

[image]

roleoroleo, 2023-06-21:
With the last hack version, the old onvif_srvd daemon is deprecated and not uset anymore.
The new binary is onvif_simple_server and it's not a daemon but a cgi.
So it shares the port 80 with httpd (no need to configure it).

wsdd is also deprecated, the new binary is wsd_simple_server (this is a daemon).

Extreme79, 2023-06-22:
I'm in this issue too. Is the Xml formatting or the httpd port sharing that doesn't like to HA?

roleoroleo, 2023-06-22:
No, the problem is that the xml produced by HA uses explicit namespace.
Something like:
```
<SOAP-ENV:Envelope>
...
<ns0:GetCapabilities xmlns:ns0="http://www.onvif.org/ver10/device/wsdl">
...
```
instead of:
```
<SOAP-ENV:Envelope xmlns:tds="http://www.onvif.org/ver10/device/wsdl">
...
<GetCapabilities>
...
```
Since there is no soap/xml engine behind onvif_simple_server, it didn't work.
And another problem.
If GetCapabilitiesResponse contains the Event section:
```
                <tt:Events>
                    <tt:XAddr>%EVENTS_SERVICE_ADDRESS%</tt:XAddr>
                    <tt:WSSubscriptionPolicySupport>false</tt:WSSubscriptionPolicySupport>
                    <tt:WSPullPointSupport>false</tt:WSPullPointSupport>
                    <tt:WSPausableSubscriptionManagerInterfaceSupport>false</tt:WSPausableSubscriptionManagerInterfaceSupport>
                </tt:Events>
```
HA asks for an event subscription although ths support is false as in the example above.

I improved the parsing of the XML request and I removed the event section.
The previous release was tested with the official tools from ONVIF Project, so I'm pretty sure that was correct.

papagdog, 2023-06-26:
Same problem here. I have the y30 and h201c flavor.... No ONVIF access... I've only upgraded the y30 to this point.....  I started to follow the same instructions posted earlier, but decided against it since that version may be different..... I was on version 0.4.9 before upgrading. I may revert back for now.. Thanks in advance.

roleoroleo, 2023-06-26:
The problem is already fixed.
There are no difference between h201c and y203c about onvif daemons.
You can try them if you want.

roleoroleo, 2023-07-04:
I don't know at the moment.
If you want, I can make a beta version.

Extreme79, 2023-07-04:
If you can, I'll (we'll) appreciate it! 💚

roleoroleo, 2023-07-04:
Here it is:
[h201c_0.5.0.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11950493/h201c_0.5.0.tar.gz)
[h305r_0.5.0.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11950494/h305r_0.5.0.tar.gz)
[h307_0.5.0.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11950496/h307_0.5.0.tar.gz)
[y23_0.5.0.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11950500/y23_0.5.0.tar.gz)
[y25_0.5.0.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11950504/y25_0.5.0.tar.gz)
[y30_0.5.0.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11950508/y30_0.5.0.tar.gz)
[y203c_0.5.0.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/11950509/y203c_0.5.0.tar.gz)

TheBandit001, 2023-10-15:
Was this confirmed working ?  I have  [ Installed: 0.5.2 - Available: 0.5.2 ]  and it does same thing.   onvif doesn't work, stream via Rstp works ok. and also in the app it works.   I was on 0.4.2  or something like that.  Just decided to update..

roleoroleo, 2023-10-16:
Yes, onvif is working.
If it doesn't work with you client, please take a network capture and share it with me.

closed 2024-05-03 by github-actions[bot]