roleoroleo/yi-hack-MStar #513: Yi 1080p Dome BFUS h201c Bricked Hard

odmyko, 2023-07-16:
Hi, i have installed yi-hack-MStar 0.5.0 to my Yi 1080p Dome BFUS h201c. And it was working ok for some time.
Than, when i have changed config in web and push save, and then reboot, camera just not getting up.
It is doing nothing. No led. No moving. No sound. Board little worm.
I read a little about unbrick. I have ttl serial adapter, so  i did try to connect and check serial console (boudrate 115200), but there is also nothing.
Board is little worm and that is it.
Is there any low level recovery way?

roleoroleo, 2023-07-17:
If the led is off, probably there is a hardware problem and it's very difficult to repair.

> Is there any low level recovery way?

There is a low recovery way only if the problem is related to the flash (bootloader error).
But I've never had a problem like this.
If the flash is corrupted the problem is with the rootfs partition or the home partition. And in this case it is always possible to rewrite the partitions by simply copying the files to the SD card.
The corruption of the bootloader is very uncommon.
But you can try this recovery way.
It requires soldering skills and a programmer device.
You should desolder the chip and reprogram it using an external programmer.
Like FTDI FT232H.
There are a lot of sites about this task. For example:
https://learn.adafruit.com/programming-spi-flash-prom-with-an-ft232h-breakout/overview

odmyko, 2023-07-17:
Thanks for reply, i have FTDI FT232, and also external programmer CH314 but i will need some instructions, offsets, and bootloader binary, or instructions howto build binary. 
Sorry if im asking too much, but i do not have any infromation about this device HW.

roleoroleo, 2023-07-18:
If you program the memory with an external programmer the bootloader is at address 0x0.
The binary for h201c is here:
[mtd0.bin.gz](https://github.com/roleoroleo/yi-hack-MStar/files/12077182/mtd0.bin.gz)
The length is 0x50000.
Make a backup of your flash before writing it.
And compare the 1st part with the the file above. If the content is the same, probably is not a flash problem.

After flashing the bootloader, if the cam is ok, you should see something through the uart.

odmyko, 2023-07-18:
Thanks,
I see cfeon qh128a-104hip  (128mbit serial flash). I need to write binary into it?
[image]

roleoroleo, 2023-07-18:
Yes.
Remove, program and reinstall it.
Pay attention to the small resistors and caps near the chip.

odmyko, 2023-07-19:
Managed to write flash, but camera still leds off, propably some HW issue.
Anyway thank for help!

closed 2023-07-19 by odmyko