roleoroleo/yi-hack-MStar #515: Yi Dome 1080P v0.5.2 Reboot Loop

idanny76, 2023-07-26:
The camera reboots many times before it connects and starts working. I had the issue on the v0.4.8 and v0.4.9. I was on the v0.4.7 since it was not causing the reboot problems. Notice you released v0.5.2 with a debug log option and was hoping maybe the debug log would provide you with some insight into why its doing a reboot loop. Posted here the debug log.... hopefully it may show the problem. Like on my previous report issue on the 0.4.8 and 0.4.9, the problem was not on all the 1080p Dome cameras.

[hack_debug.log](https://github.com/roleoroleo/yi-hack-MStar/files/12174966/hack_debug.log)

ethempekin, 2023-07-26:
I think got same problem for my Dome 1080p. Which version should I downgrade for it to work hassle-free? Leaving for holidays so need to sort this out.

idanny76, 2023-07-26:
> I think got same problem for my Dome 1080p. Which version should I downgrade for it to work hassle-free? Leaving for holidays so need to sort this out.

For me version 0.4.7 did not have the reboot issue. If you are needing to reboot sometimes like me for certain reasons and the camera gets stuck on the reboot loop.... 0.4.7 is your fix.

roleoroleo, 2023-08-16:
I thought this commit might be the problem:
https://github.com/roleoroleo/yi-hack-MStar/commit/0db334bea30fad0445cbde22e4f0f973c9f54ece
when you have time to spend on this issue, please try to install 0.5.2 and rename file yi-hack/script/wd_rtsp.sh to yi-hack/script/wd_rtsp.sh.ori

closed 2024-05-03 by github-actions[bot]