roleoroleo/yi-hack-MStar #516: Not able to connect after reset

gmag11, 2023-08-04:
I've changed my network setup at home and I needed to change WiFi configuration on Yi Home BFUS camera (Yi Dome 1080). Hack was already installed and I was able to log into Web UI before.

After pushing reset button I'm not able to reconfigure camera again. I've tried to install latest release and set it up using Yi Home app. I scan QR and it shows "No Wi-Fi Networks Found" on wifi configuration. I have an access point in 2.4GHz. And there is quite a bunch of networks around my home.

I've tried to install a firmware < 4.0.0 and copy wifi configuration to SD as of [wiki](https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials-(deprecated-from-0.4.0)).

No success. I attach logs.

[log_start_reset.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/12259628/log_start_reset.tar.gz)

Is there anything more that I can do? Thanks

roleoroleo, 2023-08-19:
Try these steps:
- install a previuos version of the hack
- reset with ths back button
- configure the cam with the app
- upgrade to the last version of the hack

gmag11, 2023-08-21:
Hi, I tried downgrading to 4.0.1 what, as far as I remember, the first release I installed. Thre is no way to get wifi configured.

I wonder how is wifi settings provisioned, is it via Bluetooth? Does camera setup an AP?

I've got access to serial console but there is nothing strange there. There is no possibility to get Linux console through serial, is it?

Finally I had to install 0.3.9 and provision wifi through startup script. Now I've updated to latest version and working fine again. I haven't registered camera on cloud, but this is not important to me. Thanks.

gmag11, 2023-08-21:
Please, could you explain or point to the explanation on how app provision is technically achieved? Thanks again.

closed 2023-08-21 by gmag11

roleoroleo, 2023-08-24:
> I wonder how is wifi settings provisioned, is it via Bluetooth? Does camera setup an AP?

Several methods, it depends on the model.
1. The cam can read a qr code generated by the app.
2. The cam can hear a sound (like an old analog modem) through the microphone, generated by the app.
5. Maybe a wifi AP (but I'm not sure).

> I've got access to serial console but there is nothing strange there. There is no possibility to get Linux console through serial, is it?

Serial console should be a linux console, if I remember properly.