roleoroleo/yi-hack-MStar #520: v0.49 Directories are wrongly deleted by the auto cleaner

denven, 2023-09-03:
I think this may rarely happen, but it happened to my device today,

I wrote a real-time uploader program to upload files to OneDrive, and I check video files from OneDrive when I am not home. It works for several months. Today I found that some video files are missing, I thought the camera ran into failure or something, but I can access the camera from Yi-Home official app. Then I supposed the uploader program may have some bug results in some upload failures.

I checked the uploader history, the last time my uploader program's restart date was a long time ago, and it is working without reboot, so the camera never rebooted recently. I checked the uploaded folders in OneDrive and the existing folders in my camera sd card. I found folders between some time ranges are missing (not uploaded).
- 2023090221-2023090223
- 2023090300
- 2023090303-2023090304 (2hours)

OneDrive already has some folders that are not shown on SD card (which were deleted by yi-hack auto-cleaner), but some older directories on SD card are still there 2023Y09M01D05H, 2023Y09M01D10H.

[image]

[image]

[image]

[image]

So the issue is that: **yi-hack has some bug that it doesn't always clean files/directories from old to newer consecuitively, some newer files are deleted wrongly, and older files/directories are still kept.**

the yi camera hack auto video cleaning configuration, I set the free space to 90% (I originally took it as the threshold percentage to clean) 
[image]

denven, 2023-09-03:
I guess the cleaner may not work or calculate and find the right folders to delete properly when the used sd card storage ratio is smaller.

**Setting a smaller value like 10-50% is likely to avoid or reduce the risk of newer files/directories wrong deletion since each time it will clean only a smaller portion of the old files from the existing folders.**

roleoroleo, 2023-09-08:
The script seems to be ok, it's very simple.
Maybe there is another cleaning procedure inside the yi processes.
I didn't check this case.

denven, 2023-09-09:
> I guess the cleaner may not calculate and find the right folders to delete properly when the used SD card storage ratio is smaller.
> 
> **Setting a smaller value like 10-50% is likely to avoid or reduce the risk of newer files/directories wrong deletion since each time it will clean only a smaller portion of the old files/folders from the existing folders.**

I checked the script `clean_records.sh` code, and it looks fine, however, it doesn't work properly for `ls -lt` to sort the folders by modification time. Otherwise, I cannot understand why the folder's modification time is changed and results in the wrong sort. I dont know if any other process that will change the record folder mtime.

[image]

I am not sure if it is a bug of the `ls`, but when removing the `-t` option, it will list folders by name, I think it is okay to remove it since all the folders are named by yyyy**Y**mm**M**dd**D**hh**H** format, they will be sorted and ordered by the created time from earlier to recent time by default.

[image]

[image]

Then the fix is to change the `t` to `r`
` ls -lt | grep -v tmp | grep -v timelapse | tail -n1 | awk '{print $9}'`
to
` ls -lr | grep -v tmp | grep -v timelapse | tail -n1 | awk '{print $9}'`
``

closed 2024-05-03 by github-actions[bot]