roleoroleo/yi-hack-MStar #522: Proxychains-ng not working

vin-w, 2023-09-17:
Steps:
1.  new cam installed the firmware 0.5.2
2. proxychains added Server list in webpage
3. on restart  vi /home/yi-hack/etc/proxychains.conf => emptyfile

Another Qns: how to confirm if proxychains is running on cam?

vin-w, 2023-09-21:
fallback to old firmware and it works

closed 2023-09-21 by vin-w

roleoroleo, 2023-09-25:
> Steps:
> 
>     1. new cam installed the firmware 0.5.2
> 
>     2. proxychains added Server list in webpage
> 
>     3. on restart  vi /home/yi-hack/etc/proxychains.conf => emptyfile

Is the file empty after clicking "SAVE AND TEST CONFIGURATION" before restarting?

> Another Qns: how to confirm if proxychains is running on cam?

From webgui it's not possible, you can check with a ssh session with ps command.