roleoroleo/yi-hack-MStar #524: 0.5.3 Switch camera OFF, than ON leads to not working RTSP, ONVIF services.

thoemmes86, 2023-11-01:
Hi,

I installed the version 0.5.3 to my Yi Dome X (y30).
Since this update the services for RTSP and ONVIF, which I use for the integration with the Synology Surveillance Station, are not working anymore if you switched camera OFF and ON again.

Only a reboot of the cam gets RTSP and ONVIF working again.
I tried to use web service to restart RTSP and ONVIF instead, but it is not working.

http://.../cgi-bin/service.sh?name=rtsp&action=start
--> after sending this web request, the status changes from stopped to started. But the stream does not work at all.

http://.../cgi-bin/service.sh?name=onvif&action=start
--> after sending this web request, the status stays in stopped and will not change to started.

As it worked fine for 0.5.2, I think this is a bug new to the 0.5.3.

Kind Regards

roleoroleo, 2023-11-08:
Yes, it's a bug.
The definition of CAMERA_CONF_FILE is missing in wd_rtsp.sh

Please add it:
```
#!/bin/sh

CONF_FILE="etc/system.conf"
**CAMERA_CONF_FILE="etc/camera.conf"**

YI_HACK_PREFIX="/home/yi-hack"
MODEL_SUFFIX=$(cat /home/yi-hack/model_suffix)

#LOG_FILE="/tmp/sd/wd_rtsp.log"
LOG_FILE="/dev/null"
...

```

thoemmes86, 2023-11-10:
> Yes, it's a bug. The definition of CAMERA_CONF_FILE is missing in wd_rtsp.sh
>
> Please add it:
>
> ```
> #!/bin/sh
>
> CONF_FILE="etc/system.conf"
> CAMERA_CONF_FILE="etc/camera.conf" [...]

Tried it out and seems to fix the issue.
Cameras are working stable now for about two days.

Thank you for fixing this issue and maintaining this project.

closed 2023-11-10 by thoemmes86