roleoroleo/yi-hack-MStar #526: Online FW upgrade not working?

vin-w, 2023-11-20:
Recently I found all camera does not show available version for new firmware.

"Online FW upgrade 
Installed: 0.5.3 - Available:"

roleoroleo, 2023-11-20:
Did you filter the http/https connection?

vin-w, 2023-11-28:
I remember I once enable proxychain but disable afterwards.

Here is some output I checked.

```
/home/yi-hack/bin # wget -O -  https://api.github.com/repos/roleoroleo/yi-hack-MStar/releases/latest
Connecting to api.github.com (20.205.243.168:443)
wget: note: TLS certificate validation not implemented
wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer

/home/yi-hack/bin # cat /home/yi-hack/etc/proxychains.conf

```

roleoroleo, 2023-11-29:
proxychains is not a problem if you don't use the Yi platform.
The log `wget: note: TLS certificate validation not implemented` is normal.

This is not ok:
```
wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer
```

Check your connection, firewall/proxy, router.
It should work.
Try with a temporary tethering.

marchidaniele, 2023-12-20:
I have same problem.
I have a cam y25 with base version 4.2.0.0H_201909041620 and Firmware Version 0.4.9.

Online update don't work 
[image]

and if I connect via ssh and download latest firmware

```
/tmp/sd # wget -O -  https://api.github.com/repos/roleoroleo/yi-hack-MStar/releases/latest
Connecting to api.github.com (140.82.121.6:443)
wget: note: TLS certificate validation not implemented
wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer
```
if you download another file similar error:
```
/tmp/sd # wget https://ubuntu.mirror.garr.it/releases/jammy/ubuntu-22.04.3-live-server-amd64.iso
Connecting to ubuntu.mirror.garr.it (90.147.160.70:443)
wget: note: TLS certificate validation not implemented
wget: TLS error from peer (alert code 40): handshake failure
wget: error getting response: Connection reset by peer
```
and if I add switch --no-check-certificates don't help:
```
/tmp/sd # wget -O -  https://api.github.com/repos/roleoroleo/yi-hack-MStar/releases/latest --no-check-certificates
wget: unrecognized option '--no-check-certificates'
```

any chance for upgrade my cam?

roleoroleo, 2023-12-22:
I need to update busybox version.
Meanwhile you can upgrade your cam with this procedure:
https://github.com/roleoroleo/yi-hack-MStar/wiki/Manual-firmware-upgrade

roleoroleo, 2023-12-23:
It's a bigger problem than it initially seemed.
Currently I'm unable to fix it.

roleoroleo, 2024-01-09:
Fixed here:
https://github.com/roleoroleo/yi-hack-MStar/commit/e86b722f85cb042ca630e2b805e462eecc9db0b0

Traoma, 2024-02-04:
Hello, What's the way to upgrade to the last version with very old release? 
I'm with the 0.3.5 that has not yet the local firmware update implementation (came since 0.4.8)

roleoroleo, 2024-02-05:
Save the config.
Copy home_xxx and sys_xxx to your sd card.
Reboot.
Restore the config if you lost it.

roleoroleo, 2024-02-05:
Are you sure that you are using the correct files for your model?

Traoma, 2024-02-05:
It's OK. The problem was the sd card ... I put on old (but working) 4gb card. Tried with a new 32gb and the update has worked. Thanks

Madaraxa, 2024-04-12:
Online Update from 0.5.4 to 0.5.5 wont work on my two Mstar cams.
(my two allwinner worked from 0.3.8 to 0.3.9)

closed 2024-04-12 by roleoroleo

Extreme79, 2024-05-13:
Happened to me today, from 0.53, in 3 cam. 
Solved upgrading manually (but I lost settings).
1 identical camera, 0.53, upgraded flawlessy without loosing config.

roleoroleo, 2024-05-15:
The problem was fixed with 0.5.4.
If you come from 0.5.3 it doesn't work.

> Online Update from 0.5.4 to 0.5.5 wont work on my two Mstar cams. I guess this is fixed from 0.5.5 and above? So this is for now the last manual Update needed? (my two allwinner worked from 0.3.8 to 0.3.9)

Reading better your message, this is not correct. The update should have worked.
Let me know if you have problems with the next future update.

Madaraxa, 2025-12-19:
Hi @roleoroleo . I just tried to online Update from 0.5.5 to 0.5.6 and it didnt work 4 Times. I came to github to See if there is a problem and found the rtsp stream Not working issue which shall be fixed in 0.5.7. I stopped trying to Update now because of the upcoming 0.5.7.
But shouldnt the Update work from 0.5.5 to 0.5.6? Or did you delete 0.5.6?

Madaraxa, 2026-09-27:
Hi @roleoroleo I'm on 0.5.7 and the Web Interface Update to 0.5.8 still isnt working (See above)

[image]

Madaraxa, 2026-09-28:
Tried manual Upgrade. Downloaded correct file (y25) and renamed 0.5.8 with x.x.x. Then i put it in tmp/sd (couldnt rename it inside tmp/sd so i had to rename it Prior to SSH it to the sd card).
Clicked on Update Button. Now its stuck at 'waiting for Camera to come back online' (or however its named). I closed the Window but cant access Web if nor SSH.
I know i can get it back to working by using my backed up config plus installing like Initial install. But its wird that I have to Do it like this every Update :) my both allwinner cams have no issues with the Web if Upgrade :)

roleoroleo, 2026-09-29:
Sorry for this problem.
The only difference I see compared to other models is that the home file system size is very close to the partition limit.
But I no longer have this model and am unable to test it.

Madaraxa, 2026-09-29:
Well. Its not that bad. I can just save config, reset, hack and load config. I just wanted to 'tell' about the Problem.
Sadly i know nothing about programming. So I wont be a big help solving this issue i guess

roleoroleo, 2026-09-29:
But now that you're on 0.5.8, does the web update work?

Madaraxa, 2026-09-29:
I can check. I just install 0.5.8 over 0.5.8? Ah and how would i See if it was installed again? The Problem was that the cam restarted after 'Web Update' but didnt Update at All.
If i just Do a Web Update 0.5.8 over 0.5.8 and it restarts i dont See of its the 0.5.8 from yesterday or today.

roleoroleo, 2026-10-09:
Enter the cam with a ssh session and run the command
`echo 0.5.7 > /home/yi-hack/version`
The cam will see 0.5.7 version and you could do a web update to 0.5.8.
After the reboot you should see again 0.5.8 if the update is successfully completed.

Madaraxa, 2026-10-10:
I did this. The Main Web Page (Status) Shows 0.5.7. The Maintenance Page too. (Installed: 0.5.7 - Available: 0.5.8).
When i click on Upgrade Firmware it shows 'No SD detected'. Then it loads the 'Status' Page and is still on 0.5.7.

roleoroleo, 2026-10-13:
Is there a SD card into the cam?
You need it to update the firmware.

Madaraxa, 2026-10-13:
Ofcourse. Your hack doesnt work without sd, correct?
The 'no sd' error only showd after the Echo command in ssl that you wrote. When I tried to Update normally it just didnt Update (it said something like getting firmware, and then rebooting).
I'm just trying to help figuring Our why the online Update wont work on my cam for some Versions

roleoroleo, 2026-10-16:
> Ofcourse. Your hack doesnt work without sd, correct?

MStar hack works with or without sd card.

> I'm just trying to help figuring Our why the online Update wont work on my cam for some Versions

I'm checking source code.