roleoroleo/yi-hack-MStar #530: Changing WiFi SSID

Dmario-dev, 2024-01-07:
Hi,
I'm trying to connect the camera to the new router.
When I go to wifi confirmation page on the top there is "loading..." and the list of SSID is empty.

Is there a way to provide new SSID and password for the wifi using SSH?

BR

roleoroleo, 2024-01-07:
Yes, you can use the configure_wifi.sh script.

Create a file /tmp/configure_wifi.cfg with this content:
```
# Don't use backslash char \
# Don't use double quote " around ssid and psk
# Space chars are allowed
wifi_ssid=My AP 2.4GHz
wifi_psk=password$
```
Change "My AP 2.4GHz" with your SSID and "password$" with your password.
Run /home/yi-hack/script/configure_wifi.sh and reboot the cam.

victorclaessen, 2024-03-19:
I'm having the same problem. The ajax call returns but then the browser complains that the return value is not valid json. "SyntaxError: JSON.parse: bad control character in string literal at line 3 column 8 of the JSON data". It think line 3 column 8 is the 0x0A (line feed character). Perhaps something goes wrong in the script that returns that value?

[image]

victorclaessen, 2024-03-19:
if you change this line in `/home/yi-hack/www/cgi-bin/wifi.sh`:
```
LIST=`$YI_HACK_PREFIX/bin/iwlist wlan0 scan | grep "ESSID:" | cut -d : -f 2,3,4,5,6,7,8 | grep -v -e '^$'`
```
to
```
LIST=`$YI_HACK_PREFIX/bin/iwlist wlan0 scan | grep "ESSID:" | cut -d : -f 2 | grep -v -e '^$'`
```
then the dropdown list gets populated.

roleoroleo, 2024-03-21:
Please, try this command:
```
LIST=`$YI_HACK_PREFIX/bin/iwlist wlan0 scan | grep "ESSID:" | sed 's/^[ \t]*ESSID://g' | grep -v -e '^$'`
```

victorclaessen, 2024-03-22:
> Please, try this command:
> 
> ```
> LIST=`$YI_HACK_PREFIX/bin/iwlist wlan0 scan | grep "ESSID:" | sed 's/^[ \t]*ESSID://g' | grep -v -e '^$'`
> ```

Yes, that gives an identical result in my setup. (Checked visually and also compared output with md5sum.)

roleoroleo, 2024-03-22:
Ok, I prefer this solution to avoid problems parsing the essid when it contains special chars.

victorclaessen, 2024-03-22:
Ok, I updated the pull request for your convenience. Thanks for all the excellent work in active development on this firmware!

roleoroleo, 2024-03-22:
Thank you for your pr.

closed 2024-07-21 by github-actions[bot]