roleoroleo/yi-hack-MStar #531: ftp upload not working with 4FUS/y25/0.5.3 due to TZ translation of filename

pultars, 2024-01-15:
First and foremost, thank you @roleoroleo for your hard work on the Mstar/Allwinner/Allwinnner-v2 projects. I have been using them for a few years now and greatly appreciate it!

The problem I am having is that in ftppush.sh, the function translateFULLFNWithTz does not translate the filename properly. As a result, the file with the bad name is not found and uploaded. I think it is probably similar to https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/101. (BTW, I think I had used that fix in my Allwinner-v2 cameras!)  This ftppush was working for 4FUS/y25 in previous versions.

Also relating to TZ, when I try a zone other than GMT0, the Events page Date&Time column shows the wrong date. The directory name is correct. When entered into the directory, the Time column shows the wrong time (hourwise), but the filename has the correct time/name. The osd in the recorded file is correct.

roleoroleo, 2024-01-15:
> The problem I am having is that in ftppush.sh, the function translateFULLFNWithTz does not translate the filename properly. As a result, the file with the bad name is not found and uploaded. I think it is probably similar to [roleoroleo/yi-hack-Allwinner-v2#101](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/101). (BTW, I think I had used that fix in my Allwinner-v2 cameras!) This ftppush was working for 4FUS/y25 in previous versions.

Do you mean the problem related to the different naming convention?
The script should handle this situation automatically.
Could you please run the command manually and share the log?

About timezone, did you stop and start mp4record with web services or with another method?
Does the problem exist immediately when you start the cam?

pultars, 2024-01-16:
> Do you mean the problem related to the different naming convention? The script should handle this situation automatically. Could you please run the command manually and share the log?

The code seems to breakdown the filename/path in to chunks and then concatenate in a certain format. the TR_SUFFIX variable gets an incorrect value, and thus the concatenated string is invalid. 

The normal output of ftppush.log is:

[image]

It doesn't show much, other than the fact that none of the logAdd statements from the function uploadToFtp produced any output.

I added some logAdd statements into the translateFULLFNWithTz function to show the variable values before it returns and you can see the output here:
[image]

It seems that the TR_SUFFIX is getting an extra "S" in front and the resulting TR_RET path has two "S"s, which is a non-existing file. Also the H value seems to have changed in the TR_RET. I thought a fix similar to https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/6e084f08227871573c05f067f0ee45f23507d2f8 might work from the Allwinner-v2 issue I had mentioned might fix this, however I think you have refactored the handling of TZ since then.

> About timezone, did you stop and start mp4record with web services or with another method? Does the problem exist immediately when you start the cam?

I have been starting and stoping the camera from the ui. the first recording is of a date that does not make sense but my guess is that it is the date/time the camera has hardcoded and starts with it until ntp updates the date/time. (The 2020 file in the above screenshots is due to this I believe.) after the ntp update, the recordings have the proper filenames. You might see the errors I am experiencing about the UI not displaying the time correctly in the following screenshots:

[image]
[image]
[image]

roleoroleo, 2024-01-16:
You are right.
Please, try this beta:
[y25_0.5.3.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/13954824/y25_0.5.3.tar.gz)

closed 2024-01-19 by roleoroleo