roleoroleo/yi-hack-MStar #532: 9FUS bricked

raelbsd, 2024-01-18:
Hi folks, I was using this camera with the hack without problems for months, until suddenly died. It couldn't connect to the YI app, so I tried to reflash it from SD with the original firmware but after that it didn't boot anymore. I've solder the serial pins to see where it's failing, I took a dump of the boot process, and looks like the firmware I used is wrong version. 
[dump_yi_9FUS.txt](https://github.com/roleoroleo/yi-hack-MStar/files/13981288/dump_yi_9FUS.txt)
The board says Y203C_MB_RT1.0 2018/10/22. Do I have any chances to fix it? 
Thanks for the great work you are doing, regards

roleoroleo, 2024-01-19:
Flash the hack for y25.
It should work.

raelbsd, 2024-01-19:
Thanks! I used the y25 hack and saved me!

closed 2024-01-19 by raelbsd