roleoroleo/yi-hack-MStar #533: Found a bug from a mouse screenshot - incorrect screenshot timestamp

denven, 2024-01-20:
- Background:  I saw some animal paw prints in the snow in the yard this morning, and I was very curious about which guy had visited last night. To my surprise, I found a mouse in the camera video screenshot.  The interesting thing is that this screenshot also gave me a chance 👀 to identify a bug.

[image]

- The bug appears in the lower right corner of the screenshot. The screenshot timestamp is not correct. The camera time is 20:**29** when it was recorded, however, the timestamp put in the screenshot is not correct, it shows 20:**37**.

roleoroleo, 2024-01-20:
This is very very strange.
But I don't understand what I'm looking at.
Is it a frame of the mp4 or the related jpg.
Because the timestamp shouldn't appear in the video.

denven, 2024-01-20:
> This is very very strange. But I don't understand what I'm looking at. Is it a frame of the mp4 or the related jpg. Because the timestamp shouldn't appear in the video.
> 
> Another question: which version are you using?

It's a jpg file, you can see the red border marked filename
not sure if you can get some clue by checking `imggrabber.c` and `thumb.sh`

roleoroleo, 2024-01-20:
You are right.
The thumbnail is created from the mp4 file but the watermark is added with the wrong time.
Is the time when imggrabber is executed and not the time related to the content of the video file.
To fix this problem, I need to change imggrabber.

denven, 2024-01-22:
@roleoroleo 
some ideas about the `thumb.sh` and `imggrabber` performance:

### current situation:
1. it seems this thumbnail feature is cpu expensive, on my camera, it costs about 10% cpu usage
[image]

2. `thumb.sh` will run every minute, and `imggrabber` has delay in grab frame to generate jpg file  (like about 8 minutes in this bug)

### question:
what's the most time cost for `imggrabber`? why there is too much delay when grabbing the thumbnail file?

probably it is caused by the `checkFiles()` in `thumb.sh` , 
````bash
echo "${L_FILE_LIST}" | while read file; do
```

my thoughts:
1. make sure `imggrabber` can be called every minute, evaluate `imggrabber` process time and tune `SLEEP_CYCLE_SECONDS`
2. only use current time to grab the latest 0/1/2 video file to avoid too much delay, do not find and check from all
- 0: no video file generated for the past minute
- 1: one video file generated and not processed
- 2: one video file generated and another one is not processed in time due to time deviation (less likely)

roleoroleo, 2024-01-22:
I already trying to change the code.
The solution I'm trying is very simple.
imggrabber accept a new option useful to pass the timestamp manually.
So the script thumb.sh can call imggrabber and pass the value derived from the file name.
https://github.com/roleoroleo/yi-hack-MStar/commit/040c90af1e069d0b4b21082b7880ada2a064e718

If you want to test it:
[thumb.sh.gz](https://github.com/roleoroleo/yi-hack-MStar/files/14010396/thumb.sh.gz)
[imggrabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/14010397/imggrabber.gz)

denven, 2024-01-23:
replaced the two files and reboot, got camera infinite reboot loop. perhaps thecause is I didn't change the overwriten files permission. Then I rehacked the camera with version 0.54 firmware, I previously used 0.49.

Will find another time to test again.

denven, 2024-01-23:
tested, the time minute is close to the jpg filename, but it is always a little greater (no more than one minute) than it.

[image]

roleoroleo, 2024-01-23:
Yes.
The last version fixes a very old bug related to the time.
Now the cam uses the correct timezone (localtime).
But I changed all other pages and features related to the time, so the only difference is the time of the folder structure in /tmp/sd/record. I hope...

denven, 2024-01-24:
> Yes. The last version fixes a very old bug related to the time. Now the cam uses the correct timezone (localtime). But I changed all other pages and features related to the time, so the only difference is the time of the folder structure in /tmp/sd/record. I hope...

By adding and using `get_timestamp` function (convert UTC time to local time), we can put local time into the thumbnail file.

```bash
# input sample: /tmp/sd/record/2024Y01M24D09H/24M00S60
# will output: 2024-01-24 09:24:00
get_timestamp() {
        local_time=$(date +"%Y-%m-%d %H:%M:%S")
        utc_time=$(date -u +"%Y-%m-%d %H:%M:%S")

        local_ts=$(date -d "$local_time" +%s)
        utc_ts=$(date -d "$utc_time" +%s)

        local_utc_offset=$((local_ts-utc_ts)) # seconds

	# get utc time path without minute: 2024Y01M24D09H
        # utc_path=$(echo $1 | grep -oE "[0-9]{4}Y[0-9]{2}M[0-9]{2}D[0-9]{2}H")
        # utc_path_ts=$(date -d "${utc_path:0:4}-${utc_path:5:2}-${utc_path:8:2} ${utc_path:11:2}:00:00" +%s)
        # local_path_ts=$((utc_path_ts+local_utc_offset))

        # output local date path: 2024Y01M24D01H
        # echo $(date -d "@$local_path_ts" +"%YY%mM%dD%HH")

	# get utc time path with minute and second: 2024Y01M24D09H24M00S
        utc_path=$(echo $1 | grep -oE "[0-9]{4}Y[0-9]{2}M[0-9]{2}D[0-9]{2}H/[0-9]{2}M[0-9]S" | tr -d "/")
        utc_path_ts=$(date -d "${utc_path:0:4}-${utc_path:5:2}-${utc_path:8:2} ${utc_path:11:2}:${utc_path:14:2}:${utc_path:17:2}" +%s)
        local_path_ts=$((utc_path_ts+local_utc_offset))

        # output local time: 2024-01-24 09:24:00
        echo $(date -d "@$local_path_ts" +"%Y-%m-%d %H:%M:%S")
}

checkFiles ()
{
    ....
			# TIME_STAMP="${file:15:4}-${file:20:2}-${file:23:2} ${file:26:2}:${file:30:2}:${file:33:2}"
			TIME_STAMP=$(get_timestamp "$file")
    ....
}
```

roleoroleo, 2024-01-24:
> By adding and using get_timestamp function (convert UTC time to local time), we can put local time into the thumbnail file.

Sorry but I don't understand.
The time in the thumbnail jpeg should already be localtime.
Is it in UTC for your cam?

Could you try to switch on the option "Enable/disable time OSD" and check the time in the upper left corner?
[image]

roleoroleo, 2024-01-24:
A little fix here:
[imggrabber.gz](https://github.com/roleoroleo/yi-hack-MStar/files/14040944/imggrabber.gz)
Not related to the time zone.

roleoroleo, 2024-01-27:
How is the time in the snapshot (from web gui for example)?

> The delay is likely caused by the cron job of thumb.sh execution every minute

thumb.sh is executed with a delay, but, if you are using the version I posted above, when it runs imggrabber it passes the timestamp as an argument:

```
...
TIME_STAMP="${file:15:4}-${file:20:2}-${file:23:2} ${file:26:2}:${file:30:2}:${file:33:2}"
imggrabber -f $BASE_NAME.h26x -r low -w -t "$TIME_STAMP" > $BASE_NAME.jpg
...
```

thumb.sh reads the time from the name of the file (and the dir) and passes it to imggrabber.
Imggrabber uses this time to print the watermark.
Since the image is extracted from the first frame, the time of this frame is the same as the time extracted from the name of the file.

denven, 2024-01-27:
> How is the time in the snapshot (from web gui for example)?
>
>
>
>
>
>
> thumb.sh is executed with a delay, but, if you are using the version I posted above, when it runs imggrabber it passes the timestamp [...]

currently, the time in thumbnail is correct after using your new imggrabber. However, it is still in UTC time.
z
The video records directories are orgnized in UTC date hour time. do you convert the time to local timezone in imggrabber?

roleoroleo, 2024-01-27:
Something is wrong, I don't understand.
With the last release, the file/folder structure in /tmp/sd/record should be oganized in local time. Not in UTC.

I will send a whole image to test.

denven, 2024-01-27:
> Something is wrong, I don't understand. With the last release, the file/folder structure in /tmp/sd/record should be oganized in local time. Not in UTC.
> 
> I will send a whole image to test.

[image]

I noticed that the record folders' naming was changed to UTC since version 0.49 or earlier.
I thought it was a universal change since that version, so that's why I had to write the time conversion code to handle this.

I am not sure which setup of mine is not correct or if some other thing is wrong at my end because you've already adjusted this.

roleoroleo, 2024-01-28:
This commit (included in 0.5.4) changes the behavior of the cam.
The cam should work in localtime and save the files/folders in localtime.
I don't understand why your cam works in UTC...

A question.
What time does the cam use when cloud is on?

denven, 2024-01-28:
from the Yi App stream, the OSD time is local. I think I never turned off the cloud future, but I don't use their cloud storage plans.

[image]

denven, 2024-01-30:
@roleoroleo 
I've mentioned that the camera stops to record video files in the previous [issue discussion](https://github.com/roleoroleo/yi-hack-MStar/issues/533#issuecomment-1909686617) as well.

And it is not true, it is recording files, and using the local time to do it, the files in utc-time directories are overwritten by the new files.
[image]

So now the issue is **when OSD is disabled, the camera still uses UTC time to organize folders.**

roleoroleo, 2024-02-14:
Check this issue:
https://github.com/roleoroleo/yi-hack-MStar/issues/436

closed 2024-06-14 by github-actions[bot]