roleoroleo/yi-hack-MStar #538: Error from stream worker: Error opening stream (ERRORTYPE_111, Connection refused) rtsp://192.168.1.187/ch0_0.h264

GiaMisu, 2024-02-16:
Hi, I'm looking for support for my Yi-hack MStar, i have a log issue in home assistant

my configuration:

Firmware Version | 0.5.4
Base Version | 4.5.0.0C_201910080934
Model Suffix | y203c
Serial Number | 4FUSY21KO********

/home/yi-hack/etc # cat system.conf
HTTPD=yes
TELNETD=yes
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=yes
RTSP=yes
RTSP_ALT=no
RTSP_STREAM=high
RTSP_AUDIO=pcm
RTSP_AUDIO_NR_LEVEL=0
SPEAKER_AUDIO=yes
SNAPSHOT=yes
SNAPSHOT_VIDEO=no
SNAPSHOT_LOW=no
TIMELAPSE=no
TIMELAPSE_FTP=no
TIMELAPSE_FTP_SAME_NAME=no
TIMELAPSE_DT=60
TIMELAPSE_VDT=
ONVIF=yes
ONVIF_WSDD=yes
ONVIF_PROFILE=high
ONVIF_WM_SNAPSHOT=yes
TIME_OSD=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
RTSP_PORT=554
HTTPD_PORT=80
USERNAME=
PASSWORD=
FREE_SPACE=0
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=
CRONTAB=
DEBUG_LOG=no

Log error:

Logger: homeassistant.components.stream.stream.camera.yi_hack_m_03c4bb_cam
Source: components/stream/__init__.py:431
Integration: Stream ([documentation](https://www.home-assistant.io/integrations/stream), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+stream%22))
First occurred: 09:07:28 (6 occurrences)
Last logged: 09:11:38
Error from stream worker: Error opening stream (ERRORTYPE_111, Connection refused) rtsp://192.168.1.187/ch0_0.h264

The binary_sensor.yi_hack_m_03c4bb_motion_detection always remains active.

[image]

[image]

I formatted the SD and flashed the firmware several times, also downgrading to version 0.5.2 and then bringing it back to version 0.5.4 by performing the flash manually following the procedure described but the error messages continue to appear, I also deleted the yi-hack_ha_integration integration by deleting it from the folder inside HA and restarting HA and then reinstalling it, strange that everything works correctly anyway.

I honestly don't understand what I can do to eliminate the error logs, I've looked for guides but none have solved the problem, RTSP works, otherwise streaming wouldn't work but why the message?

roleoroleo, 2024-02-16:
I think the cam is disabled.
Swith on the cam with the switch in ha.

GiaMisu, 2024-02-16:
> I think the cam is disabled. Swith on the cam with the switch in ha.

the cam activates and deactivates without problems from HA but the RTSP connection refused error appears in the logs when it is turned off and on lovelace the cam goes off and then becomes "unavailable" and after 1 minute it correctly returns to off, it is strange behavior, furthermore after updating to version 0.5.4 the entity The binary_sensor.yi_hack_m_03c4bb_motion_detection always remains active, the "sgombro" function does not seem to work correctly as you can see in the image I posted above.

closed 2024-06-16 by github-actions[bot]