roleoroleo/yi-hack-MStar #540: RTSP not working for Synology Surveillance Station (9.1.4-11002)

lightxx, 2024-03-20:
So, I flashed my YI Cam (4FUSY) with 0.5.4, and I can't get it to work with  Surveillance Station.
I can add it, get its capabilities, get a still image, but after activation Surveillance Station says it's disconnected. 

Works with VLC of course. 

Tried both live555 and the alternative RTSP server.

Any ideas?

roleoroleo, 2024-03-20:
> Maybe a regression?

I completely changed the onvif daemon.
If you tell me your model suffix, I will send a beta for testing.

lightxx, 2024-03-21:
Thank you so much.
y203c

roleoroleo, 2024-03-21:
Here it is:
[y203c_0.5.4.tgz](https://github.com/roleoroleo/yi-hack-MStar/files/14691510/y203c_0.5.4.tgz)

lightxx, 2024-03-21:
Awesome. Great job. Works. Thank you so much dude!

roleoroleo, 2024-03-21:
Which onvif option do you use to make it work?

lightxx, 2024-03-21:
Uhm… I just flashed your version and added the cam in surveillance station. Defaults everywhere. Worked out of the box.

lightxx, 2024-03-21:
Is there any information I can provide you with?

lightxx, 2024-03-21:
Resolved

closed 2024-03-21 by lightxx

roleoroleo, 2024-03-21:
I don't understand.
I added a special option to improve compatibility with Synology.
[image]

But it works without settings this option.

lightxx, 2024-03-21:
I don’t understand it either then.
The beta you provided me with works out of the box with surveillance station. I didn’t set any options.