roleoroleo/yi-hack-MStar #542: Error in "Manual firmware upgrade" wiki?

Madaraxa, 2024-04-12:
Hey,
I just tried the online update on my two MStar yicams. It didnt work (from 0.5.4 to 0.5.5.). It just rebooted and showed 0.5.4 again.

So I did the manual update on one of the cams like its described in the wiki (downloaded the file, renamed it with x.x.x on the cam in the correct folder, used the Update button in Web interface)
Since then the cam wont boot up (yellow light is on and stays on the whole time, no blinking, nothing).

On my second MStar I unzipped the tgz file and transfered the tar file to my cam, renamed it with x.x.x and this cam updated "manual" without problems.
Is there a typo in your manual? Because there it says the tgz should be renamed and copied.

both are 9FUS (y25) cams.

roleoroleo, 2024-04-12:
> I just tried the online update on my two MStar yicams. It didnt work (from 0.5.4 to 0.5.5.).

Does "Maintenance page" showed the label "Available: local SD"?

As you can read in the code here:
https://github.com/roleoroleo/yi-hack-MStar/blob/42d08f9ff709ffa8b6f179a21a53e143e61b9e30/src/www/httpd/cgi-bin/fw_upgrade.sh#L65
the cam doesn't use a file with .tar extension.
If ${MODEL_SUFFIX}_x.x.x.tgz exists, then process it.
Probably you copied home_y25 and sys_25 to your sd card. In this case, update procedure works.

Madaraxa, 2024-04-12:
Thats really wierd as I am 100% sure that I only copied the tar file to the second yi.
Maybe somehow the yi got the online Update through the Internet the second time and didnt use the tar at all... I only tried online Update once before I did the manual Update.
I got the First one running again after reset Into Stock Firmware Into reflash Into importing settings.

closed 2024-08-11 by github-actions[bot]